enjoy-digital/litex #1415: litex_json2renode for Vexriscv-smp

amr-25, 2022-09-03:
The renode script generated by litex_json2renode is unable to take flash binary.  It gives, Warning! There is no flash memory to load binaries to. I input the csr.json file that is generated inside the directory /build/arty_a7/ as follow:

$ litex_json2renode csr.json --bios-binary=opensbi.bin --firmware-binary=Image --flash-binary=rootfs.cpio:0x41000000 --repl=thisplat --resc=thiscript
Skipping `opensbi` since it is included in `main_ram`
Skipping io region: ethmac
Skipping io region: csr
Skipping io region: clint
Skipping io region: plic
Skipping unsupported peripheral `ethphy` at 0xf0002800
Skipping unsupported peripheral `icap_bit` at 0xf0003800
Skipping unsupported peripheral `identifier_mem` at 0xf0004000
Skipping unsupported peripheral `leds` at 0xf0004800
Skipping unsupported peripheral `mmcm` at 0xf0005000
Skipping unsupported peripheral `rgb_led_b0` at 0xf0005800
Skipping unsupported peripheral `rgb_led_g0` at 0xf0006000
Skipping unsupported peripheral `rgb_led_r0` at 0xf0006800
Skipping unsupported peripheral `sdblock2mem` at 0xf0007000
Skipping unsupported peripheral `sdcore` at 0xf0007800
Skipping unsupported peripheral `sdirq` at 0xf0008000
Skipping unsupported peripheral `sdmem2block` at 0xf0008800
Skipping unsupported peripheral `spiflash_core` at 0xf000a800
Skipping unsupported peripheral `switches` at 0xf000b000
Skipping unsupported peripheral `xadc` at 0xf000b800
Warning! There is no flash memory to load binaries to

I just want to run my own elf file, so I tried to change the change the rootfs.cpio with my own rootfs.cpio in default existing renode.resc script but it doesn't work that way. So I am trying to generate the scripts, platform files through litex_json2renode.

P.S: I used $ ./make.py --board=arty_a7 --cpu-count=2 --dcache-size=4096 -icache-size=4096 --build 
to build arty_a7.bit and csr.json files. I think renode doesn't support 8192 as -dcache-size, so had to choose 4096.

mateusz-holenko, 2022-09-05:
Thanks @amr-25 for reporting the issue, we'll look into it.

MateuszKarlic, 2022-10-03:
For the record, we’re working on the update for this script. This is still in progress but a sample repl for multicore VexRiscv can be found here: https://github.com/renode/renode/issues/374#issuecomment-1254859841

MateuszKarlic, 2022-11-08:
We now have a PR that adds the multicore support ready here: https://github.com/enjoy-digital/litex/pull/1496

enjoy-digital, 2024-01-14:
We can probably close now that https://github.com/enjoy-digital/litex/pull/1496 is merged.

closed 2024-01-14 by enjoy-digital