enjoy-digital/litex #1417: axi stream buffer

1n3o1, 2022-09-03:
thanks in advance 
should not you buffer in axi stream be like this 
class Buffer(Module):
    """Pipe ready and valid and all data to cut timing path"""
    def __init__(self, layout):
        self.sink   = sink   = Endpoint(layout)
        self.source = source = Endpoint(layout)
        
        # # #

        bufready = PipeReady(layout)
        self.submodules += bufready
        
        bufvalid = PipeValid(layout)
        self.submodules += bufvalid
        
        self.comb += sink.connect(bufready.sink)
        self.comb += bufready.source.connect(bufvalid.sink)
        self.comb += bufvalid.source.connect(source)
instead of 
class Buffer(PipeValid): pass # FIXME: Replace Buffer with PipeValid in codebase?

I mean to be a buffer and to break the latency path you have to buffer both valid and ready signal of axi stream 
(skid buffer ) but this buffer that has been used inside buffered endpoint of dma only buffer valid signal so its no buffer that 
breaks the latency 
I have tried the new build of litepcie with dma with this new buffer it improved timing a bit

enjoy-digital, 2022-09-07:
Thanks @1n3o1 for the feedback. Depending the case, you'll sometimes only want to pipeline valid/data, sometimes only ready and in other cases both (what you call a skid buffer). 

For the Buffer, we could in fact integrate both PipeValid/PipeReady and allow enabling/disabling both through parameters.

I'll have a look at this.

enjoy-digital, 2022-09-07:
Buffer can now integrate PipeValid and/or PipeReady with https://github.com/enjoy-digital/litex/commit/a6acfb9a37e121f5458fd1bf6139f85e7114e503. For now default behavior defaults to the previous one (PipeValid). More tests would need to be done on designs to see if we should defaults to both PipeValid and PipeReady.

enjoy-digital, 2024-01-14:
Closing since implemented.

closed 2024-01-14 by enjoy-digital