enjoy-digital/litex #1421: axprot signals error while using axi_lite

bilalahmed-RS, 2022-09-08:
File "/nfs_scratch/scratch/litex/moin_ijaz/litex_public/litex/litex/soc/interconnect/stream.py", line 80, in __getattr__
    return getattr(object.__getattribute__(self, "param"), name)
AttributeError: 'Record' object has no attribute 'prot'

We have seen in some examples  of yours as well, that this signal is tied to zero.

enjoy-digital, 2022-09-08:
Hi @bilalahmed-RS,

thanks, can you provide a minimal code snippet to be able to reproduce and have a closer look at the issue and fix it if this is an issue? The more precise the issue will be described, the easier it will be for me (while also maximizing the chance to do what you want).

enjoy-digital, 2022-09-08:
@bilalahmed-RS: I think https://github.com/enjoy-digital/litex/commit/e5de4b356a3e4b20278e1aa58bba175f411cc2a4 should do it, please re-open if not.

closed 2022-09-08 by enjoy-digital