enjoy-digital/litex #1422: axregion and axuser signals in axi full produce error

bilalahmed-RS, 2022-09-08:
In the bellow wrapper you can see that these inputs are tied to zero and outputs are set Open().

https://github.com/enjoy-digital/litex_verilog_axi_test/blob/fedb6e507265010a98fe1e0c89cc0823ca9c34ff/verilog_axi/axi/axi_fifo.py#L137

enjoy-digital, 2022-09-09:
Thanks @bilalahmed-RS, this is now possible with https://github.com/enjoy-digital/litex/commit/15f72174cef1bfa24d43f8654b697e740300d540 and demonstrated by https://github.com/enjoy-digital/litex_verilog_axi_test/commit/cf072cf304cd96f7e1dfe83fb8e0f288db6b2769.

closed 2022-09-09 by enjoy-digital

bilalahmed-RS, 2022-09-09:
@enjoy-digital axregion is working, however, axuser is still throwing the following errors.

"/nfs_scratch/scratch/litex/moin_ijaz/RapidSilicon/dev2/IP_Catalog/RapidSilicon/IP/axi_fifo/v1_0/litex_sim/axi_fifo_litex_wrapper.py", line 154, in __init__
    o_m_axi_awuser      = m_axi.aw.user,
  File "/nfs_scratch/scratch/litex/moin_ijaz/litex_public/litex/litex/soc/interconnect/stream.py", line 80, in __getattr__
    return getattr(object.__getattribute__(self, "param"), name)
AttributeError: 'Record' object has no attribute 'user'

enjoy-digital, 2022-09-09:
@bilalahmed-RS: The signal is not present when width is set to 0. To have it, you have to set `aw_user_width=x` in the AXIInterface parameters (this is similar for the other channels).