enjoy-digital/litex #1427: litex-micropython: Missing header files and "ABI is incompatible with that of the selected emulation"

r4d10n, 2022-09-13:
While trying to compile litex-micropython following the instructions in the [wiki](https://github.com/enjoy-digital/litex/wiki/Run-MicroPython-CircuitPython-On-Your-SoC), I am getting the following errors:

```
riscv64-unknown-elf-gcc -DFFCONF_H=\"lib/oofatfs/ffconf.h\" -DFFCONF_H=\"lib/oofatfs/ffconf.h\" -march=rv32i2p0_m     -mabi=ilp32 -D__vexriscv__ -I. -Ibuild -I../.. -I/home/rax/lr/litex-boards/litex_boards/targets/build/digilent_arty/software/include -I/home/rax/lr/litex/litex/soc/software/include -I/home/rax/lr/litex/litex/soc/cores/cpu/vexriscv -Wall -Wno-error -std=gnu11 -ggdb  -Og -Wdouble-promotion -c -MD -o build//home/rax/lr/litex/litex/soc/software/liblitesdcard/sdcard.o /home/rax/lr/litex/litex/soc/software/liblitesdcard/sdcard.c
/home/rax/lr/litex/litex/soc/software/liblitesdcard/spisdcard.c:16:10: fatal error: libfatfs/ff.h: No such file or directory
   16 | #include <libfatfs/ff.h>
      |          ^~~~~~~~~~~~~~~
compilation terminated.
make: *** [../../py/mkrules.mk:77: build//home/rax/lr/litex/litex/soc/software/liblitesdcard/spisdcard.o] Error 1
make: *** Waiting for unfinished jobs....
/home/rax/lr/litex/litex/soc/software/liblitesdcard/sdcard.c:17:10: fatal error: libfatfs/ff.h: No such file or directory
   17 | #include <libfatfs/ff.h>
      |          ^~~~~~~~~~~~~~~
compilation terminated.
make: *** [../../py/mkrules.mk:77: build//home/rax/lr/litex/litex/soc/software/liblitesdcard/sdcard.o] Error 1
```
If I comment out the sdcard references in Makefile, I am getting this error: 

> ABI is incompatible with that of the selected emulation:   target emulation `elf64-littleriscv' does not match `elf32-littleriscv'

```
LINK build/firmware.elf
riscv64-unknown-elf-gcc -nostartfiles -T linker.ld -Wl,-Map=build/firmware.elf.map -Wl,--cref -L/home/rax/lr/litex-boards/litex_boards/targets/build/digilent_arty/software/include -march=rv32i2p0_m -mabi=ilp32  -o build/firmware.elf build/software/libbase/crt0.o build/py/mpstate.o build/py/nlr.o build/py/nlrx86.o build/py/nlrx64.o build/py/nlrthumb.o build/py/nlraarch64.o build/py/nlrpowerpc.o build/py/nlrxtensa.o build/py/nlrsetjmp.o build/py/malloc.o build/py/gc.o build/py/pystack.o build/py/qstr.o build/py/vstr.o build/py/mpprint.o build/py/unicode.o build/py/mpz.o build/py/reader.o build/py/lexer.o build/py/parse.o build/py/scope.o build/py/compile.o build/py/emitcommon.o build/py/emitbc.o build/py/asmbase.o build/py/asmx64.o build/py/emitnx64.o build/py/asmx86.o build/py/emitnx86.o build/py/asmthumb.o build/py/emitnthumb.o build/py/emitinlinethumb.o build/py/asmarm.o build/py/emitnarm.o build/py/asmxtensa.o build/py/emitnxtensa.o build/py/emitinlinextensa.o build/py/emitnxtensawin.o build/py/formatfloat.o build/py/parsenumbase.o build/py/parsenum.o build/py/emitglue.o build/py/persistentcode.o build/py/runtime.o build/py/runtime_utils.o build/py/scheduler.o build/py/nativeglue.o build/py/pairheap.o build/py/ringbuf.o build/py/stackctrl.o build/py/argcheck.o build/py/warning.o build/py/profile.o build/py/map.o build/py/obj.o build/py/objarray.o build/py/objattrtuple.o build/py/objbool.o build/py/objboundmeth.o build/py/objcell.o build/py/objclosure.o build/py/objcomplex.o build/py/objdeque.o build/py/objdict.o build/py/objenumerate.o build/py/objexcept.o build/py/objfilter.o build/py/objfloat.o build/py/objfun.o build/py/objgenerator.o build/py/objgetitemiter.o build/py/objint.o build/py/objint_longlong.o build/py/objint_mpz.o build/py/objlist.o build/py/objmap.o build/py/objmodule.o build/py/objobject.o build/py/objpolyiter.o build/py/objproperty.o build/py/objnone.o build/py/objnamedtuple.o build/py/objrange.o build/py/objreversed.o build/py/objset.o build/py/objsingleton.o build/py/objslice.o build/py/objstr.o build/py/objstrunicode.o build/py/objstringio.o build/py/objtuple.o build/py/objtype.o build/py/objzip.o build/py/opmethods.o build/py/sequence.o build/py/stream.o build/py/binary.o build/py/builtinimport.o build/py/builtinevex.o build/py/builtinhelp.o build/py/modarray.o build/py/modbuiltins.o build/py/modcollections.o build/py/modgc.o build/py/modio.o build/py/modmath.o build/py/modcmath.o build/py/modmicropython.o build/py/modstruct.o build/py/modsys.o build/py/moduerrno.o build/py/modthread.o build/py/vm.o build/py/bc.o build/py/showbc.o build/py/repl.o build/py/smallint.o build/py/frozenmod.o build/extmod/moduasyncio.o build/extmod/moductypes.o build/extmod/modujson.o build/extmod/modure.o build/extmod/moduzlib.o build/extmod/moduheapq.o build/extmod/modutimeq.o build/extmod/moduhashlib.o build/extmod/moducryptolib.o build/extmod/modubinascii.o build/extmod/virtpin.o build/extmod/machine_mem.o build/extmod/machine_pinbase.o build/extmod/machine_signal.o build/extmod/machine_pulse.o build/extmod/machine_i2c.o build/extmod/machine_spi.o build/extmod/modbluetooth.o build/extmod/modussl_axtls.o build/extmod/modussl_mbedtls.o build/extmod/modurandom.o build/extmod/moduselect.o build/extmod/moduwebsocket.o build/extmod/modwebrepl.o build/extmod/modframebuf.o build/extmod/vfs.o build/extmod/vfs_blockdev.o build/extmod/vfs_reader.o build/extmod/vfs_posix.o build/extmod/vfs_posix_file.o build/extmod/vfs_fat.o build/extmod/vfs_fat_diskio.o build/extmod/vfs_fat_file.o build/extmod/vfs_lfs.o build/extmod/utime_mphal.o build/extmod/uos_dupterm.o build/lib/embed/abort_.o build/lib/utils/printf.o build/main.o build/isr.o build/modmachine.o build/moduos.o build/modutime.o build/modlitex.o build/litex_led.o build/litex_dma.o build/litex_video.o build/machine_hw_spi.o build/machine_timer.o build/machine_pwm.o build/machine_pin.o build/lib/utils/sys_stdio_mphal.o build/lib/utils/stdout_helpers.o build/lib/utils/interrupt_char.o build/lib/utils/pyexec.o build/lib/timeutils/timeutils.o build/lib/mp-readline/readline.o build/drivers/bus/softspi.o build//home/rax/lr/litex/litex/soc/software/libbase/system.o build//home/rax/lr/litex/litex/soc/software/libbase/uart.o -Wl,-lc -Wl,-lm
/home/rax/lr/riscv64-unknown-elf-gcc-10.1.0-2020.08.2-x86_64-linux-ubuntu14/bin/../lib/gcc/riscv64-unknown-elf/10.1.0/../../../../riscv64-unknown-elf/bin/ld: /home/rax/lr/riscv64-unknown-elf-gcc-10.1.0-2020.08.2-x86_64-linux-ubuntu14/bin/../lib/gcc/riscv64-unknown-elf/10.1.0/../../../../riscv64-unknown-elf/lib/libc.a(lib_a-memcmp.o): ABI is incompatible with that of the selected emulation:
  target emulation `elf64-littleriscv' does not match `elf32-littleriscv'
/home/rax/lr/riscv64-unknown-elf-gcc-10.1.0-2020.08.2-x86_64-linux-ubuntu14/bin/../lib/gcc/riscv64-unknown-elf/10.1.0/../../../../riscv64-unknown-elf/bin/ld: failed to merge target specific data of file /home/rax/lr/riscv64-unknown-elf-gcc-10.1.0-2020.08.2-x86_64-linux-ubuntu14/bin/../lib/gcc/riscv64-unknown-elf/10.1.0/../../../../riscv64-unknown-elf/lib/libc.a(lib_a-memcmp.o)
/home/rax/lr/riscv64-unknown-elf-gcc-10.1.0-2020.08.2-x86_64-linux-ubuntu14/bin/../lib/gcc/riscv64-unknown-elf/10.1.0/../../../../riscv64-unknown-elf/bin/ld: /home/rax/lr/riscv64-unknown-elf-gcc-10.1.0-2020.08.2-x86_64-linux-ubuntu14/bin/../lib/gcc/riscv64-unknown-elf/10.1.0/../../../../riscv64-unknown-elf/lib/libc.a(lib_a-memcpy.o): ABI is incompatible with that of the selected emulation:
  target emulation `elf64-littleriscv' does not match `elf32-littleriscv'

... <snip> ...

/home/rax/lr/riscv64-unknown-elf-gcc-10.1.0-2020.08.2-x86_64-linux-ubuntu14/bin/../lib/gcc/riscv64-unknown-elf/10.1.0/../../../../riscv64-unknown-elf/bin/ld: failed to merge target specific data of file /home/rax/lr/riscv64-unknown-elf-gcc-10.1.0-2020.08.2-x86_64-linux-ubuntu14/bin/../lib/gcc/riscv64-unknown-elf/10.1.0/libgcc.a(_clzsi2.o)
/home/rax/lr/riscv64-unknown-elf-gcc-10.1.0-2020.08.2-x86_64-linux-ubuntu14/bin/../lib/gcc/riscv64-unknown-elf/10.1.0/../../../../riscv64-unknown-elf/bin/ld: /home/rax/lr/riscv64-unknown-elf-gcc-10.1.0-2020.08.2-x86_64-linux-ubuntu14/bin/../lib/gcc/riscv64-unknown-elf/10.1.0/libgcc.a(_clz.o): ABI is incompatible with that of the selected emulation:
  target emulation `elf64-littleriscv' does not match `elf32-littleriscv'
/home/rax/lr/riscv64-unknown-elf-gcc-10.1.0-2020.08.2-x86_64-linux-ubuntu14/bin/../lib/gcc/riscv64-unknown-elf/10.1.0/../../../../riscv64-unknown-elf/bin/ld: failed to merge target specific data of file /home/rax/lr/riscv64-unknown-elf-gcc-10.1.0-2020.08.2-x86_64-linux-ubuntu14/bin/../lib/gcc/riscv64-unknown-elf/10.1.0/libgcc.a(_clz.o)
[... cut ...] -memcmp.o): file class ELFCLASS64 incompatible with ELFCLASS32
/home/rax/lr/riscv64-unknown-elf-gcc-10.1.0-2020.08.2-x86_64-linux-ubuntu14/bin/../lib/gcc/riscv64-unknown-elf/10.1.0/../../../../riscv64-unknown-elf/bin/ld: final link failed: file in wrong format
collect2: error: ld returned 1 exit status
make: *** [Makefile:142: build/firmware.elf] Error 1
```

Litex and riscv-gcc are updated to latest git revision. I tried to compile riscv-toolchain with multilib support. The same error persists.

enjoy-digital, 2022-09-15:
Hi @r4d10n,

@suarezvictor would probably be able to help on this since has been working on the micropython port recently.

suarezvictor, 2022-09-16:
Yes let's start to check the branch used, at least the version I'm using wotks ok with my toolchain

r4d10n, 2022-09-16:
Thanks Florent and Victor... 

The branch I tried was: https://github.com/litex-hub/micropython/tree/litex-rebase

Toolchains tried: riscv64-unknown-elf-gcc-10.1.0-2020.08.2-x86_64-linux
                           riscv64-unknown-elf-gcc-8.3.0-2019.08.0-x86_64-linux

OS: Manjaro Linux and Ubuntu 20.04.1

suarezvictor, 2022-09-17:
a line in Makefile needs to correctly set LIBLITESDCARD_DIRECTORY, but after invoking variables.mak (so, needs to be moved below):

```
include $(BUILD_DIRECTORY)/software/include/generated/variables.mak
LIBLITESDCARD_DIRECTORY = liblitesdcard
```
This is fixed since long ago in the current development repo (on my account), but if the one at litex-hub works, I'd recommend it

r4d10n, 2022-09-17:
For the litex-hub repo, the above fix solves the first sdcard issue. I am still facing the ABI incompatibility error. 

I tried checking out your dev repo. After modifying Makefile to remove references to files in third-party directory and some header files (<liblitesdk/litex.h> in main.c and <litex.h> in isr.c), my system is reporting the same ABI incompatibility error during linking. 

Could you let me know the riscv toolchain version being used ?

suarezvictor, 2022-09-17:
Let's stick with the version on litex-hub, it should work. I'm using riscv64-unknown-elf-gcc 8.3.0

```
$ riscv64-unknown-elf-gcc --version
riscv64-unknown-elf-gcc (SiFive GCC 8.3.0-2019.08.0) 8.3.0
Copyright (C) 2018 Free Software Foundation, Inc.
This is free software; see the source for copying conditions.  There is NO
warranty; not even for MERCHANTABILITY or FITNESS FOR A PARTICULAR PURPOSE.
```

suarezvictor, 2022-09-17:
As seen (just so far), the ABI issue has to do with a change in vexriscv core (litex/soc/cores/cpu/vexriscv/core.py):
all intances of `-march=rv32i` were changed to `-march=rv32i2p0` and that prevented compilation

suarezvictor, 2022-09-17:
as tried on the litex-hub repo:
```
$ litex_term --speed 1000000 --kernel build/firmware.bin /dev/ttyUSB1
```
and in another console
```
 ./digilent_arty.py --timer-uptime --uart-baudrate=1000000 --with-pmod-gpio --with-spi-sdcard --ident "LiteX" --build --load
```

gives
```
        __   _ __      _  __
       / /  (_) /____ | |/_/
      / /__/ / __/ -_)>  <
     /____/_/\__/\__/_/|_|
   Build your hardware, easily!

 (c) Copyright 2012-2022 Enjoy-Digital
 (c) Copyright 2007-2015 M-Labs

 BIOS built on Sep 17 2022 13:06:00
 BIOS CRC passed (a0d79f06)

 Migen git sha1: ac70301
 LiteX git sha1: 09ae2f10

--=============== SoC ==================--
CPU:		VexRiscv @ 100MHz
BUS:		WISHBONE 32-bit @ 4GiB
CSR:		32-bit data
ROM:		128KiB
SRAM:		8KiB
L2:		8KiB
SDRAM:		262144KiB 16-bit @ 800MT/s (CL-7 CWL-5)

--========== Initialization ============--
Initializing SDRAM @0x40000000...
Switching SDRAM to software control.
Read leveling:
  m0, b00: |00000000000000000000000000000000| delays: -
  m0, b01: |00000000000000000000000000000000| delays: -
  m0, b02: |11111111111100000000000000000000| delays: 06+-06
  m0, b03: |00000000000001111111111111100000| delays: 20+-07
  m0, b04: |00000000000000000000000000000111| delays: 31+-02
  m0, b05: |00000000000000000000000000000000| delays: -
  m0, b06: |00000000000000000000000000000000| delays: -
  m0, b07: |00000000000000000000000000000000| delays: -
  best: m0, b03 delays: 20+-07
  m1, b00: |00000000000000000000000000000000| delays: -
  m1, b01: |00000000000000000000000000000000| delays: -
  m1, b02: |11111111111100000000000000000000| delays: 05+-05
  m1, b03: |00000000000000111111111111110000| delays: 21+-07
  m1, b04: |00000000000000000000000000000011| delays: 00+-02
  m1, b05: |00000000000000000000000000000000| delays: -
  m1, b06: |00000000000000000000000000000000| delays: -
  m1, b07: |00000000000000000000000000000000| delays: -
  best: m1, b03 delays: 21+-07
Switching SDRAM to hardware control.
Memtest at 0x40000000 (2.0MiB)...
  Write: 0x40000000-0x40200000 2.0MiB     
   Read: 0x40000000-0x40200000 2.0MiB     
Memtest OK
Memspeed at 0x40000000 (Sequential, 2.0MiB)...
  Write speed: 37.0MiB/s
   Read speed: 48.7MiB/s

--============== Boot ==================--
Booting from serial...
Press Q or ESC to abort boot completely.
sL5DdSMmkekro
[LITEX-TERM] Received firmware download request from the device.
[LITEX-TERM] Uploading build/firmware.bin to 0x40000000 (335256 bytes)...
[LITEX-TERM] Upload calibration... (inter-frame: 10.00us, length: 64)
[LITEX-TERM] Upload complete (85.9KB/s).
[LITEX-TERM] Booting the device.
[LITEX-TERM] Done.
Executing booted program at 0x40000000

--============= Liftoff! ===============--
MicroPython v1.16-61-gc9e72b5fc on 2022-09-17; LiteX SoC with VexRiscv CPU
```
commands tried
```
>>> import umachine
>>> umachine.identifier()
b'LiteX 2022-09-17 13:05:45'
>>> 

```

r4d10n, 2022-09-17:
Thanks @suarezvictor for the prompt guidance... 

Setting CPU_FLAGS in Makefile can be considered as a temporary hack as referenced in the above commit. 

I am now able to run Micropython on  [Colorlight i5 board](https://github.com/wuxx/Colorlight-FPGA-Projects), by building Litex VexRiscV with the following parameters:

```
litex/litex-boards/litex_boards/targets$ ./colorlight_i5.py --timer-uptime --uart-baudrate=1000000 --integrated-sram-size 32768 --cpu-type=vexriscv --cpu-variant=full --build

litex/litex-boards/litex_boards/targets$ ecpdap program --freq 5000 build/colorlight_i5/gateware/colorlight_i5.bit`

litex-micropython/ports/litex$ make clean && export BUILD_DIRECTORY=$HOME/litex/litex-boards/litex_boards/targets/build/colorlight_i5 && make V=1 -j`nproc`

litex-micropython/ports/litex$ litex_term /dev/ttyACM0 --kernel=build/firmware.bin --speed 1000000
```
```
        __   _ __      _  __
       / /  (_) /____ | |/_/
      / /__/ / __/ -_)>  <
     /____/_/\__/\__/_/|_|
   Build your hardware, easily!

 (c) Copyright 2012-2022 Enjoy-Digital
 (c) Copyright 2007-2015 M-Labs

 BIOS built on Sep 17 2022 21:39:10
 BIOS CRC passed (aed0c222)

 LiteX git sha1: 89bb6885

--=============== SoC ==================--
CPU:		VexRiscv_Full @ 60MHz
BUS:		WISHBONE 32-bit @ 4GiB
CSR:		32-bit data
ROM:		128KiB
SRAM:		32KiB
L2:		8KiB
FLASH:		2048KiB
SDRAM:		8192KiB 32-bit @ 60MT/s (CL-2 CWL-2)

--========== Initialization ============--
Initializing SDRAM @0x40000000...
Switching SDRAM to software control.
Switching SDRAM to hardware control.
Memtest at 0x40000000 (2.0MiB)...
  Write: 0x40000000-0x40200000 2.0MiB     
   Read: 0x40000000-0x40200000 2.0MiB     
Memtest OK
Memspeed at 0x40000000 (Sequential, 2.0MiB)...
  Write speed: 22.2MiB/s
   Read speed: 30.2MiB/s

Initializing GD25Q16 SPI Flash @0x00200000...
Enabling Quad mode...
SPI Flash clk configured to 30 MHz
Memspeed at 0x200000 (Sequential, 4.0KiB)...
   Read speed: 3.2MiB/s
Memspeed at 0x200000 (Random, 4.0KiB)...
   Read speed: 1.6MiB/s

--============== Boot ==================--
Booting from serial...
Press Q or ESC to abort boot completely.
sL5DdSMmkekro
[LITEX-TERM] Received firmware download request from the device.
[LITEX-TERM] Uploading build/firmware.bin to 0x40000000 (278592 bytes)...
[LITEX-TERM] Upload calibration... (inter-frame: 10.00us, length: 64)
[LITEX-TERM] Upload complete (46.5KB/s).
[LITEX-TERM] Booting the device.
[LITEX-TERM] Done.
Executing booted program at 0x40000000

--============= Liftoff! ===============--
MicroPython v1.16-60-g76bb2204c-dirty on 2022-09-17; LiteX SoC with VexRiscv CPU
>>> import 
builtins        micropython     cmath           gc
litex           math            uarray          ucollections
uio             umachine        uos             ustruct
usys            utime
>>> import uos
>>> uos.uname()
(sysname='LiteX (WISHBONE bus)', nodename='LiteX (WISHBONE bus)', release='1.16.0', version='v1.16-60-g76bb2204c-dirty on 2022-09-17', machine='LiteX SoC with VexRiscv CPU')
>>> import math
>>> math.pi
3.141592653589793
>>> math.sqrt(2)
1.414213562373095
>>> a=10
>>> b=5
>>> a+b
15
>>> 
```

Cheers!

closed 2022-09-17 by r4d10n

suarezvictor, 2022-09-17:
Glad to know it works in the colorlight i5 board!
I posted comments in regards to this fix in your PR

Eumeryx, 2023-07-04:
For people like me looking for a solution to the problem, an easy solution is to edit `micropython/ports/litex/Makefile` :

```
$(Q)$(CC) $(LDFLAGS) -o $@ $^ $(LIBS)
```

to be replaced with

```mk
$(Q)$(CC) $(LDFLAGS) -march=rv32i -o $@ $^ $(LIBS)
```

The reason is that `newlib` in my toolchain only supports the following targets:

```bash
$ ls /usr/lib/gcc/riscv64-elf/12.2.0/rv*
rv32i/      rv32im/     rv32imafc/  rv64imafdc/ 
rv32iac/    rv32imac/   rv64imac/
```
but as suarezvictor says it is set to `-march=rv32i2p0`.

Also, if you need LTO, replace it with:

```
$(Q)$(CC) $(LDFLAGS) -march=rv32i -misa-spec=2.2 -o $@ $^ $(LIBS)
```

suarezvictor, 2023-07-04:
Actually the current version of the project is this one:
https://github.com/suarezvictor/micropython

suarezvictor, 2023-07-04:
This branch, specifically
https://github.com/suarezvictor/micropython/tree/litex-rebase