enjoy-digital/litex #1428: litex and litepcie: modules not found.

anandsime, 2022-09-14:
I have been trying to use litepcie for building some of my SoCs. I have installed python, icarus verilog (instead of verilator) and have also followed all the steps given in the installation guide. But I am unable to execute the migen code which is in the core or phy or tlp or etc....... Am getting an error as attached below. Please help me resolve it.
[image]
[image]

cklarhorst, 2022-09-27:
Hi,
so, the lower image looks like you had problems to install litex (because it couldn't even find the main litex python module).

I think the most important step is `./litex_setup.py --init --install --user (--user to install to user directory) --config=(minimal, standard, full)` from the quick start guide. After that, the litex python module should be installed.

Maybe you can run that again (without `--init` I think because last time it should already have downloaded all gits) and look for errors there.

enjoy-digital, 2024-01-12:
Closing since answered and no feedback. Thanks  @cklarhorst!

closed 2024-01-12 by enjoy-digital