enjoy-digital/litex #1431: Litex_sim naxriscv with sdram broken

cklarhorst, 2022-09-16:
@enjoy-digital 
The last commit (d36f98bf) broke `litex_sim --cpu-type naxriscv --with-sdram --sdram-module=MT41K128M16 --sdram-data-width=16`.

```
Error Message:
  File "X/bin/litex_sim", line 33, in <module>
    sys.exit(load_entry_point('litex', 'console_scripts', 'litex_sim')())
  File "X/litex/litex/tools/litex_sim.py", line 464, in main
    soc = SimSoC(
  File "X/litex/litex/tools/litex_sim.py", line 194, in __init__
    self.add_sdram("sdram",
  File "X/litex/litex/soc/integration/soc.py", line 1525, in add_sdram
    self.submodules += axi.AXIUpConverter(
  File "X/litex/litex/soc/interconnect/axi/axi_full.py", line 196, in __init__
    self.comb += axi_from.w.connect(w_converter.sink, omit={"id"})
  File "X/migen/migen/genlib/record.py", line 165, in connect
    r += self_e.connect(getattr(slave, field), keep=keep, omit=omit)
  File "X/migen/migen/genlib/record.py", line 158, in connect
    r += [getattr(slave, field).eq(self_e) for slave in slaves]
  File "X/migen/migen/genlib/record.py", line 158, in <listcomp>
    r += [getattr(slave, field).eq(self_e) for slave in slaves]
AttributeError: 'Record' object has no attribute 'dest'
```

enjoy-digital, 2022-09-16:
Thanks @cklarhorst for reporting, this should be fixed with https://github.com/enjoy-digital/litex/commit/c24bbedb68ee72d81159ff064c3a64ffe596429a.

cklarhorst, 2022-09-16:
yep, Thanks.

closed 2022-09-16 by cklarhorst