enjoy-digital/litex #1439: Digital video: blue and red signals seem swapped

suarezvictor, 2022-09-23:
Digital video output have swapped R and B signals, according to https://en.wikipedia.org/wiki/Digital_Visual_Interface, and to my tests using analog VGA (which has clearer channel assignment and produces the expected colors).
Note also that sync signals should be sent on DATA0 (blue clannel) and not "r" channel as in the current code (change to be tested).

See: https://github.com/enjoy-digital/litex/blob/master/litex/soc/cores/video.py#L802
`c2d  = {"r": 0, "g": 1, "b": 2}` should be changed to
`c2d  = {"r": 2, "g": 1, "b": 0}` (same for all other lines where that code appears)

and `self.comb += encoder.c.eq(Cat(sink.hsync, sink.vsync) if color == "r" else 0)`
to `self.comb += encoder.c.eq(Cat(sink.hsync, sink.vsync) if color == "b" else 0)`

trabucayre, 2022-09-23:
I had noticed this swap too (I don't know why I haven't sent a PR :-/).
Most all the phy must be updated accordingly (and all targets using pn_swap too).

enjoy-digital, 2022-09-26:
Thanks @suarezvictor, I'll try to integrate your fix in the next days and do the swap in the platforms.

suarezvictor, 2022-09-26:
Please consider that chaging the sync signal channel may break things, so test are needed. I'll report an issue with that when I can dig more and report a easier reproducible bug

trabucayre, 2022-09-28:
PR for both litex and litex-boards sent!

enjoy-digital, 2022-09-29:
Thanks @suarezvictor, @trabucayre. I remember having to swap "r" and "b" on some design in the past and now understand why :)

closed 2022-09-29 by enjoy-digital