enjoy-digital/litex #1443: litex demo cannot compile

YiminGao0113, 2022-09-28:
When I try to compile the demo project, it gives me : fatal error: picolibc.h: No such file or directory.
But everything I have is up to date, including the picolibc directory. Please let me know if you know how to solve the issue. Thanks!

cklarhorst, 2022-09-28:
@YiminGao0113 
Could you please post all steps leading to this error?
So you build a specific target (which one?) and then run the demo.py with `--build-path` pointing to that build dir?
Could you post the output of that demo.py run.

YiminGao0113, 2022-09-28:
Yes, I built the target for xilinx_vc707, and run demo.py  with --build-path pointing to xilinx_vc707 build dir, here is the output I get:

[image]

cklarhorst, 2022-09-28:
So the `picolibc.h` is build when you generate the target.
When I run `python3 -m litex_boards.targets.xilinx_vc707 --build` it generates the file. You should find it in `build/xilinx_vc707/software/libc/picolibc.h`. So please recheck for any errors in the output of the xilinx_vc707 target script if that file is missing on your end.
Or please provide the output of that script or any parameters you used.

YiminGao0113, 2022-09-29:
Thanks! I can compile the demo app now. I think now the question is, does demo app support serv cpu? I can see things after liftoff with vexriscv but cannot see anything when I build with --cpu-type serv. (Because I am adding support to another core, referring to the serv wrapper)

YiminGao0113, 2022-09-29:
Thanks, I found out

closed 2022-09-29 by YiminGao0113