enjoy-digital/litex #1445: litex demo not working while adding support to a new core, ask for suggestions

YiminGao0113, 2022-09-29:
Hi, as I am adding support for a new CPU, I get to a point where I can see LiteX bios running on hardware from serial port, but I cannot run litex demo app, nothing pops up after Liftoff. I did some mem_test at 0x40000000 so I believe the memory is working properly.
I am kind of stuck, please let me know if you have any suggestions on how I can debug now. Thanks!
[image]
[image]

cklarhorst, 2022-09-29:
Maybe test your CPU with the demo in litex_sim. Then you can also see all singal traces.
Some args you may need are (mostly the same as if you use your current target):
- --with-sdram
- --with-sdram-init [your demo bin]
- --cpu-type [your cpu]
- --trace

YiminGao0113, 2022-10-04:
I cannot see the output of demo app in litex_sim now. First I did: litex_bare_metal_demo --build-path=build/sim
Then: litex_sim --with_sdram --sdram_init=demo.bin --cpu-type=vexriscv
But I cannot see anything after liftoff in litex_sim, did I miss anything?

cklarhorst, 2022-10-04:
Did you use `--with-sdram` in the first litex_sim run? I think that is required. So the idea is that the first litex_sim run needs to have the same args as the final one except the `init` arg, so that the final hardware layout like CSRs are mapped to the same addresses.
I tested:
```
[clean build dir]
litex_sim --with-sdram --cpu-type=vexriscv
litex_bare_metal_demo --build-path build/sim/
litex_sim --with-sdram --sdram-init=demo.bin  --cpu-type=vexriscv
```
And got a liftoff (but I'm not on the latest master, I'm at 8ffdc53)

YiminGao0113, 2022-10-05:
Thanks! 
I can debug in gtkwave now, but still couldn't figure out why the demo app doesn't work for my processor.
For the software part, what I did is I reuse the cpu serv's wrapper (copy the folder soc/cores/cpu/serv), I change the cpu name and the logic inside the wrapper to map to my cpu. So I didn't really do much on the software part. Anyway, the bios works fine for my processor while the demo app doesn't work. 
Do you might have any suggestions on which part I should focus on to make demo app working? Any help would be appreciated! :)

tcal-x, 2022-10-05:
Typically I would look at the PC (program counter) to see at which instruction program execution stalls (either it freezes at a single instruction, or is caught in a loop, possibly involving a trap).   How does your core handle illegal instruction traps?