enjoy-digital/litex #1451: Compilation for target-board efinix_titanium_ti60_f225_dev_kit fails

djhardrich, 2022-10-02:
When using `python3 /home/user/litex/litex-boards/litex_boards/targets/efinix_titanium_ti60_f225_dev_kit.py --cpu-type=vexriscv_smp --with-hyperram --with-sdcard --build` , compilation fails with Interface Designer errors, log attached:

[litex_ti60_log.txt](https://github.com/enjoy-digital/litex/files/9693374/litex_ti60_log.txt)

Dolu1990, 2022-10-02:
I just tested, but can't reproduce it with efinix Version: 2021.2.323.4.6

Can you zip the efinix project ?

djhardrich, 2022-10-02:
Thanks for the reply, project files attached! Currently just trying to bring up the example and build linux-on-litex; am using Efinity 2022.1.226.1.9 which is supposed to have some rather large improvements to routing. Will try to build using 2021.2.323.4.6 and report back as well!
[efinix_titanium_ti60_f225_dev_kit.zip](https://github.com/enjoy-digital/litex/files/9693617/efinix_titanium_ti60_f225_dev_kit.zip)

Dolu1990, 2022-10-03:
Ahh opening it with the efinity tool, i get some warnings : 
[image]
Could it be that those warning are considered as error in 2022.1.226.1.9 ?

enjoy-digital, 2022-10-12:
Thanks @djhardrich for the feedback and @Dolu1990 for looking at it. I'll update the Efinity version on my machine and will try to reproduce/fix.

enjoy-digital, 2022-10-14:
The build has been tested sucessfully with:
- https://github.com/enjoy-digital/litex/commit/24cfbd8d67b29f3b5425d2198f2002e9901e6cec
- https://github.com/litex-hub/litex-boards/commit/23c6acd013f1987c1c7e1dfe5e028b5865088b40

Note that I had to do a local change to pt/bin/writer/pinout.py to get the build working, line 2254:
```
- table.add_rows(table_rows)
+ for table_row in table_rows:
+   table.add_row(table_row)
```

This would be interesting to see if this is related to my setup (ie you don't have errors without this) or if a real issue.

closed 2022-10-14 by enjoy-digital