enjoy-digital/litex #1457: Tag 2022.08 points to missing commit in pythondata_software_picolibc

tari, 2022-10-10:
If I run `litex_setup.py` against tag 2022.08, it fails with the following output:

```
[   1.431] Cloning pythondata-software-picolibc Git repository...
Cloning into 'pythondata-software-picolibc'...
remote: Enumerating objects: 612, done.
remote: Counting objects: 100% (144/144), done.
remote: Compressing objects: 100% (105/105), done.
remote: Total 612 (delta 68), reused 113 (delta 39), pack-reused 468
Receiving objects: 100% (612/612), 111.36 KiB | 5.30 MiB/s, done.
Resolving deltas: 100% (277/277), done.
Submodule 'pythondata_software_picolibc/data' (https://github.com/picolibc/picolibc) registered for path 'pythondata_software_picolibc/data'
Cloning into '/home/tari/litex/pythondata-software-picolibc/pythondata_software_picolibc/data'...
remote: Enumerating objects: 221770, done.
remote: Counting objects: 100% (2256/2256), done.
remote: Compressing objects: 100% (1004/1004), done.
remote: Total 221770 (delta 1384), reused 1923 (delta 1233), pack-reused 219514
Receiving objects: 100% (221770/221770), 134.43 MiB | 12.60 MiB/s, done.
Resolving deltas: 100% (176668/176668), done.
fatal: remote error: upload-pack: not our ref 580d4f13de47a3eaf66f10cfc8537721b21dfc31
fatal: Fetched in submodule path 'pythondata_software_picolibc/data', but it did not contain 580d4f13de47a3eaf66f10cfc8537721b21dfc31. Direct fetching of that commit failed.
Traceback (most recent call last):
  File "/home/tari/litex/./litex_setup.py", line 448, in <module>
    main()
  File "/home/tari/litex/./litex_setup.py", line 422, in main
    litex_setup_init_repos(config=args.config, tag=args.tag, dev_mode=args.dev)
  File "/home/tari/litex/./litex_setup.py", line 213, in litex_setup_init_repos
    subprocess.check_call("git clone {url} {options}".format(
  File "/usr/lib/python3.10/subprocess.py", line 369, in check_call
    raise CalledProcessError(retcode, cmd)
subprocess.CalledProcessError: Command 'git clone https://github.com/litex-hub/pythondata-software-picolibc.git --recursive' returned non-zero exit status 128.
```

It looks like the commit referred to by the submodule no longer (or never?) existed in the repository on github. Tag 2022.04 works as expected, so it doesn't look like a problem with my system.

staticfloat, 2022-10-10:
I believe there may be a gitref that was dropped.

enjoy-digital, 2022-11-02:
We can probably close since fixed with https://github.com/litex-hub/pythondata-auto/issues/36.

closed 2022-11-02 by enjoy-digital