enjoy-digital/litex #1462: CVA6 Simulation / Build error

adamhlt, 2022-10-14:
Hello,
I'm trying to work with the CVA6 cpu and I get some errors when trying to do a simulation or a build.

- Board : Digilent Basys 3
- Vivado 2019.1
- Verilator 4.110 (tested also with 4.028 and 4.228)
- Toolchain : riscv64-unknown-elf-gcc-10.1.0-2020.08.2-x86_64-linux-ubuntu14 (provided by LiteX)
- Ubuntu 20.04

Simulation :

>./digilent_basys3.py --cpu-type=cva6 

```
INFO:S7MMCM:Creating S7MMCM, speedgrade -1.
INFO:S7MMCM:Registering Single Ended ClkIn of 100.00MHz.
INFO:S7MMCM:Creating ClkOut0 sys of 75.00MHz (+-10000.00ppm).
INFO:S7MMCM:Creating ClkOut1 vga of 40.00MHz (+-10000.00ppm).
INFO:SoC:        __   _ __      _  __  
INFO:SoC:       / /  (_) /____ | |/_/  
INFO:SoC:      / /__/ / __/ -_)>  <    
INFO:SoC:     /____/_/\__/\__/_/|_|  
INFO:SoC:  Build your hardware, easily!
INFO:SoC:--------------------------------------------------------------------------------
INFO:SoC:Creating SoC... (2022-10-14 15:56:31)
INFO:SoC:--------------------------------------------------------------------------------
INFO:SoC:FPGA device : xc7a35t-CPG236-1.
INFO:SoC:System clock: 75.000MHz.
INFO:SoCBusHandler:Creating Bus Handler...
INFO:SoCBusHandler:32-bit wishbone Bus, 4.0GiB Address Space.
INFO:SoCBusHandler:Adding reserved Bus Regions...
INFO:SoCBusHandler:Bus Handler created.
INFO:SoCCSRHandler:Creating CSR Handler...
INFO:SoCCSRHandler:32-bit CSR Bus, 32-bit Aligned, 16.0KiB Address Space, 2048B Paging, big Ordering (Up to 32 Locations).
INFO:SoCCSRHandler:Adding reserved CSRs...
INFO:SoCCSRHandler:CSR Handler created.
INFO:SoCIRQHandler:Creating IRQ Handler...
INFO:SoCIRQHandler:IRQ Handler (up to 32 Locations).
INFO:SoCIRQHandler:Adding reserved IRQs...
INFO:SoCIRQHandler:IRQ Handler created.
INFO:SoC:--------------------------------------------------------------------------------
INFO:SoC:Initial SoC:
INFO:SoC:--------------------------------------------------------------------------------
INFO:SoC:32-bit wishbone Bus, 4.0GiB Address Space.
INFO:SoC:32-bit CSR Bus, 32-bit Aligned, 16.0KiB Address Space, 2048B Paging, big Ordering (Up to 32 Locations).
INFO:SoC:IRQ Handler (up to 32 Locations).
INFO:SoC:--------------------------------------------------------------------------------
INFO:SoC:Controller ctrl added.
INFO:SoC:CPU cva6 added.
INFO:SoC:CPU cva6 adding IO Region 0 at 0x80000000 (Size: 0x80000000).
INFO:SoCBusHandler:io0 Region added at Origin: 0x80000000, Size: 0x80000000, Mode: RW, Cached: False Linker: False.
INFO:SoC:CPU cva6 overriding rom mapping from 0x00000000 to 0x10000000.
INFO:SoC:CPU cva6 overriding sram mapping from 0x01000000 to 0x20000000.
INFO:SoC:CPU cva6 setting reset address to 0x10000000.
INFO:SoC:CPU cva6 adding Bus Master(s).
INFO:SoCBusHandler:cpu_bus0 Bus adapted from Wishbone 64-bit to Wishbone 32-bit.
INFO:SoCBusHandler:cpu_bus0 added as Bus Master.
INFO:SoC:CPU cva6 adding Interrupt(s).
INFO:SoCBusHandler:rom Region added at Origin: 0x10000000, Size: 0x00020000, Mode: RX, Cached: True Linker: False.
INFO:SoCBusHandler:rom added as Bus Slave.
INFO:SoC:RAM rom added Origin: 0x10000000, Size: 0x00020000, Mode: RX, Cached: True Linker: False.
INFO:SoCBusHandler:sram Region added at Origin: 0x20000000, Size: 0x00002000, Mode: RWX, Cached: True Linker: False.
INFO:SoCBusHandler:sram added as Bus Slave.
INFO:SoC:RAM sram added Origin: 0x20000000, Size: 0x00002000, Mode: RWX, Cached: True Linker: False.
INFO:SoCIRQHandler:uart IRQ allocated at Location 0.
INFO:SoCIRQHandler:timer0 IRQ allocated at Location 1.
```

> litex_sim --cpu-type=cva6

```
  186 |   typedef fmt_unit_types_t [0:NUM_OPGROUPS-1] opgrp_fmt_unit_types_t;
      |                            ^
%Warning-UNPACKED: /home/adam/LiteX/pythondata-cpu-cva6/pythondata_cpu_cva6/system_verilog/core/include/riscv_pkg.sv:711:13: Unsupported: Unpacked struct/union
                                                                                                                           : ... In instance sim
  711 |     typedef struct {
      |             ^~~~~~
%Error: /home/adam/LiteX/litex-boards/litex_boards/targets/build/sim/gateware/sim.v:3113:1: Cannot find file containing module: 'cva6_wrapper'
 3113 | cva6_wrapper cva6_wrapper(
      | ^~~~~~~~~~~~
        ... Looked in:
             /home/adam/LiteX/pythondata-cpu-cva6/pythondata_cpu_cva6/system_verilog/common/submodules/common_cells/include/cva6_wrapper
             /home/adam/LiteX/pythondata-cpu-cva6/pythondata_cpu_cva6/system_verilog/common/submodules/common_cells/include/cva6_wrapper.v
             /home/adam/LiteX/pythondata-cpu-cva6/pythondata_cpu_cva6/system_verilog/common/submodules/common_cells/include/cva6_wrapper.sv
             /home/adam/LiteX/pythondata-cpu-cva6/pythondata_cpu_cva6/system_verilog/common/submodules/common_cells/src/cva6_wrapper
             /home/adam/LiteX/pythondata-cpu-cva6/pythondata_cpu_cva6/system_verilog/common/submodules/common_cells/src/cva6_wrapper.v
             /home/adam/LiteX/pythondata-cpu-cva6/pythondata_cpu_cva6/system_verilog/common/submodules/common_cells/src/cva6_wrapper.sv
             /home/adam/LiteX/pythondata-cpu-cva6/pythondata_cpu_cva6/system_verilog/common/local/util/cva6_wrapper
             /home/adam/LiteX/pythondata-cpu-cva6/pythondata_cpu_cva6/system_verilog/common/local/util/cva6_wrapper.v
             /home/adam/LiteX/pythondata-cpu-cva6/pythondata_cpu_cva6/system_verilog/common/local/util/cva6_wrapper.sv
             cva6_wrapper
             cva6_wrapper.v
             cva6_wrapper.sv
             obj_dir/cva6_wrapper
             obj_dir/cva6_wrapper.v
             obj_dir/cva6_wrapper.sv
%Error: Exiting due to 1 error(s)
make: *** [/home/adam/LiteX/litex/litex/build/sim/core/Makefile:38: sim] Error 1
make: Leaving directory '/home/adam/LiteX/litex-boards/litex_boards/targets/build/sim/gateware'
```

I already tried to integrate the CVA6 wrapper (looking at this #1377) but with no result I get more errors.

Then for the build :

>./digilent_basys3.py --cpu-type=cva6 --build

```
INFO: [Common 17-349] Got license for feature 'Synthesis' and/or device 'xc7a35t'
INFO: [Device 21-403] Loading part xc7a35tcpg236-1
INFO: Launching helper process for spawning children vivado processes
INFO: Helper process launched with PID 34165 
ERROR: [Synth 8-1766] cannot open include file common_cells/registers.svh [/home/adam/LiteX/pythondata-cpu-cva6/pythondata_cpu_cva6/system_verilog/core/fpu/src/fpnew_cast_multi.sv:14]
```

It seems that the `register.svh` don't have the right path. 
I tried to "patch" the error by putting `register.svh` at the right path but I got another error returned by vivado.

```
ERROR: [Synth 8-1031] ariane_pkg is not declared [/home/adam/LiteX/pythondata-cpu-cva6/pythondata_cpu_cva6/system_verilog/core/mmu_sv32/cva6_mmu_sv32.sv:30]
ERROR: [Synth 8-1031] ariane_pkg is not declared [/home/adam/LiteX/pythondata-cpu-cva6/pythondata_cpu_cva6/system_verilog/core/mmu_sv32/cva6_mmu_sv32.sv:34]
ERROR: [Synth 8-1031] ariane_pkg is not declared [/home/adam/LiteX/pythondata-cpu-cva6/pythondata_cpu_cva6/system_verilog/core/mmu_sv32/cva6_mmu_sv32.sv:34]
ERROR: [Synth 8-1031] icache_areq_o_t is not declared [/home/adam/LiteX/pythondata-cpu-cva6/pythondata_cpu_cva6/system_verilog/core/mmu_sv32/cva6_mmu_sv32.sv:42]
ERROR: [Synth 8-1031] icache_areq_i_t is not declared [/home/adam/LiteX/pythondata-cpu-cva6/pythondata_cpu_cva6/system_verilog/core/mmu_sv32/cva6_mmu_sv32.sv:43]
ERROR: [Synth 8-1031] exception_t is not declared [/home/adam/LiteX/pythondata-cpu-cva6/pythondata_cpu_cva6/system_verilog/core/mmu_sv32/cva6_mmu_sv32.sv:47]
```

Now, the ariane_pkg can't be found.
I checked in `Flist.cv32a6_imac_sv0` and `Flist.cv64a6_imafdc_sv39`, the file is listed in both files and the file is present in the project.

I would like to know if the CVA6 cpu is currently working with LiteX and if you have some advice on how should I figure out these errors.

enjoy-digital, 2022-10-14:
Hello @adamhlt,

I just verified and  litex_sim  --cpu-type=cva6 is working on my machine.

- Verilator: 4.219 devel rev v4.218-43-ga9cb9bac
- pythondata-cpu-cva6 sha1: 0x8d27ff8

Can you tried this commit for pythondata-cpu-cva6?

adamhlt, 2022-10-14:
It seems to be better, but it is not working.
[Simulation Demo](https://streamable.com/7lqkdh)
The simulation use all my resources and I have not enough ram to compute the simulation, I don't know if it is a normal behavior.

Then for the build I got the same error with `register.svh` and when I try to patch it, I got another error of missing files.

```
ERROR: [Synth 8-1766] cannot open include file common_cells/registers.svh [/home/adam/LiteX/pythondata-cpu-cva6/pythondata_cpu_cva6/system_verilog/core/fpu/src/fpnew_cast_multi.sv:14]
ERROR: [Synth 8-2715] syntax error near ( [/home/adam/LiteX/pythondata-cpu-cva6/pythondata_cpu_cva6/system_verilog/core/fpu/src/fpnew_cast_multi.sv:145]
ERROR: [Synth 8-2715] syntax error near , [/home/adam/LiteX/pythondata-cpu-cva6/pythondata_cpu_cva6/system_verilog/core/fpu/src/fpnew_cast_multi.sv:145]
```

adamhlt, 2022-10-15:
When I patched the missing `register.svh`, I got this error.

```
INFO: [Common 17-349] Got license for feature 'Synthesis' and/or device 'xc7a35t'
INFO: [Device 21-403] Loading part xc7a35tcpg236-1
INFO: Launching helper process for spawning children vivado processes
INFO: Helper process launched with PID 14848 
ERROR: [Synth 8-1766] cannot open include file axi/assign.svh [/home/adam/LiteX/pythondata-cpu-cva6/pythondata_cpu_cva6/system_verilog/corev_apu/axi/src/axi_atop_filter.sv:371]
ERROR: [Synth 8-1766] cannot open include file axi/typedef.svh [/home/adam/LiteX/pythondata-cpu-cva6/pythondata_cpu_cva6/system_verilog/corev_apu/axi/src/axi_atop_filter.sv:372]
ERROR: [Synth 8-2715] syntax error near ( [/home/adam/LiteX/pythondata-cpu-cva6/pythondata_cpu_cva6/system_verilog/corev_apu/axi/src/axi_atop_filter.sv:403]
WARNING: [Synth 8-2841] use of undefined macro AXI_TYPEDEF_AW_CHAN_T [/home/adam/LiteX/pythondata-cpu-cva6/pythondata_cpu_cva6/system_verilog/corev_apu/axi/src/axi_atop_filter.sv:403]
ERROR: [Synth 8-988] addr_t is already declared [/home/adam/LiteX/pythondata-cpu-cva6/pythondata_cpu_cva6/system_verilog/corev_apu/axi/src/axi_atop_filter.sv:403]
```

I checked the `Flist.cva6_wrapper` file and the includes are here and the files are not missing.
```
+incdir+${CVA6_REPO_DIR}/corev_apu/axi/include/
+incdir+${CVA6_REPO_DIR}/corev_apu/register_interface/include/
+incdir+${CVA6_REPO_DIR}/cva6_wrapper/
```

gatecat, 2023-01-17:
It seems like the problem is, the updated https://github.com/litex-hub/pythondata-cpu-cva6 is missing the wrapper that litex needs.

Using https://github.com/litex-hub/pythondata-cpu-cva6-old works fine, but I'm not sure how we should be fixing this and distribute the wrapper long term (as afaik the pythondata-* repos shouldn't really have custom changes to upstream in them?)

I have a project planned with an ASIC tapeout of a CVA6 CPU and eFPGA fabric that I am currently planning to use LiteX for the integration for. So I'd be happy to work on improving the CVA6 support given a rough idea of the direction wanted.

enjoy-digital, 2023-01-17:
Hi @gatecat, nice for the planned ASIC tapeout!

I share your opinion and also think the pythondata-* repos should not include modifications (unless maybe for some specific cases like VexRiscv/VexRiscv-SMP where there is a close collaboration between the projects and for which the repos don't really exist otherwise).

For CVA6, we should just have the modifications required for LiteX integrated in `cores/cpu/cva6`  and eventually use a fixed commit for the `pythondata-*` repo in case these modifications are tighly coupled to the CVA6 version and an update would probably break them.

So we could:
- Integrate the changes introduced in https://github.com/litex-hub/pythondata-cpu-cva6-old to `cores/cpu/cva6`.
- Adapt `core.py` of CVA9 to these changes to point to the correct files.
- Make sure this work with https://github.com/litex-hub/pythondata-cpu-cva6.

Thanks!

closed 2023-02-09 by adamhlt