enjoy-digital/litex #1468: Does LiteX support Wishbone pipelined mode?

T-Platz, 2022-10-24:
Hi everyone,

I'm currently working on an adapter for interfacing the BlackParrot processor to a LiteX Wishbone bus. Sadly, I didn't find much documentation about what modes and signals the Wishbone bus and connected LiteX peripherals support. In particular, I'm asking whether I can drive the Wishbone signals in pipelined mode, which requires a stall signal.

I've seen some cores mention the Wishbone stall signal like [VexRISC-V SMP](https://github.com/enjoy-digital/litex/blob/ded3bad1781b37af1eada47276b313f7f47ec0bd/litex/soc/cores/cpu/vexriscv_smp/core.py#L335) for instance, but I'm not sure if this really means that I can rely on the bus and LiteX peripherals supporting the Wishbone B4 pipelined mode.

Thanks for your help in advance!

mithro, 2022-10-24:
Take a look at https://antmicro.com/blog/2022/05/faster-soc-interconnects-with-test-driven-fpga-development-and-cocotb/

T-Platz, 2022-10-24:
@mithro Thanks for the link, although I believe the article is talking about burst transfers using registered feedback cycles described in chapter 4 of the [Wishbone B4 specification](https://cdn.opencores.org/downloads/wbspec_b4.pdf). What I am looking for is the "Pipelined wishbone protocol" defined in section 3.1.3.2 of the manual.