enjoy-digital/litex #1469: Demo app not enough size to run my benchmark

YiminGao0113, 2022-10-25:
I want to replace the litex demo app with my own C files to run a neural network benchmark. 
First I replace the demo app with some simple C programs (Hello world, sort) and it works pretty well. 

However, when I replace the demo app with my neural network C programs, it says that '.data' will not fit in region 'sram'

[image]

I thought the demo app goes into the main RAM which is DRAM for my FPGA board Xilinx_VC707. How can I fix this and run large program using litex bare metal? Thanks!

cklarhorst, 2022-10-26:
The linker file for the demo app is [here](https://github.com/enjoy-digital/litex/blob/master/litex/soc/software/demo/linker.ld).
I have not tested it, but I think you need to remove `> sram AT` in this line: https://github.com/enjoy-digital/litex/blob/611b84ccee41cbc527be6bd4e8658554fabd852e/litex/soc/software/demo/linker.ld#L43

The `>sram At` was added because "initialized global variables don't work with user libraries". I'm not sure what that means, maybe just be cautious and check if your program still behaves correctly.

YiminGao0113, 2022-10-26:
Thanks!
I tried to do that and run litex demo app, but after liftoff it goes back to the bios again instead of showing demo app..

closed 2022-11-08 by YiminGao0113

phillippZZ, 2024-07-30:
Hi, there.

I'm having similar trouble now. 
I don't know how to set up the link so that I can have big array to test read/write speed. Removing "> sram AT" makes the program stuck at liftoff.