enjoy-digital/litex #1471: Add support for POWER / microwatt in the CI

mithro, 2022-10-27:
With the renewed interest in microwatt running inside LiteX thanks to the LibreBMC project, it would be good to test POWER / microwatt in CI in the same way that RISC-V and OpenRISC1000 are tested.

See https://github.com/enjoy-digital/litex/blob/e570b612b2a9d8f8d2002d79497bda0dc35b936a/.github/workflows/ci.yml#L32-L40 and https://github.com/enjoy-digital/litex/blob/master/test/test_cpu.py#L59

mithro, 2022-10-27:
@shenki - Would you be interested in taking a look at this?

shenki, 2022-10-29:
Yes, I have started working on this.

I reworked the [vhd2v support to just use GHDL](https://github.com/enjoy-digital/litex/pull/1475) (instead of GJDL+Yosys) now that it can convert VHDL to Verilog on its own. This exposed [a bug](https://github.com/ghdl/ghdl/issues/2234) in GHDL that is now fixed. (Tristan added part of neorv32 to the ghdl test suite!)

When adding microwatt to the test_cpu testcase it passes locally, but times out when run on CI. I am investigating this.

shenki, 2022-11-14:
https://github.com/enjoy-digital/litex/pull/1503

shenki, 2022-11-15:
Merged in https://github.com/enjoy-digital/litex/commit/b90080ab7472ab24de68c00428f98e6ec6357d9b

shenki, 2022-11-17:
@enjoy-digital this can be closed now

closed 2022-11-17 by enjoy-digital