enjoy-digital/litex #1477: Regression on JTAG UART with removal of jtag_atlantic for Altera boards

tomaz-suller, 2022-10-30:
The removal of `jtag_atlantic` seems to have been premature, given neither `nios2-terminal` not `litex_term` are able to connect to the JTAG UART synthesised by passing `--uart-name jtag_uart` when building a target. A simple solution would be to bring `jtag_atlantic` back (which is fairly straightforward) and assess why `jtag_uart` isn't working as expected.

# Description

I'm getting started on LiteX on my [DE10-Nano](https://www.terasic.com.tw/cgi-bin/page/archive.pl?Language=English&CategoryNo=167&No=1046) SoC from Intel/Altera, in which the board UART port can't be accessed directly from the FPGA fabric.  The board has already been ported to LiteX, as per https://github.com/litex-hub/litex-boards/issues/23, and the author of the original issue is able to connect to `litex_term` using JTAG UART through `jtag_atlantic`, which should have been superseded by `jtag_uart` according to 1c5d91dce18cf128fec3d5fd85ed8d6608e4afaa. With that, building my target using `--uart-name jtag_uart` should produce gateware that allows the user to connect to the UART through the boards USB Blaster.

However, that is not the case, and I haven't been able to make JTAG UART work on the master branch of LiteX at all.

# Steps to reproduce

I only have access to the DE10-Nano, so I'm using it in this example, and I assume Quartus is already installed in the system.

1. Install LiteX
2. Build the DE10-Nano target with from `litex-boards` (which should come with a normal LiteX install) with the `--uart-name jtag_uart` option
```bash
python -m litex_boards.targets.terasic_de10nano --uart-name jtag_uart --build
```
3. Load the gateware into the DE10-Nano
```bash
python -m litex_boards.targets.terasic_de10nano --load
```
4. Access the serial terminal.
```bash
nios2-terminal
```

# Results

## Expected
LiteX boot console appears on the screen.

## Actual
`nios2-terminal` error message:
```
nios2-terminal: There are no JTAG UARTs available which match the --device and
nios2-terminal: --instance options you provided.
```

# Workaround

I manually copied some snippets from 6f6a10db5c070a6547e5c8323b5a8887d2ec0c2d (i.e. the parent of the commit which removed `jtag_atlantic`) into some LiteX files to reintroduce `jtag_atlantic`
| File path | Modifications |
|-----------|----------|
| soc/integration/soc.py | Reintroduce `jtag_atlantic` to `supported_uarts` and the Atlantic if statement |
| tools/litex_term.py | Reintroduce `jtag_atlantic` to `jtag_name` if statement |
| soc/cores/jtag.py | C-c C-v the entire `JTAGAtlantic` class | 

Building with `--uart-name jtag_atlantic` and calling `litex_term` with `--jtag-name jtag_atlantic` both become possible again with these modifications, and the synthesised gateware is recognised by `nios2-terminal`. Running it again then outputs the expected LiteX boot console.

I'm willing to reintroduce `jtag_atlantic` on a fork and then submit a PR, though I'm not sure if I could fix `jtag_uart` without help.

# Environment

Arch Linux x86_64 (6.0.2-arch1-1)
Python 3.10.8

<details>
<summary>litex_setup.py --freeze</summary>

```
          __   _ __      _  __         
         / /  (_) /____ | |/_/         
        / /__/ / __/ -_)>  <           
       /____/_/\__/\__/_/|_|         
     Build your hardware, easily!      
          LiteX Setup utility.         

[   0.001] LiteX Setup auto-update...
[   0.696] LiteX Setup is up to date.
[   0.697] Freezing config of Git repositories...
[   0.697] --------------------------------------
git_repos = {
    "migen" : GitRepo(url="https://github.com/m-labs/",
        clone   = "recursive",
        develop = True,
        sha1    = 0x639e66f,
        branch  = "master"
    ),
    "pythondata-software-picolibc" : GitRepo(url="https://github.com/litex-hub/",
        clone   = "recursive",
        develop = True,
        sha1    = 0x78a5e73,
        branch  = "master"
    ),
    "pythondata-software-compiler_rt" : GitRepo(url="https://github.com/litex-hub/",
        clone   = "regular",
        develop = True,
        sha1    = 0xfcb0324,
        branch  = "master"
    ),
    "litex" : GitRepo(url="git@github.com:enjoy-digital/",
        clone   = "regular",
        develop = True,
        sha1    = 0xe3c3319,
        branch  = "master"
    ),
    "liteeth" : GitRepo(url="https://github.com/enjoy-digital/",
        clone   = "regular",
        develop = True,
        sha1    = 0x022207c,
        branch  = "master"
    ),
    "litedram" : GitRepo(url="https://github.com/enjoy-digital/",
        clone   = "regular",
        develop = True,
        sha1    = 0xc770dd6,
        branch  = "master"
    ),
    "litepcie" : GitRepo(url="https://github.com/enjoy-digital/",
        clone   = "regular",
        develop = True,
        sha1    = 0xbdd335a,
        branch  = "master"
    ),
    "litesata" : GitRepo(url="https://github.com/enjoy-digital/",
        clone   = "regular",
        develop = True,
        sha1    = 0x6b6e45b,
        branch  = "master"
    ),
    "litesdcard" : GitRepo(url="https://github.com/enjoy-digital/",
        clone   = "regular",
        develop = True,
        sha1    = 0x15955d8,
        branch  = "master"
    ),
    "liteiclink" : GitRepo(url="https://github.com/enjoy-digital/",
        clone   = "regular",
        develop = True,
        sha1    = 0xecdfe9f,
        branch  = "master"
    ),
    "litescope" : GitRepo(url="https://github.com/enjoy-digital/",
        clone   = "regular",
        develop = True,
        sha1    = 0x7105b47,
        branch  = "master"
    ),
    "litejesd204b" : GitRepo(url="https://github.com/enjoy-digital/",
        clone   = "regular",
        develop = True,
        sha1    = 0xb18880f,
        branch  = "master"
    ),
    "litespi" : GitRepo(url="https://github.com/litex-hub/",
        clone   = "regular",
        develop = True,
        sha1    = 0x69d1331,
        branch  = "master"
    ),
    "valentyusb" : GitRepo(url="https://github.com/litex-hub/",
        clone   = "regular",
        develop = True,
        sha1    = 0x0b534cc,
        branch  = "hw_cdc_eptri"
    ),
    "litex-boards" : GitRepo(url="https://github.com/litex-hub/",
        clone   = "regular",
        develop = True,
        sha1    = 0x4dc8f72,
        branch  = "master"
    ),
    "pythondata-misc-tapcfg" : GitRepo(url="https://github.com/litex-hub/",
        clone   = "regular",
        develop = True,
        sha1    = 0xfbcb024,
        branch  = "master"
    ),
    "pythondata-misc-usb_ohci" : GitRepo(url="https://github.com/litex-hub/",
        clone   = "regular",
        develop = True,
        sha1    = 0x1f9c773,
        branch  = "master"
    ),
    "pythondata-cpu-lm32" : GitRepo(url="https://github.com/litex-hub/",
        clone   = "regular",
        develop = True,
        sha1    = 0x594f206,
        branch  = "master"
    ),
    "pythondata-cpu-mor1kx" : GitRepo(url="https://github.com/litex-hub/",
        clone   = "regular",
        develop = True,
        sha1    = 0x92628ec,
        branch  = "master"
    ),
    "pythondata-cpu-naxriscv" : GitRepo(url="https://github.com/litex-hub/",
        clone   = "regular",
        develop = True,
        sha1    = 0x1124b40,
        branch  = "master"
    ),
    "pythondata-cpu-serv" : GitRepo(url="https://github.com/litex-hub/",
        clone   = "regular",
        develop = True,
        sha1    = 0xcdede3b,
        branch  = "master"
    ),
    "pythondata-cpu-vexriscv" : GitRepo(url="https://github.com/litex-hub/",
        clone   = "regular",
        develop = True,
        sha1    = 0xe75700d,
        branch  = "master"
    ),
    "pythondata-cpu-vexriscv-smp" : GitRepo(url="https://github.com/litex-hub/",
        clone   = "recursive",
        develop = True,
        sha1    = 0xb360267,
        branch  = "master"
    ),
}
```

</details>

enjoy-digital, 2022-10-30:
Hi @tomaz-suller,

jtag_atlantic/nios2-terminal support has been removed in favor of a common JTAG protocol for all vendors/boards (through OpenOCD):

On DE0Nano, this can be tested with:

```
python3 -m litex_boards.targets.terasic_de0nano --uart-name=jtag_uart --build --load
litex_term jtag --jtag-config openocd_max10_blaster.cfg
```
This should also work on the DE10-Nano.

tomaz-suller, 2022-10-30:
I've just tried it out, and it doesn't work still, possibly because the OpenOCD configuration would be different for each development boards? (Sorry for my ignorance, I've never used OpenOCD before.)

After executing
```bash
litex_term jtag --jtag-config openocd_max10_blaster2.cfg
```
since my board has a USB Blaster II (and since using `openocd_max10_blaster.cfg` I can't even get the board detected), I get this error message
```
 Open On-Chip Debugger 0.11.0
Licensed under GNU GPL v2
For bug reports, read
        http://openocd.org/doc/doxygen/bugs.html
Info : only one transport option; autoselect 'jtag'
jtagstream_serve
Info : Altera USB-Blaster II found (Firm. rev. = 1.39)
Info : This adapter doesn't support configurable speed
Info : JTAG tap: 10m50.tap tap/device found: 0x02d020dd (mfg: 0x06e (Altera), part: 0x2d02, ver: 0x0)
Warn : JTAG tap: 10m50.tap       UNEXPECTED: 0x02d020dd (mfg: 0x06e (Altera), part: 0x2d02, ver: 0x0)
Error: JTAG tap: 10m50.tap expected 1 of 14: 0x031810dd (mfg: 0x06e (Altera), part: 0x3181, ver: 0x0)
Error: JTAG tap: 10m50.tap expected 2 of 14: 0x0318a0dd (mfg: 0x06e (Altera), part: 0x318a, ver: 0x0)
Error: JTAG tap: 10m50.tap expected 3 of 14: 0x031820dd (mfg: 0x06e (Altera), part: 0x3182, ver: 0x0)
Error: JTAG tap: 10m50.tap expected 4 of 14: 0x031830dd (mfg: 0x06e (Altera), part: 0x3183, ver: 0x0)
Error: JTAG tap: 10m50.tap expected 5 of 14: 0x031840dd (mfg: 0x06e (Altera), part: 0x3184, ver: 0x0)
Error: JTAG tap: 10m50.tap expected 6 of 14: 0x0318d0dd (mfg: 0x06e (Altera), part: 0x318d, ver: 0x0)
Error: JTAG tap: 10m50.tap expected 7 of 14: 0x031850dd (mfg: 0x06e (Altera), part: 0x3185, ver: 0x0)
Error: JTAG tap: 10m50.tap expected 8 of 14: 0x031010dd (mfg: 0x06e (Altera), part: 0x3101, ver: 0x0)
Error: JTAG tap: 10m50.tap expected 9 of 14: 0x0310a0dd (mfg: 0x06e (Altera), part: 0x310a, ver: 0x0)
Error: JTAG tap: 10m50.tap expected 10 of 1: 0x031020dd (mfg: 0x06e (Altera), part: 0x3102, ver: 0x0)
Error: JTAG tap: 10m50.tap expected 11 of 1: 0x031030dd (mfg: 0x06e (Altera), part: 0x3103, ver: 0x0)
Error: JTAG tap: 10m50.tap expected 12 of 1: 0x031040dd (mfg: 0x06e (Altera), part: 0x3104, ver: 0x0)
Error: JTAG tap: 10m50.tap expected 13 of 1: 0x0310d0dd (mfg: 0x06e (Altera), part: 0x310d, ver: 0x0)
Error: JTAG tap: 10m50.tap expected 14 of 1: 0x031050dd (mfg: 0x06e (Altera), part: 0x3105, ver: 0x0)
Info : JTAG tap: auto0.tap tap/device found: 0x4ba00477 (mfg: 0x23b (ARM Ltd), part: 0xba00, ver: 0x4)
Error: Trying to use configured scan chain anyway...
Warn : AUTO auto0.tap - use "jtag newtap auto0 tap -irlen 4 -expected-id 0x4ba00477"
Warn : Bypassing JTAG setup events due to errors
Warn : gdb services need one or more targets defined
```
followed by a bunch of binary data which the terminal can't decode. A sample of that would be
```
����������������������������������������������������������������aaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaXXXXXXXXXXXXXXXXXXXXXXXXXXXXXXXXXXXXXXXXXXXXXXXXXXXXXXXXXXXXXXXX________________________________________________________________������������������������������������������������������������������������������������������������������������������������������������������������������������������������������������������������������������������������������������������������������������������������������������������������������������������������������������������������������������������������������������������������________________________________________________________________����������������������������������������������������������������������������������������������������������������������������������������������������������������������������������������������������������������������������������������������������������������������������������������������������������������������������������������������������������������������������������������������������������������������������������������������������������������������������������������������������������������pppppppppppppppppppppppppppppppppppppppppppppppppppppppppppppppp����������������������������������������������������������������0000000000000000000000000000000000000000000000000000000000000000����������������������������������������������������������������
```

enjoy-digital, 2022-10-30:
This is indeed probably the issue. We would need to add support for this use-case.

tomaz-suller, 2022-10-31:
I'm closing this issue and opening another specific one then. Sorry for having misunderstood what was going on.