enjoy-digital/litex #1478: PLL issues on target terasic_de10lite

udif, 2022-10-30:
I'm new to LiteX, trying out the default SoC for terasic_de10lite.
When building it, with Quartus 21.1, I get failures due to the PLL:
```
Error (15065): Clock input port inclk[0] of PLL "altpll:ALTPLL|altpll_qm91:auto_generated|pll1" must be driven by a non-inverted input pin or another PLL, optionally through a Clock Control block
	Info (15024): Input port INCLK[0] of node "altpll:ALTPLL|altpll_qm91:auto_generated|pll1" is driven by clk50~buf0 which is COMBOUT output port of Combinational cell type node clk50~buf0
```
I traced this to this commit: d531a07719afa31b4f53bdfa1b93d3ccb7cf7746 by @tpwrules in `litex/build/altera/quartus.py` :
```
    attr_translate = {
        "keep": ("keep", 1),
    }
```
Before this change, I get the following code in terasic_de10lite.v:
```
module terasic_de10lite (
    (* keep = "true" *)
    input  wire          clk50, 
    output wire          sdram_clock,
```
Which works fine.

After the change, I get:
```
module terasic_de10lite (
    (* keep = 1 *)
    input  wire          clk50, 
    output wire          sdram_clock,
```
Which yields the error above.
I also tried various variations such as `(* keep *)` and only got  `(* keep="true" *)` to work.
The way the attribute appears in the code on a separate line hinted me it somehow applies to all the ports when in fact it merely applies to the next port definition, and should probably be written as:
```
module terasic_de10lite (
    (* keep = 1 *) input  wire          clk50, 
    output wire          sdram_clock,
```
to be more clear.

Another variation that works is:
```
module terasic_de10lite (
    input  wire          clk50, // synthesis keep
    output wire          sdram_clock,
```
But I think Verilog attributes are more recommended.

I'm not pushing a pull request because I know very little about LiteX or Quartus in general and I would prefer someone with more LiteX and Quartus experience would take a look at this.

tpwrules, 2022-10-30:
I'm pretty sure the variants you say "work" just work because Quartus doesn't recognize the attribute anymore.

It looks like this is issue only turned up with MAX10 because of how restrictive its PLLs are. I think the correct fix would be to not specify the "keep" attribute for clock nets which are also ports. But I don't know how to best go about doing this.

udif, 2022-10-30:
> I'm pretty sure the variants you say "work" just work because Quartus doesn't recognize the attribute anymore.

I'm not so sure, because dropping the attribute completely also doesn't work, so if Quartus couldn't parse it, it should have behaved the same as no attribute at all.
When googling for this, I kept hitting `(* keep="true" *)` for Vivado. Perhaps Quartus has copied this in one of the recent releases? I could not find a single reference as how to use this Verilog attribute with Quartus (not even the Quartus help files).

There is also the issue of where the attribute is placed (above the signals, instead of at the beginning of the input clock line).

tpwrules, 2022-10-30:
Here is where I found the Quartus syntax: https://www.intel.com/content/www/us/en/docs/programmable/683283/18-1/keep-combinational-node-implement-as.html

I don't get the same results as you. On my system with Quartus 20.1.1.720 and MAX 10 support, I get the same error you do with `(* keep = 1 *)` (no matter where it is placed), and do not get any error if it is removed or `1` is replaced with `"true"`.

This commit (https://github.com/enjoy-digital/litex/commit/b6e672a060ce142e79782ef46517a3e77f1b8a02) moved the attribute location if you want to try reverting it but I would be surprised if anything changed.

enjoy-digital, 2022-11-11:
This is fixed with https://github.com/enjoy-digital/litex/commit/62e869296f14fcae9e27f9b3edf7b5c05fe82af1 and https://github.com/enjoy-digital/litex/commit/c1885b333fbf607aa111b817dec9b8c7d85c3b92.

closed 2022-11-11 by enjoy-digital

tpwrules, 2022-11-11:
I'm fairly certain this fix will just result in the constraint not being applied to anything because the node name will change.

Like I said earlier LiteX needs to somehow detect and disable the "keep" attribute for clocks which are on ports. It somehow knows here: https://github.com/enjoy-digital/litex/blob/c1885b333fbf607aa111b817dec9b8c7d85c3b92/litex/build/altera/quartus.py#L91-L102 but at that point the Verilog is already generated so I don't know how it could go back and remove the attribute.

enjoy-digital, 2022-11-11:
@tpwrules: This was indeed a fix to allow the build, but we need to fix it correctly. I was also thinking about disabling the keep attribute on clock ports, but haven't had the time to do it today. I'll have another look at the issue next week.