enjoy-digital/litex #1483: Litex_sim VCD file

foisal-a, 2022-11-01:
I'm trying to observe the signal while running the litex_sim using vexriscv. v file. That work well, I can see all my dumped signals in the sim.vcd file with gtkwave.

But I have a problem for the timescale, by default, the function module->dump() records signal with timescale at 1ps:

$timescale 1ps $end

Do you know how to change it ? I need to change the timescale to 100ps.

Thanks,

enjoy-digital, 2022-11-02:
Hi @foisal-a,

that's a feature I wanted to add to the Verilog generation recently. I'll add it and we could discuss how to use/configure it for your use-case.

enjoy-digital, 2022-11-04:
The generated verilog now integrate timescale with https://github.com/enjoy-digital/litex/commit/adea7879d73c3740d93e4ed54d424fb26317ae1b, a next step will be to expose this in targets and verilator simulation. I'll try to do this soon.

foisal-a, 2022-11-04:
> 

Thanks for your update. 
Please remember one thing that after tracing the litex_sim, in the saved sim.vcd file the time scale shows 1ps but the signal in the gtkwave is changing at 1us when using Vexriscv CPU. It is difficult to observe the signal values in the sim.vcd file for this mismatching the time scale. We need to observe the signal at the same time scale at 1us. Thanks.