enjoy-digital/litex #1491: Too many Verilator warnings?

suarezvictor, 2022-11-04:
When doing simulations with verilator, it usually generates many warnings that pollute the console.
Maybe those are not really dangerous.

The [Makefile](https://github.com/enjoy-digital/litex/blob/master/litex/build/sim/core/Makefile#L53) could be changed to disable those warnings, it needs to just add the following to the already contemplated ones:
`-Wno-TIMESCALEMOD -Wno-COMBDLY -Wno-CASEINCOMPLETE`

enjoy-digital, 2022-11-05:
Hi @suarezvictor,

thanks, I'll have a closer look, we could indeed probably extend the list of disabled warnings.

enjoy-digital, 2022-11-07:
`-Wno-COMBDLY -Wno-CASEINCOMPLETE` has been added with https://github.com/enjoy-digital/litex/commit/f43b92103aa2cb2a3a71056dd73e6558e80e97af. Not sure `-Wno-TIMESCALEMOD` is required now that we are generating timescale in the verilog with https://github.com/enjoy-digital/litex/commit/adea7879d73c3740d93e4ed54d424fb26317ae1b.

closed 2022-11-07 by enjoy-digital

suarezvictor, 2022-11-07:
I had to suggest -Wno-TIMESCALEMOD since in my tests not all the generated verilog files specified timings, particularly cores generated by the "gen.py" files. Maybe there should be an option to make possible the addition of the timescales on such generators