enjoy-digital/litex #1498: Broken build with Altera boards

lapnd, 2022-11-09:
Hi,
I report that the build is broken now for all Altera boards. I have tried with de2-115, deca, and de10lite. All come to the same error as
(this is error log for de10lite)

```sh
Warning (15899): PLL "altpll:ALTPLL|altpll_ij21:auto_generated|pll1" has parameters clk1_multiply_by and clk1_divide_by specified but port CLK[1] is not connected File: /opt/litex-essential/litex-boards/litex_boards/targets/build/terasic_deca/gateware/db/altpll_ij21.tdf Line: 28
Error (15065): Clock input port inclk[0] of PLL "altpll:ALTPLL|altpll_ij21:auto_generated|pll1" must be driven by a non-inverted input pin or another PLL, optionally through a Clock Control block File: /opt/litex-essential/litex-boards/litex_boards/targets/build/terasic_deca/gateware/db/altpll_ij21.tdf Line: 28
    Info (15024): Input port INCLK[0] of node "altpll:ALTPLL|altpll_ij21:auto_generated|pll1" is driven by clk50~buf0 which is COMBOUT output port of Combinational cell type node clk50~buf0 File: /opt/litex-essential/litex-boards/litex_boards/targets/build/terasic_deca/gateware/db/altpll_ij21.tdf Line: 28
Error: Quartus Prime Analysis & Synthesis was unsuccessful. 1 error, 287 warnings
    Error: Peak virtual memory: 812 megabytes
    Error: Processing ended: Wed Nov  9 23:03:13 2022
    Error: Elapsed time: 00:00:31
    Error: Total CPU time (on all processors): 00:00:49
```
I'm using latest commit of litex (1e2ad2250dc18c17c23540e861a8cbbed75a4562) and litex-boards(58489ebebf9bd66a742fb19129dc15bfe4cd53e4), Quartus 21.3

lapnd, 2022-11-09:
Sorry for duplication, the error is the same with https://github.com/enjoy-digital/litex/issues/1455

enjoy-digital, 2022-11-11:
This is fixed with https://github.com/enjoy-digital/litex/commit/62e869296f14fcae9e27f9b3edf7b5c05fe82af1 and https://github.com/enjoy-digital/litex/commit/c1885b333fbf607aa111b817dec9b8c7d85c3b92.

closed 2022-11-11 by enjoy-digital