enjoy-digital/litex #1516: Refactored DNA block sometimes returns incorrect results

bunnie, 2022-11-18:
The change to the DNA block introduced in b24d744f8ed553c023e6ab54e05d237de3dadbb7 seems to have some sort of a timing margin problem. 

I haven't traced down the root cause, but, the symptom is that on some boots I get this DNA string:

`c7589783a285c`

and on other boots I get this string:

`c7789783a285c`

Note the change of 5->7 on the third digit from the left. 

The previous version of the core always returned `c7589783a285c` for this device, so it seems a 0 bit flipped to a 1 sometimes.

The strange part is that I only get the wrong DNA value *sometimes*. It's not consistently reproducible. I only know this because I can see it in the full hardware CI testbench logs; and the failure is very noticeable because the device DNA is used as part of the authentication data for decrypting our secured databases. My immediate suspicion is maybe the value is being sampled on the wrong edge of the clock or something, so a hold-time violation may be at play, but I would imagine that would pop out of the timing analyzer but it doesn't.

My resolution to this is going to be to just fork out the old DNA core and use it as-is -- since the refactor doesn't seem to introduce any new feature that I need, and the old core has been reliable for some years now, I will just stick with the old code that works.

Thus this is not a blocking or priority bug for me, but more of an FYI for future users who hope to rely on the DNA to be deterministic bit-for-bit on every boot.

enjoy-digital, 2022-11-18:
Thanks @bunnie, I'll have a look at this and will do more tests. Can you provide your `sys_clk` freq to be able to test in the same conditions?

bunnie, 2022-11-18:
I'm running a 100MHz `sys_clk` on a -1L speed grade part. The problem seems more reproducible on some chips than others, but that could be explained by either a timing sensitivity, or a dependency on the bit pattern of the DNA code (which varies from chip to chip and can't be modified for testing).

enjoy-digital, 2022-11-18:
Could you try by using a clock at `dna_clk_count[0]` instead of `dna_clk_count[3]`? If working, we could make it configurable and default to 0.

bunnie, 2022-11-18:
On an N=1 test, using `dna_clk_count[0]` works, when it previously did not.

Strange though, the block is rated for 100MHz or so operation according to the datasheet. Could also just be luck of the draw with place and route if it's a hold time issue.

enjoy-digital, 2022-11-18:
With `dna_clk_count[0]` the clk freq is 50MHz and similar to what was used before. I was trying to see if the clk freq change was the reason. If so, as I was saying, we could default to use `dna_clk_count[0]` so sys_clk/2.

closed 2023-02-10 by enjoy-digital