enjoy-digital/litex #1525: Issue with CXX build, triggered by "#include <algorithm>"

tcal-x, 2022-11-29:
In CFU Playground, I'm trying to bump LiteX and related dependencies (https://github.com/google/CFU-Playground/pull/743), and encountered this issue.  I was able to reproduce in a straight LiteX setup.

To reproduce, cd to `litex-boards/litex_boards/targets/` and build Arty (likely any board will work!):
```
./digilent_arty.py --build
```
Edit the following file to **add a line** `#include <algorithm>`:
```
vi ../../../litex/litex/soc/software/demo/hellocpp.cpp
```
Then build the demo with CXX:
```
litex_bare_metal_demo --build-path build/digilent_arty --with-cxx
```
I see the following output:
```
cp: -r not specified; omitting directory '/usr/local/google/home/tcal/enjoy-digital/litex/litex/soc/software/demo/__pycache__'
 CC       donut.o
 CC       helloc.o
 CC       main.o
 CX       hellocpp.o
In file included from /usr/local/google/home/tcal/riscv64-unknown-elf-gcc-10.1.0-2020.08.2-x86_64-linux-ubuntu14/riscv64-unknown-elf/include/stdlib.h:18,
                 from /usr/local/google/home/tcal/riscv64-unknown-elf-gcc-10.1.0-2020.08.2-x86_64-linux-ubuntu14/riscv64-unknown-elf/include/c++/10.1.0/cstdlib:75,
                 from /usr/local/google/home/tcal/riscv64-unknown-elf-gcc-10.1.0-2020.08.2-x86_64-linux-ubuntu14/riscv64-unknown-elf/include/c++/10.1.0/bits/stl_algo.h:59,
                 from /usr/local/google/home/tcal/riscv64-unknown-elf-gcc-10.1.0-2020.08.2-x86_64-linux-ubuntu14/riscv64-unknown-elf/include/c++/10.1.0/algorithm:62,
                 from hellocpp.cpp:2:
/usr/local/google/home/tcal/riscv64-unknown-elf-gcc-10.1.0-2020.08.2-x86_64-linux-ubuntu14/riscv64-unknown-elf/include/sys/reent.h:202:28: error: expected identifier before '*' token
  202 |   _READ_WRITE_RETURN_TYPE (*_read) (struct _reent *, void *,
      |                            ^
/usr/local/google/home/tcal/riscv64-unknown-elf-gcc-10.1.0-2020.08.2-x86_64-linux-ubuntu14/riscv64-unknown-elf/include/sys/reent.h:203:17: error: '_READ_WRITE_BUFSIZE_TYPE' has not been declared
  203 |         char *, _READ_WRITE_BUFSIZE_TYPE);
      |                 ^~~~~~~~~~~~~~~~~~~~~~~~
/usr/local/google/home/tcal/riscv64-unknown-elf-gcc-10.1.0-2020.08.2-x86_64-linux-ubuntu14/riscv64-unknown-elf/include/sys/reent.h:203:41: error: '_READ_WRITE_RETURN_TYPE' declared as function returning a function
  203 |         char *, _READ_WRITE_BUFSIZE_TYPE);
      |                                         ^
/usr/local/google/home/tcal/riscv64-unknown-elf-gcc-10.1.0-2020.08.2-x86_64-linux-ubuntu14/riscv64-unknown-elf/include/sys/reent.h:204:28: error: expected identifier before '*' token
  204 |   _READ_WRITE_RETURN_TYPE (*_write) (struct _reent *, void *,
      |                            ^
/usr/local/google/home/tcal/riscv64-unknown-elf-gcc-10.1.0-2020.08.2-x86_64-linux-ubuntu14/riscv64-unknown-elf/include/sys/reent.h:206:10: error: '_READ_WRITE_BUFSIZE_TYPE' has not been declared
  206 |          _READ_WRITE_BUFSIZE_TYPE);
      |          ^~~~~~~~~~~~~~~~~~~~~~~
/usr/local/google/home/tcal/riscv64-unknown-elf-gcc-10.1.0-2020.08.2-x86_64-linux-ubuntu14/riscv64-unknown-elf/include/sys/reent.h:206:34: error: '_READ_WRITE_RETURN_TYPE' declared as function returning a function
  206 |          _READ_WRITE_BUFSIZE_TYPE);
      |                                  ^
/usr/local/google/home/tcal/riscv64-unknown-elf-gcc-10.1.0-2020.08.2-x86_64-linux-ubuntu14/riscv64-unknown-elf/include/sys/reent.h:292:26: error: conflicting declaration 'typedef struct __sFILE __FILE'
  292 | typedef struct __sFILE   __FILE;
      |                          ^~~~~~
In file included from hellocpp.cpp:1:
/usr/local/google/home/tcal/enjoy-digital/pythondata-software-picolibc/pythondata_software_picolibc/data/newlib/libc/tinystdio/stdio.h:300:23: note: previous declaration as 'typedef struct __file __FILE'
  300 | typedef struct __file __FILE;
      |                       ^~~~~~
make: *** [Makefile:46: hellocpp.o] Error 1
```

I see the same thing whether I use a SiFive riscv toolchain, or one that is installed via Conda from litex-hub.

FYI @alanvgreen

enjoy-digital, 2022-12-01:
@tcal-x: Haven't really tried to look at the issue yet, but just testing ChatGPT to see if it could help on Github issues :):

[image]

tcal-x, 2022-12-01:
@enjoy-digital , that is....interesting.    Did it deduce "helloc.c" from the "helloc.o" in the message output?   I'm puzzled since "helloc.c" isn't mentioned at all.   But in any case, it is mistaken thinking that I'm including `<algorithm>` in a C file.

In any case, I'm mainly hoping that someone else using C++ had encountered this issue and had an easy fix.

mithro, 2023-01-03:
@tcal-x - Have you tried a different RISC-V compiler?

tcal-x, 2023-01-18:
> @tcal-x - Have you tried a different RISC-V compiler?

I see the same thing with these:
* `