enjoy-digital/litex #1532: OpenOCD fails to connect to litex-server

fritzbauer, 2022-12-06:
### Issue
OpenOCD fails to connect to litex-server and exits with the error: 
> remote bridge closed network connection

It seems litex_server was extended to send server information on connection in [this commit](https://github.com/enjoy-digital/litex/commit/35afd59956ee330df1201e7dbf344000a109b114#diff-8d3f7a152f506d85bf434410f8ea1d69e999287027053edef7cf052a693ef6ddR111), however openocd_riscv has not been adjusted to handle it.
[vexriscv_network_read()](https://github.com/SpinalHDL/openocd_riscv/blob/24dc44f2eda6777b7f39bc12523e91de077bb7c8/src/target/vexriscv.c#L1155) is always reading 20 bytes, regardless of the TCP packet length. It seems the first socket read from openocd is when checking [vexriscv_is_halted()](https://github.com/SpinalHDL/openocd_riscv/blob/24dc44f2eda6777b7f39bc12523e91de077bb7c8/src/target/vexriscv.c#L2067). It will receive the server info (which is 23 bytes for me), instead of the vexriscv halted status and fail on subsequent socket reads returning only 3 bytes.

### Personal learning
Since I am quite new to the litex environment, I was wondering whether I really am the only one trying to use openocd since the change in August?! 
Are there some fancier debugging tools/more recommended debugging approaches I am missing out?

fritzbauer, 2022-12-08:
Yes, that's right. My current workaround is to comment `self._send_server_info(client_socket)` out in litex_server.py for debugging sessions. However, need to comment it in for litex_cli to work...

Ch1psN3t, 2023-02-02:
Yes, I also have the problem of connection with litex_server when I launch OpenOCD Server.

fritzbauer, 2023-02-02:
fwiw - I am running on Arch Linux with latest kernel. Not sure whether there might be differences on the network stack between Windows and Linux which might cause that it is only failing on linux?!