enjoy-digital/litex #1536: Cannot build on windows

Peter-van-Tol, 2022-12-14:
The `.bat` file generated on Windows is invalid. Contents of the file are:
```
@echo off
rem Autogenerated by LiteX / git: cb85a8ca

yosys -l colorlight_5a_75e.rpt colorlight_5a_75e.ys
 || exit /bnextpnr-ecp5 --json colorlight_5a_75e.json --lpf colorlight_5a_75e.lpf --textcfg colorlight_5a_75e.config  --25k --package CABGA256 --speed 6 --timing-allow-fail --seed 1 
 || exit /becppack  --bootaddr 0   --compress  colorlight_5a_75e.config --svf colorlight_5a_75e.svf --bit colorlight_5a_75e.bit  || exit /b
```

Problem is are the lines starting witg `||`, I think this should be formatted on the previous line.

The commands themselves are correct, as I can copy them to the command line and build the firmware.

I'm using the latest version of Litex. Thanks for this great project!

Peter-van-Tol, 2022-12-14:
The problem lays in, amongst others, in the file `litex/build/yosys_wrapper.py`. Here the command is created and the string is terminated by a new-line character, i.e.:
```
base_cmd = f"yosys -l {self._build_name}.rpt {self._build_name}.ys\n"
```

In `litex/build/yosys_nextpnr_toolchain.py` the fail-statement is added using string concatenation, so that will end up on the next line. This only affects Windows, because for Linux does not use `fail_stmt` at this moment.

To solve this issue, there are two possible solutions:

1.  Not terminating the commands with `\n`, but leave that to the `build_script` function. This means all functions for commands have to be checked (not only the yosys+nextpnr-ecp5 toolchain I use.
2.  In the `build_script` command use `.rstrip` to strip all white-space (including newlines) from the command. 

I would recommend to go for the second option. In that case the function would look like this:
```python
...
        # yosys call
        script_contents += f"{self._yosys.get_yosys_call('script').rstrip()} {fail_stmt\n"
        # nextpnr call
        script_contents += f"{self._nextpnr.get_call('script').rstrip()} {fail_stmt}\n"
...
```

If you want I can make a PR.

trabucayre, 2022-12-14:
Hi, true `\n` are not correctly uses: [PR](https://github.com/enjoy-digital/litex/pull/1537) triy to fix this error.

trabucayre, 2022-12-14:
Could you try it to validates this PR fixes your issue?