enjoy-digital/litex #1542: Run a simple c program under litex_sim

yicwang, 2022-12-22:
Hi @enjoy-digital,

I am very new to this project, and just started to play with it. I am now able to run the simulator and successfully get into the "litex>" prompt, and I am going to run a simple C hello world. I complied the code with riscv64-unknown-elf-gcc, and have something like foo.bin. The document points me to use serial port (litex_term utility) to upload  the image, but this utility requires a "real" serial port under /dev. With the simulator what do I give over there? How should I upload and run a simple hello world post getting the "litex>" prompt successfully?

Thanks very much!

Regards,
Yichen

enjoy-digital, 2023-01-05:
Hi @yicwang,

the simulation doest not use litex_term. To do what you want in simulation, you can have a look at: https://github.com/enjoy-digital/litex/wiki/SoC-Simulator

yicwang, 2023-01-06:
Perfect, that is all I want!!! Thanks very much!

closed 2023-01-06 by yicwang