enjoy-digital/litex #1545: Meson 1.0.0 build issue

AEW2015, 2022-12-25:
Tools don't like the new Meson Release Dec 23.
```
OSError: Meson version to old. Found: 1.0. Required: 0.59.
Try updating with:
- pip3 install -U meson.
```

I had to force the old Meson version 0.59.

AEW2015, 2022-12-25:
Related issue:
Meson build doesn't seem to work with the new RISC-V GCC tool chain.
```
../../../pythondata-software-picolibc/pythondata_software_picolibc/data/meson.build:35:0: ERROR: Value "c18" (of type "string") for combo option "C language standard to use" is not one of the choices. Possible choices are (as string): "none", "c89", "c99", "c11", "gnu89", "gnu99", "gnu11".
```
Do I have the wrong one on the path?

eli-schwartz, 2022-12-25:
> Tools don't like the new Meson Release Dec 23.

https://github.com/enjoy-digital/litex/blob/e8322587a06f2f3e82443cc6422ba230b11c195a/litex/soc/integration/builder.py#L256-L266

This seems like a pretty odd check to me. It's checking to see if the major version is older than 0, OR if the minor version is older than 59.

Checking the major version doesn't even make *sense*, the minimum requirement here is "0".

Anyway, this check is definitely not a real version comparison. It should use a version tuple instead:

```python
if tuple(int(i) for i in meson_version[:3]) < (0, 59, 0):
         msg = f"Meson version too old. Found ......."
```

Because it's not an either-or.

You could also do this:
```meson
from packaging.version import Version

meson_version = Version(subprocess.check_output(["meson", "-v"]).decode("utf-8"))

if meson_version < Version('0.59.0'):
    msg = f"Meson version too old. Found ......."
```

That does require the https://pypi.org/project/packaging/ module, which is reasonably common at least. It also guarantees proper "version specifiers" support.

AEW2015, 2022-12-25:
The new `apt install riscv64-linux-gnu-gcc` default to gcc 7.5 which causes the second meson error listed above.

AEW2015, 2022-12-27:
Google colab uses Ubuntu 18 and therefore only installs the riscv64 gcc 7.5.  I had to manually install the newer RISCV binaries.

enjoy-digital, 2023-01-03:
Thanks @eli-schwartz for the recommandation. @tmichalak's fix has already been useful to unlock things in CI and implementing the proper check was probably just a matter of time to spend on it. I'll have a closer look at it soon. In case you also want to provide a PR, this is also welcome.

eli-schwartz, 2023-01-04:
Done, see linked PR.

PR #1551 (2023-01-04): replace Meson version check with a specification-compliant version comparator
The current check compares the integers split out from `meson --version` one by one. This is an ad-hoc version comparison algorithm with a few flaws, notably that it doesn't truly understand how version components fit together, and that broke once Meson bumped the major version. There are other potential issues that could show up but haven't yet, such as versions with words in them (release candidates).

The packaging module is a high-quality library that provides a standard version parsing algorithm, with which you  [... cut]
commit c4ec49e1 (2023-01-04): replace Meson version check with a specification-compliant version comparator

The current check compares the integers split out from `meson --version`
one by one. This is an ad-hoc version comparison algorithm with a few
flaws, notably that it doesn't truly understand how version components
fit together, and that broke once Meson bumped the major version. There
are other potential issues that could show up but haven't yet, such as
versions with words in them (release candidates).

The packaging module is a high-quality library that provides a standard
version parsing algorithm, with which you [... cut]
closed 2023-01-04 by enjoy-digital