enjoy-digital/litex #1550: Cannot start Litex_term on/dev/ttyLXU0 (PCIe) when running Linux on Acorn

PatrickChenHZ, 2023-01-03:
Hi, I have been stuck on this issue for days where I built the linux for acorn board but cannot connect to bios or upload linux image via litex_term. More specifically my host computer would just lock up/freeze when I try to use litex_term on the virtual UART port created by litex_pcie driver. The driver loaded fine, and litexpcie_util can correctly see FPGA info and do DMA test. I didn't observer any major error when building the bitstream.
Sorry I am failing to provide more specific info on where the issue is, because I just straight up don't know. It could be FPGA side or PCIe driver side or litex_term, but there is no error message showed when lock up happens, and it happens every single time. I attached syslog when lockup happens below, hope that will help, since that's all I got.

Board:Acorn_CLE215
Project:linux-on-litex-vexriscv
Build command used: `./make.py --board=acorn_pcie --build --flash`
System lock up when executing: `litex_term /dev/ttyLXU0` or `litex_term --image wheremyimageis /dev/ttyLXU0`
Host operating system ubuntu 22.04 (also tried on different computer with ubuntu 18.04 same result)
Host computer: Intel NUC
PCIe Connection: Thunderbolt3 (also tried directly attach to motherboard with same result)

syslog when computer crashed:
```
Jan  3 02:18:50 patrickchen-NUC10i7FNH rtkit-daemon[934]: Supervising 7 threads of 4 processes of 1 users.
Jan  3 02:19:50 patrickchen-NUC10i7FNH rtkit-daemon[934]: message repeated 3 times: [ Supervising 7 threads of 4 processes of 1 users.]
Jan  3 02:20:10 patrickchen-NUC10i7FNH kernel: [ 2011.842655] litepcie: loading out-of-tree module taints kernel.
Jan  3 02:20:10 patrickchen-NUC10i7FNH kernel: [ 2011.842687] litepcie: module verification failed: signature and/or required key missing - tainting kernel
Jan  3 02:20:10 patrickchen-NUC10i7FNH kernel: [ 2011.843039] litepcie 0000:06:00.0: [Probing device]
Jan  3 02:20:10 patrickchen-NUC10i7FNH kernel: [ 2011.843054] litepcie 0000:06:00.0: enabling device (0000 -> 0002)
Jan  3 02:20:10 patrickchen-NUC10i7FNH kernel: [ 2011.863504] litepcie 0000:06:00.0: Version 
Jan  3 02:20:10 patrickchen-NUC10i7FNH kernel: [ 2011.863583] litepcie 0000:06:00.0: 1 MSI IRQs allocated.
Jan  3 02:20:10 patrickchen-NUC10i7FNH kernel: [ 2011.863595] litepcie 0000:06:00.0: Creating /dev/litepcie0
Jan  3 02:20:10 patrickchen-NUC10i7FNH kernel: [ 2011.866648] liteuart.1: ttyLXU0 at MMIO 0x0 (irq = 0, base_baud = 0) is a liteuart
Jan  3 02:20:28 patrickchen-NUC10i7FNH gnome-shell[1697]: libinput error: event10 - MX Keys Keyboard: client bug: event processing lagging behind by 29ms, your system is too slow
Jan  3 02:20:46 patrickchen-NUC10i7FNH nautilus[4678]: Called "net usershare info" but it failed: Failed to execute child process “net” (No such file or directory)
Jan  3 02:20:52 patrickchen-NUC10i7FNH rtkit-daemon[934]: Supervising 7 threads of 4 processes of 1 users.
Jan  3 02:20:52 patrickchen-NUC10i7FNH rtkit-daemon[934]: Supervising 7 threads of 4 processes of 1 users.
Jan  3 02:21:32 patrickchen-NUC10i7FNH gnome-shell[1697]: Window manager warning: last_user_time (2093887) is greater than comparison timestamp (2092697).  This most likely represents a buggy client sending inaccurate timestamps in messages such as _NET_ACTIVE_WINDOW.  Trying to work around...
Jan  3 02:21:32 patrickchen-NUC10i7FNH gnome-shell[1697]: Window manager warning: W20 appears to be one of the offending windows with a timestamp of 2093887.  Working around...
Jan  3 02:21:35 patrickchen-NUC10i7FNH gnome-shell[1697]: Window manager warning: last_user_time (2094449) is greater than comparison timestamp (2094179).  This most likely represents a buggy client sending inaccurate timestamps in messages such as _NET_ACTIVE_WINDOW.  Trying to work around...
Jan  3 02:21:35 patrickchen-NUC10i7FNH gnome-shell[1697]: Window manager warning: W20 appears to be one of the offending windows with a timestamp of 2094449.  Working around...
Jan  3 02:21:38 patrickchen-NUC10i7FNH kernel: [ 2100.237551] watchdog: BUG: soft lockup - CPU#10 stuck for 26s! [swapper/10:0]
Jan  3 02:21:38 patrickchen-NUC10i7FNH kernel: [ 2100.237557] Modules linked in: liteuart(OE) litepcie(OE) ftdi_sio pcspkr ch341 usbserial rfcomm joydev input_leds hid_generic uhid hid cmac algif_hash algif_skcipher af_alg snd_hda_codec_hdmi snd_sof_pci_intel_cnl snd_sof_intel_hda_common soundwire_intel soundwire_generic_allocation soundwire_cadence snd_sof_intel_hda snd_sof_pci snd_sof_xtensa_dsp snd_sof snd_soc_hdac_hda snd_hda_ext_core snd_ctl_led snd_soc_acpi_intel_match snd_hda_codec_realtek snd_soc_acpi soundwire_bus snd_hda_codec_generic bnep ledtrig_audio snd_soc_core snd_compress ac97_bus snd_pcm_dmaengine intel_rapl_msr intel_rapl_common snd_hda_intel x86_pkg_temp_thermal snd_intel_dspcfg snd_intel_sdw_acpi snd_hda_codec intel_powerclamp snd_hda_core snd_hwdep coretemp snd_pcm kvm_intel snd_seq_midi binfmt_misc iwlmvm snd_seq_midi_event kvm tps6598x i915 mei_hdcp typec mac80211 snd_rawmidi crct10dif_pclmul ghash_clmulni_intel libarc4 btusb ttm btrtl snd_seq btbcm aesni_intel drm_kms_helper nls_iso8859_1 crypto_simd btintel
Jan  3 02:21:38 patrickchen-NUC10i7FNH kernel: [ 2100.237592]  cryptd snd_seq_device cec iwlwifi bluetooth snd_timer rc_core rapl i2c_algo_bit fb_sys_fops intel_cstate intel_wmi_thunderbolt wmi_bmof snd ecdh_generic syscopyarea cfg80211 ee1004 ecc soundcore sysfillrect sysimgblt mei_me intel_pch_thermal mei serial_multi_instantiate mac_hid acpi_tad acpi_pad sch_fq_codel msr parport_pc ppdev lp drm parport ramoops pstore_blk reed_solomon pstore_zone efi_pstore ip_tables x_tables autofs4 sdhci_pci cqhci nvme i2c_i801 crc32_pclmul e1000e thunderbolt sdhci nvme_core i2c_smbus intel_lpss_pci xhci_pci ahci intel_lpss libahci idma64 xhci_pci_renesas wmi video pinctrl_cannonlake
Jan  3 02:21:38 patrickchen-NUC10i7FNH kernel: [ 2100.237620] CPU: 10 PID: 0 Comm: swapper/10 Tainted: G           OE     5.15.0-56-generic #62-Ubuntu
Jan  3 02:21:38 patrickchen-NUC10i7FNH kernel: [ 2100.237622] Hardware name: Intel(R) Client Systems NUC10i7FNH/NUC10i7FNB, BIOS FNCML357.0055.2021.1202.1748 12/02/2021
Jan  3 02:21:38 patrickchen-NUC10i7FNH kernel: [ 2100.237623] RIP: 0010:liteuart_timer+0x8d/0x100 [liteuart]
Jan  3 02:21:38 patrickchen-NUC10i7FNH kernel: [ 2100.237628] Code: ff 00 00 00 00 45 31 c0 44 89 e9 31 d2 be 01 00 00 00 4c 89 ff e8 c3 08 b0 c5 49 8b bc 24 f0 fe ff ff e8 36 46 ae c5 41 8b 06 <84> c0 74 9d 41 8b 84 24 40 ff ff ff 83 f8 06 7f 40 b8 01 00 00 00
Jan  3 02:21:38 patrickchen-NUC10i7FNH kernel: [ 2100.237629] RSP: 0018:ffffb65ac037ce80 EFLAGS: 00000202
Jan  3 02:21:38 patrickchen-NUC10i7FNH kernel: [ 2100.237631] RAX: 0000000000000000 RBX: ffffb65ac3d06820 RCX: ffff8bc41cfe3a10
Jan  3 02:21:38 patrickchen-NUC10i7FNH kernel: [ 2100.237632] RDX: 0000000000000000 RSI: 0000000000000006 RDI: ffff8bc400051000
Jan  3 02:21:38 patrickchen-NUC10i7FNH kernel: [ 2100.237633] RBP: ffffb65ac037cea8 R08: 0000000000000001 R09: ffffffff886898d8
Jan  3 02:21:38 patrickchen-NUC10i7FNH kernel: [ 2100.237633] R10: 0000000000000003 R11: 000000000000000a R12: ffff8bc417318200
Jan  3 02:21:38 patrickchen-NUC10i7FNH kernel: [ 2100.237634] R13: 0000000000000000 R14: ffffb65ac3d06828 R15: ffff8bc417318028
Jan  3 02:21:38 patrickchen-NUC10i7FNH kernel: [ 2100.237635] FS:  0000000000000000(0000) GS:ffff8bc7b0e80000(0000) knlGS:0000000000000000
Jan  3 02:21:38 patrickchen-NUC10i7FNH kernel: [ 2100.237636] CS:  0010 DS: 0000 ES: 0000 CR0: 0000000080050033
Jan  3 02:21:38 patrickchen-NUC10i7FNH kernel: [ 2100.237637] CR2: 00007f5eb5c06658 CR3: 0000000268e10003 CR4: 00000000003706e0
Jan  3 02:21:38 patrickchen-NUC10i7FNH kernel: [ 2100.237638] Call Trace:
Jan  3 02:21:38 patrickchen-NUC10i7FNH kernel: [ 2100.237639]  <IRQ>
Jan  3 02:21:38 patrickchen-NUC10i7FNH kernel: [ 2100.237641]  ? liteuart_startup+0x70/0x70 [liteuart]
Jan  3 02:21:38 patrickchen-NUC10i7FNH kernel: [ 2100.237643]  call_timer_fn+0x29/0x120
Jan  3 02:21:38 patrickchen-NUC10i7FNH kernel: [ 2100.237646]  __run_timers.part.0+0x1e3/0x270
Jan  3 02:21:38 patrickchen-NUC10i7FNH kernel: [ 2100.237648]  ? ktime_get+0x43/0xc0
Jan  3 02:21:38 patrickchen-NUC10i7FNH kernel: [ 2100.237650]  ? lapic_next_deadline+0x2c/0x40
Jan  3 02:21:38 patrickchen-NUC10i7FNH kernel: [ 2100.237652]  ? clockevents_program_event+0xaa/0x130
Jan  3 02:21:38 patrickchen-NUC10i7FNH kernel: [ 2100.237655]  run_timer_softirq+0x2a/0x60
[... cut ...]
Jan  3 02:22:13 patrickchen-NUC10i7FNH kernel: [ 2134.877331]  cpuidle_idle_call+0x142/0x1e0
Jan  3 02:22:13 patrickchen-NUC10i7FNH kernel: [ 2134.877332]  do_idle+0x83/0xf0
Jan  3 02:22:13 patrickchen-NUC10i7FNH kernel: [ 2134.877334]  cpu_startup_entry+0x20/0x30
Jan  3 02:22:13 patrickchen-NUC10i7FNH kernel: [ 2134.877335]  start_secondary+0x12a/0x180
Jan  3 02:22:13 patrickchen-NUC10i7FNH kernel: [ 2134.877336]  secondary_startup_64_no_verify+0xc2/0xcb
Jan  3 02:22:13 patrickchen-NUC10i7FNH kernel: [ 2134.877338]  </TASK>

```
Thank you!

enjoy-digital, 2023-01-03:
Hi @PatrickChenHZ,

I can probably help on this and have the hardware at hand. Can you send attach your bistream + litepcie kernel (including csr.h). I'll do a test to see if it's related to the hardware or to the bistream/software header generation.

I could also share a version working on my setup to do tests on yours.

PatrickChenHZ, 2023-01-04:
[kernel.zip](https://github.com/enjoy-digital/litex/files/10341703/kernel.zip)
[gateware.zip](https://github.com/enjoy-digital/litex/files/10341710/gateware.zip)

Thank you so much!

sthornington, 2024-04-20:
Just wanted to add that I am in the same spot. However, I can add that creating the target directly from litex-boards, I can get litex_term to see the BIOS works fine. I’m going to check whether adding more flags to the AcornPCIE target, when launched from make.py, to correspond to how I ran the script on the command line (e.g. —driver), produces working drivers.

sthornington, 2024-04-20:
In what may or may not be a coincidence, I’m also using a NUC 10 i7FNH with the Acorn connected via a Thunderbolt bay for the M.2 card.

sthornington, 2024-04-20:
Did you ever figure this out @PatrickChenHZ ?

sthornington, 2024-04-20:
Something definitely weird with the Linux-on-litex build, I wonder if it’s something to do with the operational/fallback stuff. In that documented anywhere? 

Raw build from the target in litex-boards seems to work reliably, but once flashed with the Linux-on-litex version, the PCIe stuff is sketchy, sometimes the board shuts down completely, with the fan turning off and the led chaser stopping even.

sthornington, 2024-04-21:
Okay I got it to load in "safe" mode, via `litex_term --images=images/boot.json --safe /dev/ttyLXU0`.  I also, personally, changed the make.py to not specifically use the fallback bitstream when flashing, since I think that might also have been causing problems?