enjoy-digital/litex #1552: Add support for Novastar cards

romanetz, 2023-01-04:
brand: Novastar MRV416
FPGA: SL3-22E-6B324C (unknown family. possibly vendor Intilligence Silicon)
Memory: ESMT M14D5121632A
Eth PHY: PF025
Additional MCU: Artery technology AT32F421G8U7 ARM Cortex-M4 based
Flash: Fudan micro FM25W32
16 HUB75 ports

enjoy-digital, 2023-01-05:
Hi @romanetz,

are you planning to add support for it? If so, I would recommend splitting this in different tasks:
1. Add support for Inilligence Silicon FPGA in LiteX (through a dev board?).
2. Reverse the Novastar MRV416 board.
3. Add support for the board in LiteX-Boards.

As you can see, some non negligeable effort is required to support this board, so unless you are going to put some efforts or know someone interested to do so, this will probably not happen.

enjoy-digital, 2024-01-12:
Closing since no feedback. Still happy to see the Novastar board supported if someone is willing to reverse them and add support for them in LiteX-Boards.

closed 2024-01-12 by enjoy-digital