enjoy-digital/litex #1555: Zephyr: Make sure tutorial is up to date and mention the --timer-uptime requirement.

enjoy-digital, 2023-01-06:


mithro, 2023-01-06:
@kgugala - Who at Antmicro should look at this?

pgielda, 2023-01-06:
@enjoy-digital can you elaborate on what tutorial you mean?

enjoy-digital, 2023-01-06:
@mithro, @pgielda: Sorry, this was just a reminder to have a look at it later, regarding a discussion on IRC: https://libera.irclog.whitequark.org/litex/2023-01-06#33611390;

mithro, 2023-01-07:
Looks like the discussion reference the https://github.com/antmicro/zephyr-on-litex-vexriscv repository and specifically https://github.com/antmicro/zephyr-on-litex-vexriscv/blob/0775f94d3537ec5e967ea2ac2b6aba0de3fa71af/soc_zephyr.py

pgielda, 2023-01-07:
Its a fork (probably created to open a PR), seems that litex-hub "zephyr-on-litex-vexriscv" repo is newer? In any case I was more asking about a tutorial that is broken/not up to date (unless there is no tutorial)

enjoy-digital, 2023-01-07:
> In any case I was more asking about a tutorial that is broken/not up to date (unless there is no tutorial)

That's mostly what I wanted to figure out when opening this PR :), just wanted to see if use of --timer-uptime was included since seems to be required. I'll have a look and provide an update, no need to spend much time on this yet.