enjoy-digital/litex #1556: vexrixcv_smp sim-build fails

pottendo, 2023-01-06:
Hi,
thanks for this great project!

I've update my litex-setup and wanted to get started with a vexriscv-smp configuration - but litex-sim fails:
```
$ litex_sim --cpu-type vexriscv_smp
[...]
verilator -Wno-fatal -O3 --cc /opt/work/src/orangeCart/supporting-tools/litex/pythondata-cpu-vexriscv-smp/pythondata_cpu_vexriscv_smp/verilog/Ram_1w_1rs_Generic.v --cc /opt/work/src/orangeCart/supporting-tools/litex/pythondata-cpu-vexriscv-smp/pythondata_cpu_vexriscv_smp/verilog/VexRiscvLitexSmpCluster_Cc1_Iw32Is4096Iy1_Dw32Ds4096Dy1_ITs4DTs4_Ood_Wm.v --cc /opt/work/src/orangeCart/supporting-tools/litex/build/sim/gateware/sim.v  --top-module sim --exe \
	-DPRINTF_COND=0 \
	sim_init.cpp /opt/work/src/orangeCart/supporting-tools/litex/litex/litex/build/sim/core/veril.cpp libdylib.o modules.o pads.o parse.o sim.o \
	--top-module sim \
	 \
	-CFLAGS "-ggdb -Wall -O3   -I/opt/work/src/orangeCart/supporting-tools/litex/litex/litex/build/sim/core" \
	-LDFLAGS "-lpthread -Wl,--no-as-needed -ljson-c -lz -lm -lstdc++ -Wl,--no-as-needed -ldl -levent" \
	--trace \
	 \
	 \
	--unroll-count 256 \
	--output-split 5000 \
	--output-split-cfuncs 500 \
	--output-split-ctrace 500 \
	 \
	-Wno-BLKANDNBLK \
	-Wno-WIDTH \
	-Wno-COMBDLY \
	-Wno-CASEINCOMPLETE \
	--relative-includes
%Error-TIMESCALEMOD: /opt/work/src/orangeCart/supporting-tools/litex/pythondata-cpu-vexriscv-smp/pythondata_cpu_vexriscv_smp/verilog/Ram_1w_1rs_Generic.v:2:8: Timescale missing on this module as other modules have it (IEEE 1800-2017 3.14.2.2)
                     /opt/work/src/orangeCart/supporting-tools/litex/pythondata-cpu-vexriscv-smp/pythondata_cpu_vexriscv_smp/verilog/VexRiscvLitexSmpCluster_Cc1_Iw32Is4096Iy1_Dw32Ds4096Dy1_ITs4DTs4_Ood_Wm.v:6:8: ... Location of module with timescale
    6 | module VexRiscvLitexSmpCluster_Cc1_Iw32Is4096Iy1_Dw32Ds4096Dy1_ITs4DTs4_Ood_Wm (
      |        ^~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
%Error: Exiting due to 1 error(s)
        ... See the manual and https://verilator.org for more assistance.
make: *** [/opt/work/src/orangeCart/supporting-tools/litex/litex/litex/build/sim/core/Makefile:44: sim] Error 1
make: Leaving directory '/opt/work/src/orangeCart/supporting-tools/litex/build/sim/gateware'
```

I found some [thread](https://github.com/enjoy-digital/litex/issues/1491) about TIMESCALEMOD, but trying to add `-Wno-TIMESCALEMOD` resulted in different errors:
```
%Error: /opt/work/src/orangeCart/supporting-tools/litex/pythondata-cpu-vexriscv-smp/pythondata_cpu_vexriscv_smp/verilog/VexRiscvLitexSmpCluster_Cc1_Iw32Is4096Iy1_Dw32Ds4096Dy1_ITs4DTs4_Ood_Wm.v:14242:17: Unsupported or unknown PLI call: '$urandom'
14242 |     buffers_0 = $urandom;
      |                 ^~~~~~~~
%Error: /opt/work/src/orangeCart/supporting-tools/litex/pythondata-cpu-vexriscv-smp/pythondata_cpu_vexriscv_smp/verilog/VexRiscvLitexSmpCluster_Cc1_Iw32Is4096Iy1_Dw32Ds4096Dy1_ITs4DTs4_Ood_Wm.v:14243:17: Unsupported or unknown PLI call: '$urandom'
14243 |     buffers_1 = $urandom;
      |                 ^~~~~~~~
%Error: /opt/work/src/orangeCart/supporting-tools/litex/pythondata-cpu-vexriscv-smp/pythondata_cpu_vexriscv_smp/verilog/VexRiscvLitexSmpCluster_Cc1_Iw32Is4096Iy1_Dw32Ds4096Dy1_ITs4DTs4_Ood_Wm.v:14191:24: Unsupported or unknown PLI call: '$urandom'
14191 |     inputArea_target = $urandom;
      |                        ^~~~~~~~
%Error: /opt/work/src/orangeCart/supporting-tools/litex/pythondata-cpu-vexriscv-smp/pythondata_cpu_vexriscv_smp/verilog/VexRiscvLitexSmpCluster_Cc1_Iw32Is4096Iy1_Dw32Ds4096Dy1_ITs4DTs4_Ood_Wm.v:14192:22: Unsupported or unknown PLI call: '$urandom'
14192 |     outputArea_hit = $urandom;
      |                      ^~~~~~~~
%Error: Exiting due to 4 error(s)
```

My setup builds and boots into the console fine with `litex_sim --cpu-type vexriscv`.

Any hints? 
pottendo

Dolu1990, 2023-01-06:
Hi, normaly, updating verilator to 4.2xxx should fix it.

pottendo, 2023-01-07:
> Hi, normaly, updating verilator to 4.2xxx should fix it.

Big thanks - Ubuntu 22.04 stays at:
  Verilator 4.038 2020-07-11 rev v4.036-114-g0cd4a57ad

I've pulled the latest stable verilator (Verilator 5.004 2022-12-14 rev v5.004-30-g8468af1a2) and now it builds and runs as expected for simple(r) invocations for vexriscv_smp (no specific options)
But for e.g. 
`litex_sim --cpu-type vexriscv_smp --integrated-sram-size 10000 --integrated-main-ram-size=0x1000000 --with-fpu --with-rvc --cpu-count 2`
the process progresses until here:
```
[...]
[gmii_ethernet] loaded (0x563725979ef0)
[ethernet] loaded (0x563725979ef0)
[xgmii_ethernet] loaded (0x563725979ef0)
[serial2console] loaded (0x563725979ef0)
[clocker] loaded
[spdeeprom] loaded (addr = 0x0)
[serial2tcp] loaded (0x563725979ef0)
[clocker] sys_clk: freq_hz=1000000, phase_deg=0
```
and stood there for a *very* long time (>30 min) until I interrupted. On process was executing 100% (VSim I think).

Pls. confirm, that this command finishes (maybe after hours - I couldn't wait that long).

Fyi, I have another target using the vexriscv_smp cpu (see [here](https://github.com/pottendo/RVCop64-pottendo). Generating a DualCore CPU takes around 10-15min on my system. A single core (with FPU, ISA-C) is even faster, maybe 5 min. So the generation of the sim target above may still have some issues. 

thanks for all the effort here - great work!
pottendo

enjoy-digital, 2023-01-10:
@pottendo: The issues seems to appears directly with `litex_sim --cpu-type vexriscv_smp --with-fpu`, adding` --wishbone-force` fix it (and also for your use-case). You can use that as a temporary workaround until this is understood.

pottendo, 2023-01-10:
Hi, 
I confirm, that the litex BIOS promt shows up now - the workaround works for me.
You may close this issue.
thanx, pottendo

PS: fyi, see fairly low memory performance:
```
$ litex_sim --cpu-type vexriscv_smp --integrated-sram-size 0x10000 --integrated-main-ram-size=0x1000000
--========== Initialization ============--
Memtest at 0x40000000 (2.0MiB)...
  Write: 0x40000000-0x40200000 2.0MiB     
   Read: 0x40000000-0x40200000 2.0MiB     
Memtest OK
Memspeed at 0x40000000 (Sequential, 2.0MiB)...
  Write speed: 1.2MiB/s
   Read speed: 771.5KiB/s
```
This is independent from `--wishbone-force' and other parameters.
I have a AMD Ryzen 5 3600 6-Core Processor.
I have only the comparison with a real FPGA + Hyperram board, where I get up to 60MiB/s read speed.
Maybe this is normal for lite_sim using this CPU - just wanted to let you know my observation.

Dolu1990, 2023-01-10:
will check it tomorrow

Dolu1990, 2023-01-11:
For the mem speed, the simulation simulate the SoC at 1 Mhz, that's why the meme speed test mesure a low speed. but relative to the clock frequancy, that's normal. Also note that if not using litedram, it use wishbone, which has half the bandwidth.

About the issue  litex_sim --cpu-type vexriscv_smp --integrated-sram-size 10000 --integrated-main-ram-size=0x1000000 --with-fpu --with-rvc --cpu-count 2

It is that the SoC is generated with a 64 bits wishbone, while litex is notified of a 32 bits wishbone (peripheral bus).

The issue is that the bus is added to wishbone in the core.py __init__
https://github.com/enjoy-digital/litex/blob/master/litex/soc/cores/cpu/vexriscv_smp/core.py#L290

while the final wishbone width is decided in the do_finalize : 
https://github.com/enjoy-digital/litex/blob/master/litex/soc/cores/cpu/vexriscv_smp/core.py#L490

So, the solution would be to move that code from core.py into the do_finalize. Is it possible ? or it would violate something ? @enjoy-digital

enjoy-digital, 2023-01-11:
@Dolu1990: Thanks, will try to look/answer in the afternoon.

enjoy-digital, 2023-01-16:
Hi @Dolu1990,

for now I changed the code to raise an error instead of silently generating an invalid configuration. I would need to see how to do that without manually specifying `--with-wishbone-memory ` since `pbus` is added to the SoC after the CPU declaration (https://github.com/enjoy-digital/litex/blob/master/litex/soc/integration/soc.py#L1089-L1090) so would have the 32-bit data_width. We could eventually keep `periph_buses ` empty and add if to the SoC when `pbus` size is really determined as you suggest. I'll do some tests.

Dolu1990, 2023-01-16:
Ok thanks :)

enjoy-digital, 2023-01-16:
@Dolu1990: BTW, what is the condition in VexRiscv-SMP to have a 64-bit data_width bus? Is it always the case when there are no direct memory buses?

Dolu1990, 2023-01-17:
It become 64 bits wishbone bus if there is no litedram and (the FPU is enabled or there is more than 1 CPU core) and no --wishbone-force-32b

enjoy-digital, 2023-01-17:
@Dolu1990: OK thanks, I'll try to add these conditions in the wrapper also then.

enjoy-digital, 2024-01-12:
We can now probably close this since answered/fixed.

closed 2024-01-12 by enjoy-digital