enjoy-digital/litex #1568: FileNotFoundError generic_fpga.pl of openc906

vietthanh85, 2023-01-14:
I got error when LiteX was generating RTL code with openc906 core (--cpu-type=openc906).

`FileNotFoundError: [Errno 2] No such file or directory: '/workspace/opt/litex-essential/openc906/C906_RTL_FACTORY/gen_rtl/filelists/generic_fpga.fl'`

Currently, there is no generic_fpga.fl on openc906 repo:
https://github.com/T-head-Semi/openc906/tree/main/C906_RTL_FACTORY/gen_rtl/filelists

May I know how to get `generic_fpga.fl' file?
Thanks in advance!

enjoy-digital, 2023-01-16:
Hi  @vietthanh85,

for now you have to use this fork: https://github.com/Icenowy/openc906

We'll probably add specific repo to litex-hub for openc906.

vietthanh85, 2023-01-17:
@enjoy-digital Thanks. I am able to generate and compile RTL successfully.

sususjysjy, 2023-09-25:
> @enjoy-digital Thanks. I am able to generate and compile RTL successfully.

i don't see the file in this repo
could you tell me some details about this file?

vietthanh85, 2023-10-01:
To Kevin SU
> i don't see the file in this repo
> could you tell me some details about this file?

Sorry for missing your comment.
The file is located here for your reference.
https://github.com/Icenowy/openc906/blob/fpga-optimization/C906_RTL_FACTORY/gen_rtl/filelists/generic_fpga.fl

sususjysjy, 2023-10-02:
> To Kevin SU
> 
> > i don't see the file in this repo
> > could you tell me some details about this file?
> 
> Sorry for missing your comment. The file is located here for your reference. https://github.com/Icenowy/openc906/blob/fpga-optimization/C906_RTL_FACTORY/gen_rtl/filelists/generic_fpga.fl

thanks your reply , it is my mistake, I didn't switch branches
thanks.:)

sususjysjy, 2023-10-29:
> To Kevin SU
> 
> > i don't see the file in this repo
> > could you tell me some details about this file?
> 
> Sorry for missing your comment. The file is located here for your reference. https://github.com/Icenowy/openc906/blob/fpga-optimization/C906_RTL_FACTORY/gen_rtl/filelists/generic_fpga.fl

thanks. have you boot Linux with C906?

vietthanh85, 2023-10-29:
> thanks. have you boot Linux with C906?

Not yet Kevin.