enjoy-digital/litex #1569: Porting to Xilinx Alveo U200

vietthanh85, 2023-01-14:
I want to run LiteX on xilinx_alveo_u200 board.
Since `xilinx_alveo_u200` is not available in LiteX so I have modified from `xilinx_alveo_u250`.
The target file was kept the same as `xilinx_alveo_u250`.
As for platform file, I only replace the FPGA part name from `xcu250-figd2104-2L-e` to `xcu200-fsgd2104-2-e`. 

Build command:
`./xilinx_alveo_u200.py --build`

Result: Memtest failed.
`--=============== SoC ==================--
CPU:		VexRiscv SMP-LINUX @ 125MHz
BUS:		WISHBONE 32-bit @ 4GiB
CSR:		32-bit data
ROM:		128KiB
SRAM:		8KiB
L2:		8KiB
SDRAM:		1048576KiB 64-bit @ 1000MT/s (CL-9 CWL-9)

--========== Initialization ============--
Initializing SDRAM @0x40000000...
Switching SDRAM to software control.
Write leveling:
  tCK equivalent taps: 612
  Cmd/Clk scan (0-306)
  |11111  |11111  |11111  |11111| best: 0
  Setting Cmd/Clk delay to 0 taps.
  Data scan:
  m0: |11111111111111111111111| delay: 00
  m1: |11111111111111111111111| delay: 00
  m2: |11111111111111111111111| delay: 00
  m3: |11111111111111111111111| delay: 00
  m4: |11111111111111111111111| delay: 00
  m5: |11111111111111111111111| delay: 00
  m6: |11111111111111111111111| delay: 00
  m7: |11111111111111111111111| delay: 00
Write latency calibration:
m0:0 m1:0 m2:0 m3:0 m4:0 m5:0 m6:0 m7:0 
Read leveling:
  m0, b00: |00000000000000000000000000000000| delays: -
  m0, b01: |00000000000000000000000000000000| delays: -
  m0, b02: |00000000000000000000000000000000| delays: -
  m0, b03: |00000000000000000000000000000000| delays: -
  m0, b04: |00000000000000000000000000000000| delays: -
  m0, b05: |00000000000000000000000000000000| delays: -
  m0, b06: |00000000000000000000000000000000| delays: -
  m0, b07: |00000000000000000000000000000000| delays: -
  best: m0, b00 delays: -
  m1, b00: |00000000000000000000000000000000| delays: -
  m1, b01: |00000000000000000000000000000000| delays: -
  m1, b02: |00000000000000000000000000000000| delays: -
  m1, b03: |00000000000000000000000000000000| delays: -
  m1, b04: |00000000000000000000000000000000| delays: -
  m1, b05: |00000000000000000000000000000000| delays: -
  m1, b06: |00000000000000000000000000000000| delays: -
  m1, b07: |00000000000000000000000000000000| delays: -
  best: m1, b00 delays: -
  m2, b00: |00000000000000000000000000000000| delays: -
  m2, b01: |00000000000000000000000000000000| delays: -
  m2, b02: |00000000000000000000000000000000| delays: -
  m2, b03: |00000000000000000000000000000000| delays: -
  m2, b04: |00000000000000000000000000000000| delays: -
  m2, b05: |00000000000000000000000000000000| delays: -
  m2, b06: |00000000000000000000000000000000| delays: -
  m2, b07: |00000000000000000000000000000000| delays: -
  best: m2, b00 delays: -
  m3, b00: |00000000000000000000000000000000| delays: -
  m3, b01: |00000000000000000000000000000000| delays: -
  m3, b02: |00000000000000000000000000000000| delays: -
  m3, b03: |00000000000000000000000000000000| delays: -
  m3, b04: |00000000000000000000000000000000| delays: -
  m3, b05: |00000000000000000000000000000000| delays: -
  m3, b06: |00000000000000000000000000000000| delays: -
  m3, b07: |00000000000000000000000000000000| delays: -
  best: m3, b00 delays: -
  m4, b00: |00000000000000000000000000000000| delays: -
  m4, b01: |00000000000000000000000000000000| delays: -
  m4, b02: |00000000000000000000000000000000| delays: -
  m4, b03: |00000000000000000000000000000000| delays: -
  m4, b04: |00000000000000000000000000000000| delays: -
  m4, b05: |00000000000000000000000000000000| delays: -
  m4, b06: |00000000000000000000000000000000| delays: -
  m4, b07: |00000000000000000000000000000000| delays: -
  best: m4, b00 delays: -
  m5, b00: |00000000000000000000000000000000| delays: -
  m5, b01: |00000000000000000000000000000000| delays: -
  m5, b02: |00000000000000000000000000000000| delays: -
  m5, b03: |00000000000000000000000000000000| delays: -
  m5, b04: |00000000000000000000000000000000| delays: -
  m5, b05: |00000000000000000000000000000000| delays: -
  m5, b06: |00000000000000000000000000000000| delays: -
  m5, b07: |00000000000000000000000000000000| delays: -
  best: m5, b00 delays: -
  m6, b00: |00000000000000000000000000000000| delays: -
  m6, b01: |00000000000000000000000000000000| delays: -
  m6, b02: |00000000000000000000000000000000| delays: -
  m6, b03: |00000000000000000000000000000000| delays: -
  m6, b04: |00000000000000000000000000000000| delays: -
  m6, b05: |00000000000000000000000000000000| delays: -
  m6, b06: |00000000000000000000000000000000| delays: -
  m6, b07: |00000000000000000000000000000000| delays: -
  best: m6, b00 delays: -
  m7, b00: |00000000000000000000000000000000| delays: -
  m7, b01: |00000000000000000000000000000000| delays: -
  m7, b02: |00000000000000000000000000000000| delays: -
  m7, b03: |00000000000000000000000000000000| delays: -
  m7, b04: |00000000000000000000000000000000| delays: -
  m7, b05: |00000000000000000000000000000000| delays: -
  m7, b06: |00000000000000000000000000000000| delays: -
  m7, b07: |00000000000000000000000000000000| delays: -
  best: m7, b00 delays: -
Switching SDRAM to hardware control.
Memtest at 0x40000000 (2.0MiB)...
  Write: 0x40000000-0x40200000 2.0MiB     
   Read: 0x40000000-0x40200000 2.0MiB     
  bus errors:  256/256
  addr errors: 0/8192
  data errors: 524288/524288
Memtest KO
Memory initialization failed`

Any suggestion is appreciated!
Thanks in advance!

enjoy-digital, 2023-01-15:
Hi @vietthanh85,

one first thing to check is if INTERNAL_VREF constraints are set correctly on your DRAM banks, ex: https://github.com/litex-hub/litex-boards/blob/master/litex_boards/platforms/xilinx_alveo_u280.py#L242-L249

vietthanh85, 2023-01-15:
@enjoy-digital Thank you. As I reused the source code from `xilinx_alveo_u250`, so INTERNAL_VREF constraints were set in the platform file.
https://github.com/litex-hub/litex-boards/blob/master/litex_boards/platforms/xilinx_alveo_u250.py#L352-L366

enjoy-digital, 2023-01-16:
Thanks @vietthanh85, the numbering was indeed not correct on the U250 and fixed with https://github.com/litex-hub/litex-boards/commit/be77f827204735129699862ead21342f1ea27fc0.

Are you planning to contribute the U200 support to LiteX-Boards? If so, could you create a PR to LiteX-Boards? This would allow me to have a look at the current code and maybe find a possible issue.

vietthanh85, 2023-01-16:
@enjoy-digital 
Sure, could you take a look into this PR https://github.com/litex-hub/litex-boards/pull/472. Thanks!

smosanu, 2023-01-19:
Hi! I went through the same thing with porting for the Alveo U280 board. In addition to the device name I also had to change the pin values for the DRAM and a few other peripherals (clk, rst). Did you do that?

vietthanh85, 2023-01-20:
@smosanu Thank you for your checking. I checked the pins of U250 and U200. Actually their pin names are the same. I referred the following xml from Xilinx:

https://github.com/Xilinx/open-nic-shell/blob/main/board_files/Xilinx/au250/1.3/part0_pins.xml
https://github.com/Xilinx/open-nic-shell/blob/main/board_files/Xilinx/au200/1.3/part0_pins.xml

If you diff them, the differences are only the part name and space characters which look like:

[image]

[image]

So I don't think we need to change to pin values when porting from U250.

smosanu, 2023-01-24:
Hi @vietthanh85,
can you please try with 2 or more vexriscv-smp cores or with a naxriscv core? I also had similar issues and it had to do with the bus width.
- Serg

vietthanh85, 2023-01-25:
@smosanu Thanks. I have tried both naxriscv and rocket cores but I got the same issue.

smosanu, 2023-01-30:
Hi @vietthanh85,

Honestly, I don't know why you are observing this behavior. Something's wrong. This is what I get on Alveo U280 board (below). I run the system at 150MHz, so that times 4 gives 600MHz for the memory. I also set `cmd_latency      = 1` here https://github.com/litex-hub/litex-boards/blob/b8abdf1b39d322101e5ec88d0e0650b42f836a80/litex_boards/targets/xilinx_alveo_u280.py#L132

```
        __   _ __      _  __
       / /  (_) /____ | |/_/
      / /__/ / __/ -_)>  <
     /____/_/\__/\__/_/|_|
   Build your hardware, easily!

 (c) Copyright 2012-2022 Enjoy-Digital
 (c) Copyright 2007-2015 M-Labs

 BIOS built on Dec  7 2022 13:41:51
 BIOS CRC passed (79504cea)

 LiteX git sha1: d6bbf655

--=============== SoC ==================--
CPU:            NaxRiscv @ 150MHz
BUS:            WISHBONE 32-bit @ 4GiB
CSR:            32-bit data
ROM:            128KiB
SRAM:           8KiB
L2:             8KiB
SDRAM:          1048576KiB 64-bit @ 1200MT/s (CL-9 CWL-9)

--========== Initialization ============--
Initializing SDRAM @0x40000000...
Switching SDRAM to software control.
Write leveling:
  tCK equivalent taps: 476
  Cmd/Clk scan (0-238)
  |0000  |00000  |00000  |00000| best: 0
  Setting Cmd/Clk delay to 0 taps.
  Data scan:
  m0: |1111111111100000000000000| delay: 00
  m1: |1111111111000000000000000| delay: 00
  m2: |1111111000000000000011111| delay: -
  m3: |1111100000000000000111111| delay: 291
  m4: |1000000000000001111111111| delay: 231
  m5: |1111100000000000000111111| delay: 300
  m6: |1111111100000000000001111| delay: 00
  m7: |1111111111100000000000001| delay: 00
Write latency calibration:
m0:6 m1:6 m2:6 m3:0 m4:0 m5:0 m6:6 m7:6
Read leveling:
  m0, b00: |00000000000000000000000000000000| delays: -
  m0, b01: |00000000000000000000000000000000| delays: -
  m0, b02: |00000000000000000000000000000000| delays: -
  m0, b03: |00000000000000000000000000000000| delays: -
  m0, b04: |11110000000000000000000000000000| delays: 24+-24
  m0, b05: |00000011111111111100000000000000| delays: 182+-103
  m0, b06: |00000000000000000000111111111111| delays: 413+-98
  m0, b07: |00000000000000000000000000000000| delays: -
  best: m0, b05 delays: 181+-102
  m1, b00: |00000000000000000000000000000000| delays: -
  m1, b01: |00000000000000000000000000000000| delays: -
  m1, b02: |00000000000000000000000000000000| delays: -
  m1, b03: |00000000000000000000000000000000| delays: -
  m1, b04: |11111110000000000000000000000000| delays: 53+-53
  m1, b05: |00000000011111111111110000000000| delays: 244+-102
  m1, b06: |00000000000000000000000011111111| delays: 444+-68
  m1, b07: |00000000000000000000000000000000| delays: -
  best: m1, b05 delays: 242+-102
  m2, b00: |00000000000000000000000000000000| delays: -
  m2, b01: |00000000000000000000000000000000| delays: -
  m2, b02: |00000000000000000000000000000000| delays: -
  m2, b03: |00000000000000000000000000000000| delays: -
[... 43 lines cut ...]
  m7, b02: |00000000000000000000000000000000| delays: -
  m7, b03: |00000000000000000000000000000000| delays: -
  m7, b04: |11111111000000000000000000000000| delays: 60+-60
  m7, b05: |00000000001111111111111000000000| delays: 253+-100
  m7, b06: |00000000000000000000000001111111| delays: 449+-62
  m7, b07: |00000000000000000000000000000000| delays: -
  best: m7, b05 delays: 252+-102
Switching SDRAM to hardware control.
Memtest at 0x40000000 (2.0MiB)...
  Write: 0x40000000-0x40200000 2.0MiB
   Read: 0x40000000-0x40200000 2.0MiB
Memtest OK
Memspeed at 0x40000000 (Sequential, 2.0MiB)...
  Write speed: 190.9MiB/s
   Read speed: 211.5MiB/s

--============== Boot ==================--
Booting from serial...
Press Q or ESC to abort boot completely.
sL5DdSMmkekro
Timeout
No boot medium found

--============= Console ================--
```

vietthanh85, 2023-02-12:
@smosanu Sorry for late reply. I've tried `cmd_latency      = 1` but mem access was still error.
BTW, does anyone have success with AU250? 

According to these xml files, AU200 and AU250 have same pins.
https://github.com/Xilinx/open-nic-shell/blob/main/board_files/Xilinx/au250/1.3/part0_pins.xml
https://github.com/Xilinx/open-nic-shell/blob/main/board_files/Xilinx/au200/1.3/part0_pins.xml

According to this user guide, they used same memory type MTA18ASF2G72PZ.
https://www.mouser.com/pdfDocs/u200-u250-UserGuide.pdf
[image]
So, I think the difference is only the FPGA part.

vietthanh85, 2023-02-13:
@smosanu I was succesful on AU200.

I downloaded design constraints (XDC) file [alveo-u200-xdc_20210909.zip].(https://www.xilinx.com/member/forms/download/xef.html?filename=alveo-u200-xdc_20210909.zip)
In `alveo-u200-xdc_20210909.zip/alveo-u200-xdc.xdc`, the only differences between AU200 and AU250 is shown here:
```
#   AU200/250 - Master XDC
#
#       FPGA PIN reference are with respect to the U200 FPGA Bank naming
#            The FPGA A200 and A250 are pin for pin compatible devices.
#               +----------------+---------------+---------------+---------------+
#               | A250 Bank      | A200 Bank     | Usage         | Voltage       |
#               +----------------+---------------+---------------+---------------+ 
#               | Bank 61,62,63  | Bank 40,41,42 | DDR4 C0 Int.  | 1.2V          |
#               +----------------+---------------+---------------+---------------+ 
#               | Bank 65,66,67  | Bank 65,66,67 | DDR4 C1 Int.  | 1.2V          |
#               +----------------+---------------+---------------+---------------+ 
#               | Bank 69,70,71  | Bank 46,47,48 | DDR4 C2 Int.  | 1.2V          |
#               +----------------+---------------+---------------+---------------+ 
#               | Bank 72,73,74  | Bank 70,71,72 | DDR4 C3 Int.  | 1.2V          |
#               +----------------+---------------+---------------+---------------+ 
```
So I modified AU200 platform file as follows:
https://github.com/litex-hub/litex-boards/pull/472/commits/ec7a5c4c0be524fb5f7a608c2ac1a948059eca47

Memtest was OK now:
```
        __   _ __      _  __
       / /  (_) /____ | |/_/
      / /__/ / __/ -_)>  <
     /____/_/\__/\__/_/|_|
   Build your hardware, easily!

 (c) Copyright 2012-2022 Enjoy-Digital
 (c) Copyright 2007-2015 M-Labs

 BIOS built on Feb 13 2023 22:43:05
 BIOS CRC passed (21e13a18)

 LiteX git sha1: 8b14e649

--=============== SoC ==================--
CPU:		VexRiscv @ 125MHz
BUS:		WISHBONE 32-bit @ 4GiB
CSR:		32-bit data
ROM:		128.0KiB
SRAM:		8.0KiB
L2:		8.0KiB
SDRAM:		16.0GiB 64-bit @ 1000MT/s (CL-9 CWL-9)
MAIN-RAM:	1.0GiB

--========== Initialization ============--
Initializing SDRAM @0x40000000...
Switching SDRAM to software control.
Write leveling:
  tCK equivalent taps: 612
  Cmd/Clk scan (0-306)
  |00000  |000000000  |000000000  |000000000| best: 195
  Setting Cmd/Clk delay to 195 taps.
  Data scan:
  m0: |11111111111111111000000| delay: 00
  m1: |11111111111111000000000| delay: 00
  m2: |01111111111111111111000| delay: 08
  m3: |11111111111000000000000| delay: 00
  m4: |11110000000000000000000| delay: -
  m5: |11111110000000000000000| delay: -
  m6: |01111111111111111111000| delay: 01
  m7: |11111111100000000000000| delay: -
Write latency calibration:
m0:6 m1:6 m2:6 m3:6 m4:6 m5:6 m6:6 m7:6 
Read leveling:
[... 69 lines cut ...]
  m7, b06: |00000000000000000000000000000000| delays: -
  m7, b07: |00000000000000000000000000000000| delays: -
  best: m7, b03 delays: 279+-129
Switching SDRAM to hardware control.
Memtest at 0x40000000 (2.0MiB)...
  Write: 0x40000000-0x40200000 2.0MiB     
   Read: 0x40000000-0x40200000 2.0MiB     
Memtest OK
Memspeed at 0x40000000 (Sequential, 2.0MiB)...
  Write speed: 108.8MiB/s
   Read speed: 93.6MiB/s

--============== Boot ==================--
Booting from serial...
Press Q or ESC to abort boot completely.
sL5DdSMmkekro
             Timeout
No boot medium found

--============= Console ================--

litex> 
```

@enjoy-digital @smosanu Could you check this PR https://github.com/litex-hub/litex-boards/pull/472. Thanks a lot.

enjoy-digital, 2023-02-16:
Hi @vietthanh85,

great that you managed to find the issue! Setting` INTERNAL_VREF` on the DRAM banks is indeed required. You PR is fine and has been merged. Thanks!

enjoy-digital, 2023-02-16:
We can probably close this now that it's working.

closed 2023-02-16 by enjoy-digital

smosanu, 2023-02-21:
Congrats @vietthanh85 !!! You did it! And it makes sense that the issue was a subtle one.