enjoy-digital/litex #1585: Create PMP-protected memory regions on the Vexriscv secure CPU.

Ch1psN3t, 2023-02-02:
Hello, 

I need to configure PMP secure memory areas in a binary. I have to do it on the Vexriscv RISC-V processor with the Secure variant which contains the 16 PMP registers. I don't see how to set up protected memory areas in LiteX.

Could you help me, please.

Dolu1990, 2023-02-02:
Hi,

You can either use https://github.com/litex-hub/pythondata-cpu-vexriscv/blob/master/pythondata_cpu_vexriscv/verilog/Makefile#L64

Either generate you own variation in a similar way

Then in litex, you can specify the --variant to select it.

Ch1psN3t, 2023-02-02:
Yes, I use the Vexrsicv Secure variant to get the PMPs. But, I want to configure this with LiteX software. I don't understand how to configure the PMPs registers.

Dolu1990, 2023-02-02:
With litex you can specify which "variant" of the pregenerated netlist you want to use.

You can add the --variant argument to select it.
For VexRiscv non smp in litex, things realy have to be pregenerated, can't generate a custom config on the flight from litex, unless you run a terminal command from the python run time itself to generate the cpu variant

Ch1psN3t, 2023-02-02:
Yes, I agree with you. For having generated the Vexriscv CPU with PMPs. I write: 
_./litex-boards/litex_boards/targets/digilent_arty.py --variant=a7-100 --cpu-type=vexriscv --cpu-variant=secure --build --load_

But, I don't see the pmp control registers in the generated csr.h file. I don't see how to configure the pmp in the csr.h file. 
_litex/soc/software/_ in firmware.

Dolu1990, 2023-02-02:
csr.h generated from litex correspond the memory mapped peripherals, not the RISC-V privileged CSR.
PMP registers are in the RISC-V privileged CSR space, not memory mapped.

Ch1psN3t, 2023-02-06:
Yes, I understand. But, I want to configure the PMPs registers for that you have to be in Machine or Supervisor mode, but I can't do it. I saw the mode in the bios, but it is in user mode. Do you have any idea ?

Dolu1990, 2023-02-06:
>supervisor 

Do you want PMP + MMU or just the PMP ?
(VexRIscv do not support both in the same hardware, but can support machine with pmp + supervisor with no mmu+ user)

Potentialy, you can do --csrPluginConfig linux-minimal  instead of --csrPluginConfig secure but you will also have to foce 
https://github.com/litex-hub/pythondata-cpu-vexriscv/blob/e75700dff2ab9662f3e26dd89ab59a5f6da65687/pythondata_cpu_vexriscv/verilog/src/main/scala/vexriscv/GenCoreDefault.scala#L88  to false

Ch1psN3t, 2023-02-06:
I just want PMP configure. I use the command `./litex-boards/litex_boards/targets/digilent_arty.py --variant=a7-100 --cpu-variant=secure --build --load` for generated my SoC with the PMPs. I want to configure the PMPs in the software.

Dolu1990, 2023-02-06:
So, https://github.com/litex-hub/pythondata-cpu-vexriscv/blob/e75700dff2ab9662f3e26dd89ab59a5f6da65687/pythondata_cpu_vexriscv/verilog/src/main/scala/vexriscv/GenCoreDefault.scala#L88 would need to be extended to have a csrPluginConfig with supervisor but no mmu, and a new variant could be added in https://github.com/litex-hub/pythondata-cpu-vexriscv/blob/e75700dff2ab9662f3e26dd89ab59a5f6da65687/pythondata_cpu_vexriscv/verilog/Makefile and https://github.com/enjoy-digital/litex/blob/a6d9955c9d3065d5c04cd4cba392b2fcb831380f/litex/soc/cores/cpu/vexriscv/core.py#L24

Ch1psN3t, 2023-02-08:
I'm just looking to set up the pmp registers. But, I wonder if I can do it in LiteX's default processor mode, because I saw that it is in USER mode, is that true? 
I want to set the pmp registers to baremetal.

Dolu1990, 2023-02-08:
Ahhhhhh, to write pmp registers, you need to do it in RISC-V C software, not from litex itself.

closed 2023-02-19 by Ch1psN3t