enjoy-digital/litex #1587: Upload to device failed due to data corruption (CRC error)

roby2014, 2023-02-07:
Hi, I was able to upload vexriscv to my Colorlight 5A-75E FPGA, however, if I reboot 2 times, it throws a data corruption error:

```
....
$ >reboot

        __   _ __      _  __
       / /  (_) /____ | |/_/
      / /__/ / __/ -_)>  <
     /____/_/\__/\__/_/|_|
   Build your hardware, easily!

 (c) Copyright 2012-2022 Enjoy-Digital
 (c) Copyright 2007-2015 M-Labs

 BIOS built on Feb  7 2023 17:08:45
 BIOS CRC passed (d05f02be)

 LiteX git sha1: c8343879

--=============== SoC ==================--
CPU:		VexRiscv @ 25MHz
BUS:		WISHBONE 32-bit @ 4GiB
CSR:		32-bit data
ROM:		32.0KiB
SRAM:		8.0KiB
MAIN-RAM:	16.0KiB

--========== Initialization ============--
Memtest at 0x40000000 (16.0KiB)...
  Write: 0x40000000-0x40004000 16.0KiB
   Read: 0x40000000-0x40004000 16.0KiB
Memtest OK
Memspeed at 0x40000000 (Sequential, 16.0KiB)...
  Write speed: 39.7MiB/s
   Read speed: 19.8MiB/s

--============== Boot ==================--
Booting from serial...
Press Q or ESC to abort boot completely.
sL5DdSMmkekro
[LITEX-TERM] Received firmware download request from the device.
upload
[LITEX-TERM] Uploading firmware.bin to 0x40000000 (5496 bytes)...
[LITEX-TERM] Upload calibration... (inter-frame: 10.00us, length: 64)
[LITEX-TERM] Upload to device failed due to data corruption (CRC error)
```

Is this a common error?

Repository I am using: https://github.com/roby2014/risc-v-colorlight-5a-75e
Programmer I am using: FT232H

enjoy-digital, 2023-02-08:
Hi @roby2014,

that's not really a common error but I'll do a test (probably with the same hardware).

Ei03, 2023-10-27:
Hi @roby2014 @enjoy-digital 
I wanna know did you guys sloved this error? if yes, plz tell me how, thanks

THLN33, 2026-01-08:
Hello, I have the same error. I'm working on Windows 10 WSL2. The serial driver is FTDI 2.12.36.20. My board is a Tang Primer 20K. And sometimes it works, I don't understand why it doesn't work every time.
I'm using the Python script `litex_term.py` under Windows and I've also tried using USBIPD-WIN, but I get the same result. See the console output below:

			__   _ __      _  __
		   / /  (_) /____ | |/_/
		  / /__/ / __/ -_)>  <
		 /____/_/\__/\__/_/|_|
	   Build your hardware, easily!

	 (c) Copyright 2012-2025 Enjoy-Digital
	 (c) Copyright 2007-2015 M-Labs

	 BIOS built on Jan  7 2026 15:15:44
	 BIOS CRC passed (19a518fb)

	 LiteX git sha1: 67932fdfb

	--=============== SoC ==================--
	CPU:            PicoRV32 @ 48MHz
	BUS:            wishbone 32-bit @ 4GiB
	CSR:            32-bit data big ordering
	ROM:            128.0KiB
	SRAM:           8.0KiB
	L2:             8.0KiB
	SDRAM:          128.0MiB 16-bit @ 192MT/s (CL-6 CWL-5)
	MAIN-RAM:       128.0MiB

	--========== Initialization ============--

	Initializing SDRAM @0x40000000...
	Switching SDRAM to software control.
	Read leveling:
	  m0, b00: |00000000| delays: -
	  m0, b01: |00000000| delays: -
	  m0, b02: |01100000| delays: 01+-00
	  m0, b03: |00000000| delays: -
	  best: m0, b02 delays: 01+-00
	  m1, b00: |00000000| delays: -
	  m1, b01: |00000000| delays: -
	  m1, b02: |01100000| delays: 01+-00
	  m1, b03: |00000000| delays: -
	  best: m1, b02 delays: 01+-00
	Switching SDRAM to hardware control.
	Memtest at 0x40000000 (2.0MiB)...
	  Write: 0x40000000-0x40200000 2.0MiB
	   Read: 0x40000000-0x40200000 2.0MiB
	Memtest OK
	Memspeed at 0x40000000 (Sequential, 2.0MiB)...
	  Write speed: 7.7MiB/s
	   Read speed: 11.7MiB/s

	--============== Boot ==================--
	Booting from serial...
	Press Q or ESC to abort boot completely.
	sL5DdSMmkekro
	[LITEX-TERM] Received firmware download request from the device.
	[LITEX-TERM] Uploading demo.bin to 0x40000000 (8332 bytes)...
	[LITEX-TERM] Upload calibration... (inter-frame: 10.00us, length: 64)
	[LITEX-TERM] Upload to device failed due to data corruption (CRC error)