enjoy-digital/litex #1588: Antmicro datacenter target on litex-boards fails in initialization

tr-rocks, 2023-02-07:
Simply running the target file for the Antmicro data center ddr4 test board as is from the litex-boards repo to write/load the bitstream and run the bios fails in initialization due to both bus errors and data errors. (The output is copied below.)

```
        __   _ __      _  __
       / /  (_) /____ | |/_/
      / /__/ / __/ -_)>  <
     /____/_/\__/\__/_/|_|
   Build your hardware, easily!

 (c) Copyright 2012-2022 Enjoy-Digital
 (c) Copyright 2007-2015 M-Labs

 BIOS built on Feb  7 2023 11:27:12
 BIOS CRC passed (7902f33f)

 LiteX git sha1: a56a1787

--=============== SoC ==================--
CPU:		VexRiscv @ 100MHz
BUS:		WISHBONE 32-bit @ 4GiB
CSR:		32-bit data
ROM:		128.0KiB
SRAM:		8.0KiB
L2:		8.0KiB
SDRAM:		16.0GiB 64-bit @ 800MT/s (CL-9 CWL-9)
MAIN-RAM:	1.0GiB

--========== Initialization ============--
Initializing SDRAM @0x40000000...
Switching SDRAM to software control.
Read leveling:
  m0, b00: |00000000000000000000000000000000| delays: -
  m0, b01: |00000000000000000000000000000000| delays: -
  m0, b02: |00000000000000000000000000000000| delays: -
  m0, b03: |00000000000000000000000000000000| delays: -
  m0, b04: |00000000000000000000000000000000| delays: -
  m0, b05: |00000000000000000000000000000000| delays: -
  m0, b06: |00000000000000000000000000000000| delays: -
  m0, b07: |00000000000000000000000000000000| delays: -
  best: m0, b02 delays: -
  m1, b00: |00000000000000000000000000000000| delays: -
  m1, b01: |00000000000000000000000000000000| delays: -
  m1, b02: |00000000000000000000000000000000| delays: -
  m1, b03: |00000000000000000000000000000000| delays: -
  m1, b04: |00000000000000000000000000000000| delays: -
  m1, b05: |00000000000000000000000000000000| delays: -
  m1, b06: |00000000000000000000000000000000| delays: -
  m1, b07: |00000000000000000000000000000000| delays: -
  best: m1, b02 delays: -
  m2, b00: |00000000000000000000000000000000| delays: -
  m2, b01: |00000000000000000000000000000000| delays: -
  m2, b02: |00000000000000000000000000000000| delays: -
  m2, b03: |00000000000000000000000000000000| delays: -
  m2, b04: |00000000000000000000000000000000| delays: -
  m2, b05: |00000000000000000000000000000000| delays: -
  m2, b06: |00000000000000000000000000000000| delays: -
  m2, b07: |00000000000000000000000000000000| delays: -
  best: m2, b02 delays: -
  m3, b00: |00000000000000000000000000000000| delays: -
  m3, b01: |00000000000000000000000000000000| delays: -
  m3, b02: |00000000000000000000000000000000| delays: -
  m3, b03: |00000000000000000000000000000000| delays: -
  m3, b04: |00000000000000000000000000000000| delays: -
  m3, b05: |00000000000000000000000000000000| delays: -
  m3, b06: |00000000000000000000000000000000| delays: -
  m3, b07: |00000000000000000000000000000000| delays: -
  best: m3, b02 delays: -
  m4, b00: |00000000000000000000000000000000| delays: -
  m4, b01: |00000000000000000000000000000000| delays: -
  m4, b02: |00000000000000000000000000000000| delays: -
  m4, b03: |00000000000000000000000000000000| delays: -
  m4, b04: |00000000000000000000000000000000| delays: -
  m4, b05: |00000000000000000000000000000000| delays: -
  m4, b06: |00000000000000000000000000000000| delays: -
  m4, b07: |00000000000000000000000000000000| delays: -
  best: m4, b02 delays: -
  m5, b00: |00000000000000000000000000000000| delays: -
  m5, b01: |00000000000000000000000000000000| delays: -
  m5, b02: |00000000000000000000000000000000| delays: -
  m5, b03: |00000000000000000000000000000000| delays: -
  m5, b04: |00000000000000000000000000000000| delays: -
  m5, b05: |00000000000000000000000000000000| delays: -
  m5, b06: |00000000000000000000000000000000| delays: -
  m5, b07: |00000000000000000000000000000000| delays: -
  best: m5, b02 delays: -
  m6, b00: |00000000000000000000000000000000| delays: -
  m6, b01: |00000000000000000000000000000000| delays: -
  m6, b02: |00000000000000000000000000000000| delays: -
  m6, b03: |00000000000000000000000000000000| delays: -
  m6, b04: |00000000000000000000000000000000| delays: -
  m6, b05: |00000000000000000000000000000000| delays: -
  m6, b06: |00000000000000000000000000000000| delays: -
  m6, b07: |00000000000000000000000000000000| delays: -
  best: m6, b01 delays: -
  m7, b00: |00000000000000000000000000000000| delays: -
  m7, b01: |00000000000000000000000000000000| delays: -
  m7, b02: |00000000000000000000000000000000| delays: -
  m7, b03: |00000000000000000000000000000000| delays: -
  m7, b04: |00000000000000000000000000000000| delays: -
  m7, b05: |00000000000000000000000000000000| delays: -
  m7, b06: |00000000000000000000000000000000| delays: -
  m7, b07: |00000000000000000000000000000000| delays: -
  best: m7, b02 delays: -
  m8, b00: |00000000000000000000000000000000| delays: -
  m8, b01: |00000000000000000000000000000000| delays: -
  m8, b02: |00000000000000000000000000000000| delays: -
  m8, b03: |00000000000000000000000000000000| delays: -
  m8, b04: |00000000000000000000000000000000| delays: -
  m8, b05: |00000000000000000000000000000000| delays: -
  m8, b06: |00000000000000000000000000000000| delays: -
  m8, b07: |00000000000000000000000000000000| delays: -
  best: m8, b01 delays: -
  m9, b00: |00000000000000000000000000000000| delays: -
  m9, b01: |00000000000000000000000000000000| delays: -
  m9, b02: |00000000000000000000000000000000| delays: -
  m9, b03: |00000000000000000000000000000000| delays: -
  m9, b04: |00000000000000000000000000000000| delays: -
  m9, b05: |00000000000000000000000000000000| delays: -
  m9, b06: |00000000000000000000000000000000| delays: -
  m9, b07: |00000000000000000000000000000000| delays: -
  best: m9, b01 delays: -
  m10, b00: |00000000000000000000000000000000| delays: -
  m10, b01: |00000000000000000000000000000000| delays: -
  m10, b02: |00000000000000000000000000000000| delays: -
  m10, b03: |00000000000000000000000000000000| delays: -
  m10, b04: |00000000000000000000000000000000| delays: -
  m10, b05: |00000000000000000000000000000000| delays: -
  m10, b06: |00000000000000000000000000000000| delays: -
  m10, b07: |00000000000000000000000000000000| delays: -
  best: m10, b01 delays: -
  m11, b00: |00000000000000000000000000000000| delays: -
  m11, b01: |00000000000000000000000000000000| delays: -
  m11, b02: |00000000000000000000000000000000| delays: -
  m11, b03: |00000000000000000000000000000000| delays: -
  m11, b04: |00000000000000000000000000000000| delays: -
  m11, b05: |00000000000000000000000000000000| delays: -
  m11, b06: |00000000000000000000000000000000| delays: -
  m11, b07: |00000000000000000000000000000000| delays: -
  best: m11, b01 delays: -
  m12, b00: |00000000000000000000000000000000| delays: -
  m12, b01: |00000000000000000000000000000000| delays: -
  m12, b02: |00000000000000000000000000000000| delays: -
  m12, b03: |00000000000000000000000000000000| delays: -
  m12, b04: |00000000000000000000000000000000| delays: -
  m12, b05: |00000000000000000000000000000000| delays: -
  m12, b06: |00000000000000000000000000000000| delays: -
  m12, b07: |00000000000000000000000000000000| delays: -
  best: m12, b02 delays: -
  m13, b00: |00000000000000000000000000000000| delays: -
  m13, b01: |00000000000000000000000000000000| delays: -
  m13, b02: |00000000000000000000000000000000| delays: -
  m13, b03: |00000000000000000000000000000000| delays: -
  m13, b04: |00000000000000000000000000000000| delays: -
  m13, b05: |00000000000000000000000000000000| delays: -
  m13, b06: |00000000000000000000000000000000| delays: -
  m13, b07: |00000000000000000000000000000000| delays: -
  best: m13, b02 delays: -
  m14, b00: |00000000000000000000000000000000| delays: -
  m14, b01: |00000000000000000000000000000000| delays: -
  m14, b02: |00000000000000000000000000000000| delays: -
  m14, b03: |00000000000000000000000000000000| delays: -
  m14, b04: |00000000000000000000000000000000| delays: -
  m14, b05: |00000000000000000000000000000000| delays: -
  m14, b06: |00000000000000000000000000000000| delays: -
  m14, b07: |00000000000000000000000000000000| delays: -
  best: m14, b02 delays: -
  m15, b00: |00000000000000000000000000000000| delays: -
  m15, b01: |00000000000000000000000000000000| delays: -
  m15, b02: |00000000000000000000000000000000| delays: -
  m15, b03: |00000000000000000000000000000000| delays: -
  m15, b04: |00000000000000000000000000000000| delays: -
  m15, b05: |00000000000000000000000000000000| delays: -
  m15, b06: |00000000000000000000000000000000| delays: -
  m15, b07: |00000000000000000000000000000000| delays: -
  best: m15, b02 delays: -
Switching SDRAM to hardware control.
Memtest at 0x40000000 (2.0MiB)...
  Write: 0x40000000-0x40200000 2.0MiB     
   Read: 0x40000000-0x40200000 2.0MiB     
  bus errors:  96/256
  addr errors: 0/8192
  data errors: 524288/524288
Memtest KO
Memory initialization failed

--============= Console ================--

litex>
```

enjoy-digital, 2023-02-08:
Hi @tr-rocks,

sorry I don't have this hardware, but I would recommend first testing with a bitstream that has already been tested successfully with this board. Maybe Antmicro is sharing a validated bistream, if not, maybe @kgugala could help and redirect you to the right persons.

enjoy-digital, 2024-01-12:
Closing since no feedback or hardware available for test. @tr-rocks Please provide more info or feedback if you still need help.

closed 2024-01-12 by enjoy-digital