enjoy-digital/litex #1589: litex_sim: command not found

CarrolXC, 2023-02-07:
I got an message saing "litex_sim: command not found" when I typed in litex_sim --cpu -type=vexriscv.
Do you have ideas or experience on that? All the previous steps are done successfully.
Appreciate it!

CarrolXC, 2023-02-07:
> could you try whereis litex_sim ?
> have you installed LiteX into as user (ie *~/.local/*) or in your system?
> If it's *~/.local* could you check PATH env var to check if it contains
> *~/.local/bin* ?

Yes, it is in my user folder: /home/gengar/.local/bin/litex_sim
It was fixed after I added the path.
Sincerely appreciate your suggestions!
BTW do you have any ideas on what caused this?I did everything following
the instructions.

trabucayre, 2023-02-07:
Great!
Could you point a link to instructions?
*~/.local/bin* isn't by default in your `PATH` so you have to add something like `export PATH=$PATH:$HOME/.local/bin` into your 
*~/.bashrc* to have this path in all your terminals. !!!WARNING!!! you have to add $PATH before or after: this command overrides $PATH variable.
using the `export` command manually into one terminal is limited to this terminal (a new/other terminal will not be affected and if you close this terminal you lost this update).

CarrolXC, 2023-02-07:
Here is the instructions I was looking at:
https://github.com/enjoy-digital/litex/wiki/Installation
Many thanks again for your help.
Best regards,

trabucayre, 2023-02-07:
it's explained [here](https://github.com/enjoy-digital/litex/blob/master/litex_setup.py#L290) but yes it's not mentionned this command is not persistent: wiki and script must be updated to explain this.

CarrolXC, 2023-02-07:
Cool! Appreciate your effort on my questions.
I will try to figure it out.
Best,
Xi