enjoy-digital/litex #1590: --cpu-type=mor1kx brakes efinix_trion_t120_bga576_dev_kit target

fjullien, 2023-02-08:
This builds without problem:

`python3 ./efinix_trion_t120_bga576_dev_kit.py --build`

However, with `--cpu-type=mor1kx` I get:

```
-- Analyzing Verilog file '/home/franck/travail/litex_install/pythondata-cpu-mor1kx/pythondata_cpu_mor1kx/verilog/rtl/verilog/pfpu32/pfpu32_rnd.v' (VERI-1482)
/home/franck/travail/litex_install/pythondata-cpu-mor1kx/pythondata_cpu_mor1kx/verilog/rtl/verilog/pfpu32/pfpu32_rnd.v(38): ERROR: cannot open include file 'mor1kx-defines.v' (VERI-1245)
/home/franck/travail/litex_install/pythondata-cpu-mor1kx/pythondata_cpu_mor1kx/verilog/rtl/verilog/pfpu32/pfpu32_rnd.v(420): ERROR: module 'pfpu32_rnd' is ignored due to previous errors (VERI-1072)
-- Verilog file '/home/franck/travail/litex_install/pythondata-cpu-mor1kx/pythondata_cpu_mor1kx/verilog/rtl/verilog/pfpu32/pfpu32_rnd.v' ignored due to errors (VERI-1483)
INFO: Analysis took 0.231896 seconds.
INFO: 	Analysis took 0.16 seconds (approximately) in total CPU time.
INFO: Analysis virtual memory usage: begin = 188.42 MB, end = 192.496 MB, delta = 4.076 MB
INFO: Analysis resident set memory usage: begin = 72.132 MB, end = 78.944 MB, delta = 6.812 MB
```

fjullien, 2023-02-08:
At some point there should be an include path in the projetc xml file:
```
    <efx:synthesis tool_name="efx_map">
        ...
        <efx:param name="include" value="../../../../litex_install/pythondata-cpu-mor1kx/pythondata_cpu_mor1kx/verilog/rtl/verilog" value_type="e_string"/>
    </efx:synthesis>
```

enjoy-digital, 2023-02-08:
Hi @fjullien,

thanks for reporting, this has probably not been tested yet :) Is it working with the added include path?

fjullien, 2023-02-09:
Yes, it does work with the added path.

enjoy-digital, 2023-02-09:
@fjullien: OK thanks, I'll have a look at this to integrate this to the Efinix build backend.