enjoy-digital/litex #1591: --cpu-type=cv32e40s is missing on supported CPU list

weik56100, 2023-02-08:
Hello to all,

I am trying to implement the cv32e40s in Litex.
The Litex hub "pythondata-cpu-cv32e40s" has already been done last year:
https://github.com/litex-hub/pythondata-cpu-cv32e40s

I followed this little guide to try to do it manually based on the cv32e40p and cv32e41p:
http://pcotret.gitlab.io/blog/processor_in_litex/

But shouldn't it already appear in the list just by installing Litex without anything else?
Because the cv32e40p and cv32e41p are present.
[image]

enjoy-digital, 2023-02-08:
Hello @weik56100,

It seems cv32e40s support has not been integrated in LiteX. We would need to add it to https://github.com/enjoy-digital/litex/tree/master/litex/soc/cores/cpu. Have you create a wrapper based on the other cv32 variants? Happy to review and merge if someone create the PR.

pcotret, 2023-09-13:
Oh, I just saw that this issue is still open. I'll be happy to help in a short time cc @pcotret (ping myself to remember this ^^)

pcotret, 2024-04-02:
Came back to this issue as I was interested in other OpenHwGroup cores.
There are several errors related to a `uvm_error` directive. For instance :

```
  134 |         `uvm_error("Alignment buffer SVA",
      |         ^~~~~~~~~~
        /opt/litex_install/pythondata-cpu-cv32e40s/pythondata_cpu_cv32e40s/system_verilog/bhv/cv32e40s_wrapper.sv:17:1: ... note: In file included from 'cv32e40s_wrapper.sv'
%Error: /opt/litex_install/pythondata-cpu-cv32e40s/pythondata_cpu_cv32e40s/system_verilog/sva/cv32e40s_alignment_buffer_sva.sv:146:9: Define or directive not defined: '`uvm_error'
                                                                                                                                    : ... Suggested alternative: '`error'
```

I'll try to solve it later this week. At least, for an initial support (without testing 40S specific modules).

pcotret, 2024-04-02:
https://github.com/openhwgroup/core-v-verif/issues/2362

A few weeks ago, I was blocked while trying to emulate the CV32E40S with their ` core-v-verif` verification environment.

I assume there's still some work to make it run with LiteX, even for a simple `litex_sim` test.