enjoy-digital/litex #1593: Add some examples for FPU usage

suarezvictor, 2023-02-08:
Here it's a demo project to test FPU of different soft processors, and a patch for easier use of naxriscv, plus a way to reproduce an issue with NaxRiscv (yet unsolved)

https://github.com/suarezvictor/litex_fpu_test/

suarezvictor, 2023-02-08:
Naxriscv is unable to run a binary generated for that architecture while the same can be run verbatim by vexriscv_smp, see the added assemble files https://github.com/suarezvictor/litex_fpu_test/tree/main/output

enjoy-digital, 2023-02-08:
Thanks @suarezvictor, I'll have a look at the examples. We could add a link to your repo in the wiki.

Dolu1990, 2023-02-08:
Ahhhh, there is something i just remember now, in order to enable the FPU, you need to set the mstatus.FS bits to something else than 00.
See https://github.com/riscv/riscv-isa-manual/releases/download/Priv-v1.12/riscv-privileged-20211203.pdf
- Table 3.3: Encoding of FS[1:0], VS[1:0], and XS[1:0] status fields.
- Machine Status Registers (mstatus and mstatush)

In VexRIscv, for backward compatibility reasons, that FS field is initialized to 01 at reset.
In NaxRiscv i followed the RISC-V spec more closely, and initialize it to 00 at reset

When at 00, if you do anything with the FPU you will get a illegal instruction trap, on purpose :)

Here is an example to initialize it : 
https://github.com/SpinalHDL/NaxSoftware/blob/3c1a611b3fa65974d4a93b760918cb9e94748434/baremetal/fpu_test/src/crt.S#L277

Dolu1990, 2023-02-09:
```c
#define csr_set(csr, val)                    \
({                                \
    unsigned long __v = (unsigned long)(val);        \
    __asm__ __volatile__ ("csrs " #csr ", %0"        \
                  : : "rK" (__v));            \
})

...

csr_set(mstatus, (1 << 13));
```

Should be enough in C ^^

suarezvictor, 2023-02-09:
Thanks Charles. I tested configuring the CSR as proposed and unfortunately, it doesn't work on Naxriscv. Setting the CSR seems to cause no harm in Vexriscv so the same binary works with the later CPU.

Here is the C code and assembly instructions with the CSR configuration added: 

```

#ifdef __NaxRiscv__
#define csr_set(csr, val)                    \
({                                \
    unsigned long __v = (unsigned long)(val);        \
    __asm__ __volatile__ ("csrs " #csr ", %0"        \
                  : : "rK" (__v));            \
})
#endif

int main(void)
{
#ifdef CONFIG_CPU_HAS_INTERRUPT
	irq_setmask(0);
	irq_setie(1);
#endif
	uart_init();

#ifdef __NaxRiscv__
    csr_set(mstatus, (1 << 13));
#endif
	float_test();
	return 0;
}
```
Full code is here: https://github.com/suarezvictor/litex_fpu_test/blob/main/main.c#L19

Resulting assembly:
```
000001bc <main>:
 1bc:   ff010113                addi    sp,sp,-16
 1c0:   00112623                sw      ra,12(sp)
 1c4:   11c000ef                jal     ra,2e0 <uart_init>
 1c8:   000027b7                lui     a5,0x2
 1cc:   3007a073                csrs    mstatus,a5
 1d0:   f55ff0ef                jal     ra,124 <float_test>
 1d4:   00c12083                lw      ra,12(sp)
 1d8:   00000513                li      a0,0
 1dc:   01010113                addi    sp,sp,16
 1e0:   00008067                ret

```

Dolu1990, 2023-02-10:
Hi,

I just tested, found one issue fixed by : 
https://github.com/enjoy-digital/litex/commit/92032b446c4f7d8b45e7263269cc2f28ef2d1821

Then on my side it worked, got a : 

```

0=0
1=0
2=1
3=1
4=2
``` 

and checked the elf file, there is FPU stuff in it :)

Did you removed the generated verilog files from litex/pythondata-cpu-naxriscv/pythondata_cpu_naxriscv/verilog ?
NaxRiscvLitex_*

The NaxRiscv stuff is caching previously generated netlist to avoid work, but when you change something you realy need to clean that folder.

let's me know how it goes :D

Charles

suarezvictor, 2023-02-10:
I tested it and the FPU now show good signs of being alive :), and checked that setting the CSR is required.
We should decide if we expect the user to enable the FPU by hand, or if it would be better to set the bit in the .S initialization file, so it's enabled by default like other CPUs.

There's the new issue that now a value get miscalculated
```
0=0
1=0
2=1
3=/
4=2
```

Please consider that I reinstalled pythondata-cpu-naxriscv module directly from litex-hub github repo (I haven't found it on default pip repo)
I also took the opportunity to improve a bit the example repo with this:

1. Deleted the comment that Naxriscv doesn't work
2. Simplified the code using `csrs()` function/macro provided by LiteX
3. Added instructions on how to install pythondata-cpu-naxriscv module with pip
4. Added instructions on how to delete previously generated Verilog files, if generation parameters change (as Charles suggested)

Anyways before publishing this in the wiki, I think we should dig into why a value is miscalculated

Dolu1990, 2023-02-10:
> We should decide if we expect the user to enable the FPU by hand, or if it would be better to set the bit in the .S initialization file, so it's enabled by default like other CPUs.

Yes, i think that's the right thing to do.

> 3=/

That's is reaaaaly weird as on my side it was ok.
Can you send me your demo.bin, demo.elf, and the wave file of your simulation ? --trace --trace-fst
(fst file take much less space :))

Dolu1990, 2023-02-10:
Pushed the crt0.S changes to enable the FPU : 
https://github.com/enjoy-digital/litex/commit/34422d2d912178a6d8f979031320939c0f24fc6e

suarezvictor, 2023-02-10:
Files sent my email

> That's is reaaaaly weird as on my side it was ok. Can you send me your demo.bin, demo.elf, and the wave file of your simulation ? --trace --trace-fst (fst file take much less space :))

Dolu1990, 2023-02-10:
Nice thanks ^^

So, the issue was that NaxRiscv wasn't applying f32 into f64 nan-boxing for memory loads in the peripheral region (onchip-ram). (and instead was using the main memory region pipeline status)
Got it fixed with : 
https://github.com/SpinalHDL/NaxRiscv/commit/57e3bf59ecca75dc394c41f6d5d7a662ea1ddc93
I also pushed the fix the the litex repo.

Thanks :D

suarezvictor, 2023-02-11:
Good to know that a bug was found and fixed!
The output is now as expected
```
0=0
1=0
2=1
3=1
4=2
```
I had to manually update the patched file with commit https://github.com/SpinalHDL/NaxRiscv/commit/57e3bf59ecca75dc394c41f6d5d7a662ea1ddc93 (as Charles referenced), while the LiteX sources expect another version: https://github.com/SpinalHDL/NaxRiscv/commit/57e3bf59 on https://github.com/enjoy-digital/litex/blob/master/litex/soc/cores/cpu/naxriscv/core.py#L241. So that line should be updated. As a workaround I used `--update-repo no` flag in the _litex_sim_ command
Let me know if the LiteX sources will be updated otherwise I'll document the workaround in the example repo.

Dolu1990, 2023-02-11:
Hmm isn't 57e3bf5 the same as 57e3bf59 ?

Dolu1990, 2023-02-17:
@suarezvictor Note, that NaxRiscv will do out of order execution load/store only on the memory mapped in litedram.
The on-chip ram (accessed via axi-lite / wishbone) is considered as IO memory (in order / low performance, as in your simulation)

suarezvictor, 2023-02-17:
I didn't hace time to continue testing this, bit ir seems right (I trust your tests). What sinulation so you refer? I wasn't brenchmarking this CPU yet

Dolu1990, 2023-02-17:
Hooo right, that was my sim, using the litex simulation ^^
I mean, the binary you provided me used the onchip ram i think.