enjoy-digital/litex #1601: Couldn't build for nexys4ddr with cva6 cpu

Thecfvgx, 2023-02-16:
Hi, 

We're trying to build cva6 for the nexys4ddr board, but got the errors pasted below.

For a bit of context : 
 - building vexriscv for nexys4ddr worked just fine, and we didn't have any issue loading it either and running basic programs on it.
 - building cva6 for a different board returned the exact same errors
 
``` bash
$ ./digilent_nexys4ddr.py --cpu-type=cva6 --build
```
The errors we got : 
``` bash 
ERROR: [Synth 8-439] module 'addr_decode' not found [/home/thecfvgx/.local/lib/python3.10/site-packages/pythondata_cpu_cva6/system_verilog/corev_apu/axi/src/axi_xbar.sv:74]
	Parameter NoIndices bound to: 4 - type: integer 
	Parameter NoRules bound to: 2 - type: integer 
	Parameter addr_t bound to: 0'b 
	Parameter rule_t bound to: 0'b 
ERROR: [Synth 8-196] conditional expression could not be resolved to a constant [/home/thecfvgx/.local/lib/python3.10/site-packages/pythondata_cpu_cva6/system_verilog/corev_apu/axi/src/axi_xbar.sv:63]
ERROR: [Synth 8-6156] failed synthesizing module 'axi_xbar' [/home/thecfvgx/.local/lib/python3.10/site-packages/pythondata_cpu_cva6/system_verilog/corev_apu/axi/src/axi_xbar.sv:18]
ERROR: [Synth 8-6156] failed synthesizing module 'axi_xbar_intf' [/home/thecfvgx/.local/lib/python3.10/site-packages/pythondata_cpu_cva6/system_verilog/corev_apu/axi/src/axi_xbar.sv:242]
ERROR: [Synth 8-6156] failed synthesizing module 'cva6_wrapper' [/home/thecfvgx/litex_and_more/litex/litex/soc/cores/cpu/cva6/cva6_wrapper/cva6_wrapper.sv:8]
ERROR: [Synth 8-6156] failed synthesizing module 'digilent_nexys4ddr' [/home/thecfvgx/litex_and_more/litex-boards/litex_boards/targets/build/digilent_nexys4ddr/gateware/digilent_nexys4ddr.v:21]
```

Thanks in advance for any help ^^

lapd-soc-one, 2023-02-17:
Hi @Thecfvgx 
I have to add a few missing system verilog files to get rid of these errors
```patch
diff --git a/pythondata_cpu_cva6/system_verilog/core/Flist.cv64a6_imafdc_sv39 b/pythondata_cpu_cva6/system_verilog/core/Flist.cv64a6_imafdc_sv39
old mode 100644
new mode 100755
index 72a038b..671a2b9
--- a/pythondata_cpu_cva6/system_verilog/core/Flist.cv64a6_imafdc_sv39
+++ b/pythondata_cpu_cva6/system_verilog/core/Flist.cv64a6_imafdc_sv39
@@ -67,6 +67,11 @@ ${CVA6_REPO_DIR}/common/submodules/common_cells/src/shift_reg.sv
 ${CVA6_REPO_DIR}/common/submodules/common_cells/src/unread.sv
 ${CVA6_REPO_DIR}/common/submodules/common_cells/src/popcount.sv
 ${CVA6_REPO_DIR}/common/submodules/common_cells/src/exp_backoff.sv
+${CVA6_REPO_DIR}/common/submodules/common_cells/src/addr_decode.sv
+${CVA6_REPO_DIR}/common/submodules/common_cells/src/spill_register_flushable.sv 
+${CVA6_REPO_DIR}/common/submodules/common_cells/src/spill_register.sv 
+${CVA6_REPO_DIR}/common/submodules/common_cells/src/stream_register.sv 
+${CVA6_REPO_DIR}/common/submodules/common_cells/src/cdc_2phase.sv 
 
 // Common Cells for example coprocessor
 ${CVA6_REPO_DIR}/common/submodules/common_cells/src/counter.sv
```

lapd-soc-one, 2023-02-17:
However,  I got another issues
```sh
CRITICAL WARNING: [Project 1-560] Could not resolve non-primitive black box cell 'tc_sram_wrapper' instantiated as 'cva6_wrapper/i_ariane/i_cva6/i_cache_subsystem/i_cva6_icache/gen_sram[0].data_sram/gen_cut[0].gen_mem.i_tc_sram_wrapper'. 36 instances of this cell are unresolved black boxes. [/opt/litex-essential/pythondata-cpu-cva6/pythondata_cpu_cva6/system_verilog/common/local/util/sram.sv:107]
Netlist sorting complete. Time (s): cpu = 00:00:00.02 ; elapsed = 00:00:00.03 . Memory (MB): peak = 3643.551 ; gain = 0.000 ; free physical = 10272 ; free virtual = 114800
INFO: [Project 1-111] Unisim Transformation Summary:
  A total of 55 instances were transformed.
  IOBUF => IOBUF (IBUF, OBUFT): 16 instances
  IOBUFDS => IOBUFDS (IBUFDS, INV, OBUFTDS(x2)): 2 instances
  OBUFDS => OBUFDS_DUAL_BUF (INV, OBUFDS(x2)): 1 instance 
  RAM32M => RAM32M (RAMD32(x6), RAMS32(x2)): 36 instances

Synth Design complete, checksum: 83050251
INFO: [Common 17-83] Releasing license: Synthesis
751 Infos, 553 Warnings, 4 Critical Warnings and 0 Errors encountered.
synth_design completed successfully
synth_design: Time (s): cpu = 00:08:03 ; elapsed = 00:08:08 . Memory (MB): peak = 3643.551 ; gain = 1087.695 ; free physical = 10553 ; free virtual = 115081
# report_timing_summary -file digilent_nexys4ddr_timing_synth.rpt
INFO: [Timing 38-35] Done setting XDC timing constraints.
INFO: [Timing 38-91] UpdateTimingParams: Speed grade: -1, Delay Type: min_max.
INFO: [Timing 38-191] Multithreading enabled for timing update using a maximum of 8 CPUs
report_timing_summary: Time (s): cpu = 00:00:33 ; elapsed = 00:00:09 . Memory (MB): peak = 3650.578 ; gain = 7.027 ; free physical = 10429 ; free virtual = 114957
# report_utilization -hierarchical -file digilent_nexys4ddr_utilization_hierarchical_synth.rpt
# report_utilization -file digilent_nexys4ddr_utilization_synth.rpt
# opt_design -directive default
Command: opt_design -directive default
INFO: [Vivado_Tcl 4-136] Directive used for opt_design is: default
Attempting to get a license for feature 'Implementation' and/or device 'xc7a100t'
INFO: [Common 17-349] Got license for feature 'Implementation' and/or device 'xc7a100t'
Running DRC as a precondition to command opt_design

Starting DRC Task
INFO: [DRC 23-27] Running DRC with 8 threads
ERROR: [DRC INBB-3] Black Box Instances: Cell 'cva6_wrapper/i_ariane/i_cva6/i_cache_subsystem/i_cva6_icache/gen_sram[0].data_sram/gen_cut[0].gen_mem.i_tc_sram_wrapper' of type 'tc_sram_wrapper' has undefined contents and is considered a black box.  The contents of this cell must be defined for opt_design to complete successfully.
ERROR: [DRC INBB-3] Black Box Instances: Cell 'cva6_wrapper/i_ariane/i_cva6/i_cache_subsystem/i_cva6_icache/gen_sram[0].data_sram/gen_cut[1].gen_mem.i_tc_sram_wrapper' of type 'tc_sram_wrapper' has undefined contents and is considered a black box.  The contents of this cell must be defined for opt_design to complete successfully.
```

TianxuL, 2023-02-20:
Hello @lapd-soc-one , 
Thank you for your first answer, it also solved the problem that was bothering me.

I also faced the same situation as you, but  I solved it by adding the following two lines also in the Flist.cv64a6_imafdc_sv39 file.

`${CVA6_REPO_DIR}/common/local/util/tc_sram_fpga_wrapper.sv`
`${CVA6_REPO_DIR}/common/local/techlib/fpga/rtl/SyncSpRamBeNx64.sv`

lapnd, 2023-02-20:
Hi @TianxuL 
Thank you for your sharing.
I'm trying on AU200 board now, the bitstream is generated successfully, however the timing is not met :( 
I'm trying to reduce system clock frequency (default is 125MHz) to see if it works.
Thanks!