enjoy-digital/litex #1605: riscv software interruption problem with uart

Ch1psN3t, 2023-02-19:
Hello,

I want to perform a software interrupt on the value of the csr `mcause` register. But, when I perform this interrupt, I lose the display in the console. So, I wanted to change the value of the register csr `mtvec` to change the mode of the interrupts in vector mode to have several interrupts. But, I still have the same error. Do you have any idea how to add a software interrupt on the csr `mcause` register without losing the other one? 

Thank you,

Dolu1990, 2023-02-20:
Hi,
Depending which CPU you are using it is possible that those registers are read only to save area.
Which CPU are you using ?

Ch1psN3t, 2023-02-20:
I use vexriscv riscv processor. I am in machine mode. I used this link [https://www.five-embeddev.com/baremetal/vectored_interrupts/#linking-the-interrupt-handlers](url), but my second interrupt does not work.

Dolu1990, 2023-02-20:
hmmm, the default configuration do not allow mtvec writes, can you try the vexriscv-smp variant ? this one is a more beefy one, which allows settuping mtvec with write.

Ch1psN3t, 2023-02-20:
However, I see that I have modified the mtvec register with the GDB debugger. And I am in the secure variant.

Dolu1990, 2023-02-20:
Ahh i guess secure variant should be good.

> to change the mode of the interrupts in vector mode

Hoo one thing, that isn't enabled by default i think. Very rarely used it.

Does it work if you don't use vector mode ?

Dolu1990, 2023-02-20:
Seems like if you want vector interrupt mode, you would need to regenerate https://github.com/litex-hub/pythondata-cpu-vexriscv/blob/master/pythondata_cpu_vexriscv/verilog/Makefile#L61 with  --machineTrapVector argument  (personnaly never used it)

Ch1psN3t, 2023-02-20:
Without the vector mode of the `mtvec` register, I can see my interrupt being performed by looking at the path with the leds. But, I don't have `printf` in the console.

Ch1psN3t, 2023-02-21:
> Seems like if you want vector interrupt mode, you would need to regenerate https://github.com/litex-hub/pythondata-cpu-vexriscv/blob/master/pythondata_cpu_vexriscv/verilog/Makefile#L61 with --machineTrapVector argument (personnaly never used it)

Same with this option, it doesn't work.

Dolu1990, 2023-02-22:
How is setup the interrupt for the UART ? You mean that the uart controller interrupt isn't making the CPU going to the mtvec PC address ?

Ch1psN3t, 2023-02-22:
The uart interrupt is configured in the `/litex/litex/soc/cores/cpu/vexriscv/` file in the `Crt0.S` assembly file `trap_entry` is the function that calls the uart usage. I add my function in a vector that calls this `trap_entry` function and mine. 
But whatever the first interrupt in the vector is, it only calls this one, even for the `mtvec` mode at 1.

Dolu1990, 2023-02-22:
hmm, can you send me the generated VexRiscv verilog you are using ?

closed 2023-03-23 by KevinQhv