enjoy-digital/litex #1606: Memory initialization of Alveo U200 failed

offnaria, 2023-02-21:
Hello.

I found AU200 files became available a few days ago, so I tried them.
I built a bitstream by command like below, and programed my AU200 using Vivado Hardware Manager.

`litex-boards/litex_boards/targets/xilinx_alveo_u200.py --output-dir output --build`

Then, LiteX BIOS starts and "Memory initialization failed."
What should I do to fix it?

Thak you.

### Tool versions
- LiteX commit 4fa0ea2f
- litex-boards commit 549905b
- LiteDRAM commit 6c8df7c
- Vivado 2021.2

### litex_term log

```
        __   _ __      _  __
       / /  (_) /____ | |/_/
      / /__/ / __/ -_)>  <
     /____/_/\__/\__/_/|_|
   Build your hardware, easily!

 (c) Copyright 2012-2023 Enjoy-Digital
 (c) Copyright 2007-2015 M-Labs

 BIOS built on Feb 21 2023 12:58:18
 BIOS CRC passed (a206a802)

 LiteX git sha1: 4fa0ea2f

--=============== SoC ==================--
CPU:            VexRiscv @ 125MHz
BUS:            WISHBONE 32-bit @ 4GiB
CSR:            32-bit data
ROM:            128.0KiB
SRAM:           8.0KiB
L2:             8.0KiB
SDRAM:          16.0GiB 64-bit @ 1000MT/s (CL-9 CWL-9)
MAIN-RAM:       1.0GiB

--========== Initialization ============--
Initializing SDRAM @0x40000000...
Switching SDRAM to software control.
Write leveling:
  tCK equivalent taps: 596
  Cmd/Clk scan (0-298)
  |00011  |00000  |000000000  |000000000| best: 16
  Setting Cmd/Clk delay to 16 taps.
  Data scan:
  m0: |11111111111100000000000| delay: 00
  m1: |11111111110000000000000| delay: -
  m2: |11111111111111000000000| delay: 00
  m3: |11111111111111111111111| delay: 00
  m4: |00000000000000000111111| delay: 261
  m5: |11111111111111111111111| delay: 00
  m6: |11111111111111000000000| delay: 00
  m7: |11110000000000000000001| delay: -
Write latency calibration:
m0:4
m1:4
m2:4
m3:6
m4:6
m5:0
m6:4
m7:0
Read leveling:
  m0, b00: |00000000000000000000000000000000| delays: -
  m0, b01: |00000000000000000000000000000000| delays: -
  m0, b02: |00000000000000000000000000000000| delays: -
  m0, b03: |00000000000000000000000000000000| delays: -
  m0, b04: |00000000000000000000000000000000| delays: -
  m0, b05: |00000000000000000000000000000000| delays: -
  m0, b06: |00000000000000000000000000000000| delays: -
  m0, b07: |00000000000000000000000000000000| delays: -
  best: m0, b05 delays: -
  m1, b00: |00000000000000000000000000000000| delays: -
  m1, b01: |00000000000000000000000000000000| delays: -
  m1, b02: |00000000000000000000000000000000| delays: -
  m1, b03: |00000000000000000000000000000000| delays: -
  m1, b04: |00000000000000000000000000000000| delays: -
  m1, b05: |00000000000000000000000000000000| delays: -
  m1, b06: |00000000000000000000000000000000| delays: -
  m1, b07: |00000000000000000000000000000000| delays: -
  best: m1, b05 delays: -
  m2, b00: |00000000000000000000000000000000| delays: -
  m2, b01: |00000000000000000000000000000000| delays: -
  m2, b02: |00000000000000000000000000000000| delays: -
  m2, b03: |00000000000000000000000000000000| delays: -
  m2, b04: |00000000000000000000000000000000| delays: -
  m2, b05: |00000000000000000000000000000000| delays: -
  m2, b06: |00000000000000000000000000000000| delays: -
  m2, b07: |00000000000000000000000000000000| delays: -
  best: m2, b05 delays: -
  m3, b00: |00000000000000000000000000000000| delays: -
  m3, b01: |00000000000000000000000000000000| delays: -
  m3, b02: |00000000000000000000000000000000| delays: -
  m3, b03: |00000000000000000000000000000000| delays: -
  m3, b04: |00000000000000000000000000000000| delays: -
  m3, b05: |00000000000000000000000000000000| delays: -
  m3, b06: |00000000000000000000000000000000| delays: -
  m3, b07: |00000000000000000000000000000000| delays: -
  best: m3, b04 delays: -
  m4, b00: |00000000000000000000000000000000| delays: -
  m4, b01: |00000000000000000000000000000000| delays: -
  m4, b02: |00000000000000000000000000000000| delays: -
  m4, b03: |00000000000000000000000000000000| delays: -
  m4, b04: |00000000000000000000000000000000| delays: -
  m4, b05: |00000000000000000000000000000000| delays: -
  m4, b06: |00000000000000000000000000000000| delays: -
  m4, b07: |00000000000000000000000000000000| delays: -
  best: m4, b05 delays: -
  m5, b00: |00000000000000000000000000000000| delays: -
  m5, b01: |00000000000000000000000000000000| delays: -
  m5, b02: |00000000000000000000000000000000| delays: -
  m5, b03: |00000000000000000000000000000000| delays: -
  m5, b04: |00000000000000000000000000000000| delays: -
  m5, b05: |00000000000000000000000000000000| delays: -
  m5, b06: |00000000000000000000000000000000| delays: -
  m5, b07: |00000000000000000000000000000000| delays: -
  best: m5, b00 delays: -
  m6, b00: |00000000000000000000000000000000| delays: -
  m6, b01: |00000000000000000000000000000000| delays: -
  m6, b02: |00000000000000000000000000000000| delays: -
  m6, b03: |00000000000000000000000000000000| delays: -
  m6, b04: |00000000000000000000000000000000| delays: -
  m6, b05: |00000000000000000000000000000000| delays: -
  m6, b06: |00000000000000000000000000000000| delays: -
  m6, b07: |00000000000000000000000000000000| delays: -
  best: m6, b05 delays: -
  m7, b00: |00000000000000000000000000000000| delays: -
  m7, b01: |00000000000000000000000000000000| delays: -
  m7, b02: |00000000000000000000000000000000| delays: -
  m7, b03: |00000000000000000000000000000000| delays: -
  m7, b04: |00000000000000000000000000000000| delays: -
  m7, b05: |00000000000000000000000000000000| delays: -
  m7, b06: |00000000000000000000000000000000| delays: -
  m7, b07: |00000000000000000000000000000000| delays: -
  best: m7, b00 delays: -
Switching SDRAM to hardware control.
Memtest at 0x40000000 (2.0MiB)...
  Write: 0x40000000-0x40200000 2.0MiB
   Read: 0x40000000-0x40200000 2.0MiB
  bus errors:  256/256
  addr errors: 0/8192
  data errors: 524288/524288
Memtest KO
Memory initialization failed

--============= Console ================--

litex>
```

lapnd, 2023-02-21:
Hi @offnaria 
I had the same issue and got it fix by using latest Vivado (2022.2)

offnaria, 2023-02-21:
Thank you @lapnd !
I'll try that!

offnaria, 2023-02-22:
I generated bitstream with Vivado 2022.2 and loaded it onto AU200, but the memory test is still failed...
Are there anything to check?
Any suggestions would be greatly appreciated

Vivado version
```
Vivado v2022.2 (64-bit)
SW Build 3671981 on Fri Oct 14 04:59:54 MDT 2022
IP Build 3669848 on Fri Oct 14 08:30:02 MDT 2022
Tool Version Limit: 2022.10
Copyright 1986-2022 Xilinx, Inc. All Rights Reserved.
```

litex_term log
```
        __   _ __      _  __
       / /  (_) /____ | |/_/
      / /__/ / __/ -_)>  <
     /____/_/\__/\__/_/|_|
   Build your hardware, easily!

 (c) Copyright 2012-2023 Enjoy-Digital
 (c) Copyright 2007-2015 M-Labs

 BIOS built on Feb 22 2023 14:40:44
 BIOS CRC passed (e64740c4)

 LiteX git sha1: 4fa0ea2f

--=============== SoC ==================--
CPU:            VexRiscv @ 125MHz
BUS:            WISHBONE 32-bit @ 4GiB
CSR:            32-bit data
ROM:            128.0KiB
SRAM:           8.0KiB
L2:             8.0KiB
SDRAM:          16.0GiB 64-bit @ 1000MT/s (CL-9 CWL-9)
MAIN-RAM:       1.0GiB

--========== Initialization ============--
Initializing SDRAM @0x40000000...
Switching SDRAM to software control.
Write leveling:
  tCK equivalent taps: 596
  Cmd/Clk scan (0-298)
  |00001  |00000  |000000000  |000000000| best: 18
  Setting Cmd/Clk delay to 18 taps.
  Data scan:
  m0: |11111111111100000000000| delay: 00
  m1: |11111111110000000000000| delay: 00
  m2: |11111111111111000000000| delay: 00
  m3: |11111111111111111111111| delay: 00
  m4: |00000000000000000111111| delay: 260
  m5: |11111111111111111111111| delay: 00
  m6: |11111111111111000000000| delay: 00
  m7: |11110000000000000000001| delay: -
Write latency calibration:
m0:4
m1:4
m2:4
m3:6
m4:6
m5:0
m6:4
m7:4
Read leveling:
  m0, b00: |00000000000000000000000000000000| delays: -
  m0, b01: |00000000000000000000000000000000| delays: -
  m0, b02: |00000000000000000000000000000000| delays: -
  m0, b03: |00000000000000000000000000000000| delays: -
  m0, b04: |00000000000000000000000000000000| delays: -
  m0, b05: |00000000000000000000000000000000| delays: -
  m0, b06: |00000000000000000000000000000000| delays: -
  m0, b07: |00000000000000000000000000000000| delays: -
  best: m0, b05 delays: -
  m1, b00: |00000000000000000000000000000000| delays: -
  m1, b01: |00000000000000000000000000000000| delays: -
  m1, b02: |00000000000000000000000000000000| delays: -
  m1, b03: |00000000000000000000000000000000| delays: -
  m1, b04: |00000000000000000000000000000000| delays: -
  m1, b05: |00000000000000000000000000000000| delays: -
  m1, b06: |00000000000000000000000000000000| delays: -
  m1, b07: |00000000000000000000000000000000| delays: -
  best: m1, b05 delays: -
  m2, b00: |00000000000000000000000000000000| delays: -
  m2, b01: |00000000000000000000000000000000| delays: -
  m2, b02: |00000000000000000000000000000000| delays: -
  m2, b03: |00000000000000000000000000000000| delays: -
  m2, b04: |00000000000000000000000000000000| delays: -
  m2, b05: |00000000000000000000000000000000| delays: -
  m2, b06: |00000000000000000000000000000000| delays: -
  m2, b07: |00000000000000000000000000000000| delays: -
  best: m2, b05 delays: -
  m3, b00: |00000000000000000000000000000000| delays: -
  m3, b01: |00000000000000000000000000000000| delays: -
  m3, b02: |00000000000000000000000000000000| delays: -
  m3, b03: |00000000000000000000000000000000| delays: -
  m3, b04: |00000000000000000000000000000000| delays: -
  m3, b05: |00000000000000000000000000000000| delays: -
  m3, b06: |00000000000000000000000000000000| delays: -
  m3, b07: |00000000000000000000000000000000| delays: -
  best: m3, b04 delays: -
  m4, b00: |00000000000000000000000000000000| delays: -
  m4, b01: |00000000000000000000000000000000| delays: -
  m4, b02: |00000000000000000000000000000000| delays: -
  m4, b03: |00000000000000000000000000000000| delays: -
  m4, b04: |00000000000000000000000000000000| delays: -
  m4, b05: |00000000000000000000000000000000| delays: -
  m4, b06: |00000000000000000000000000000000| delays: -
  m4, b07: |00000000000000000000000000000000| delays: -
  best: m4, b05 delays: -
  m5, b00: |00000000000000000000000000000000| delays: -
  m5, b01: |00000000000000000000000000000000| delays: -
  m5, b02: |00000000000000000000000000000000| delays: -
  m5, b03: |00000000000000000000000000000000| delays: -
  m5, b04: |00000000000000000000000000000000| delays: -
  m5, b05: |00000000000000000000000000000000| delays: -
  m5, b06: |00000000000000000000000000000000| delays: -
  m5, b07: |00000000000000000000000000000000| delays: -
  best: m5, b00 delays: -
  m6, b00: |00000000000000000000000000000000| delays: -
  m6, b01: |00000000000000000000000000000000| delays: -
  m6, b02: |00000000000000000000000000000000| delays: -
  m6, b03: |00000000000000000000000000000000| delays: -
  m6, b04: |00000000000000000000000000000000| delays: -
  m6, b05: |00000000000000000000000000000000| delays: -
  m6, b06: |00000000000000000000000000000000| delays: -
  m6, b07: |00000000000000000000000000000000| delays: -
  best: m6, b05 delays: -
  m7, b00: |00000000000000000000000000000000| delays: -
  m7, b01: |00000000000000000000000000000000| delays: -
  m7, b02: |00000000000000000000000000000000| delays: -
  m7, b03: |00000000000000000000000000000000| delays: -
  m7, b04: |00000000000000000000000000000000| delays: -
  m7, b05: |00000000000000000000000000000000| delays: -
  m7, b06: |00000000000000000000000000000000| delays: -
  m7, b07: |00000000000000000000000000000000| delays: -
  best: m7, b05 delays: -
Switching SDRAM to hardware control.
Memtest at 0x40000000 (2.0MiB)...
  Write: 0x40000000-0x40200000 2.0MiB
   Read: 0x40000000-0x40200000 2.0MiB
  bus errors:  256/256
  addr errors: 0/8192
  data errors: 524288/524288
Memtest KO
Memory initialization failed

--============= Console ================--

litex>
```

offnaria, 2023-02-22:
BTW, my **AU250** has a similar problem.
However, its "bus errors" are less than 256.

offnaria, 2023-02-22:
I tried to generate bitstream with Vivado's Runs (synth_1, impl_1) instead of LiteX's tcl script, but still failed...

enjoy-digital, 2023-02-23:
Hi @offnaria,

maybe we could try first to see if this is related to variation between boards or issue with the build. @vietthanh85would you mind sharing a bistream you tested succesfully so that @offnaria could do a test?

vietthanh85, 2023-02-23:
@offnaria I share my bitstream which is also built by Vivado 2022.2. Please test on your board.

```
        __   _ __      _  __
       / /  (_) /____ | |/_/
[xilinx_alveo_u200.zip](https://github.com/enjoy-digital/litex/files/10813950/xilinx_alveo_u200.zip)

      / /__/ / __/ -_)>  <
     /____/_/\__/\__/_/|_|
   Build your hardware, easily!

 (c) Copyright 2012-2022 Enjoy-Digital
 (c) Copyright 2007-2015 M-Labs

 BIOS built on Feb 23 2023 13:21:59
 BIOS CRC passed (38049151)

 LiteX git sha1: 8b14e649

--=============== SoC ==================--
CPU:		VexRiscv @ 125MHz
BUS:		WISHBONE 32-bit @ 4GiB
CSR:		32-bit data
ROM:		128.0KiB
SRAM:		8.0KiB
L2:		8.0KiB
SDRAM:		16.0GiB 64-bit @ 1000MT/s (CL-9 CWL-9)
MAIN-RAM:	1.0GiB

--========== Initialization ============--
Initializing SDRAM @0x40000000...
Switching SDRAM to software control.
Write leveling:
  tCK equivalent taps: 612
  Cmd/Clk scan (0-306)
  |00111  |000000111  |000000001  |000000000| best: 72
  Setting Cmd/Clk delay to 72 taps.
  Data scan:
  m0: |00111111111111111111100| delay: 32
  m1: |11111111111111111110000| delay: 00
  m2: |00000111111111111111111| delay: 80
  m3: |11111111111111110000000| delay: 00
  m4: |11111111100000000000000| delay: -
  m5: |11111111111100000000000| delay: 00
  m6: |00000111111111111111111| delay: 72
  m7: |11111111111111000000000| delay: 00
Write latency calibration:
m0:6 m1:6 m2:6 m3:6 m4:6 m5:6 m6:6 m7:6 
Read leveling:
  m0, b00: |00000000000000000000000000000000| delays: -
  m0, b01: |00000000000000000000000000000000| delays: -
  m0, b02: |00000000000000000000000000000000| delays: -
  m0, b03: |11111111111111000000000000000000| delays: 108+-108
  m0, b04: |00000000000000001111111111111111| delays: 381+-129
  m0, b05: |00000000000000000000000000000000| delays: -
  m0, b06: |00000000000000000000000000000000| delays: -
  m0, b07: |00000000000000000000000000000000| delays: -
  best: m0, b04 delays: 380+-130
  m1, b00: |00000000000000000000000000000000| delays: -
  m1, b01: |00000000000000000000000000000000| delays: -
  m1, b02: |00000000000000000000000000000000| delays: -
  m1, b03: |00111111111111111110000000000000| delays: 156+-133
  m1, b04: |00000000000000000000011111111111| delays: 418+-92
  m1, b05: |00000000000000000000000000000000| delays: -
  m1, b06: |00000000000000000000000000000000| delays: -
  m1, b07: |00000000000000000000000000000000| delays: -
  best: m1, b03 delays: 156+-134
  m2, b00: |00000000000000000000000000000000| delays: -
  m2, b01: |00000000000000000000000000000000| delays: -
  m2, b02: |00000000000000000000000000000000| delays: -
  m2, b03: |11110000000000000000000000000000| delays: 32+-32
  m2, b04: |00000001111111111111111100000000| delays: 238+-132
  m2, b05: |00000000000000000000000000111111| delays: 463+-48
  m2, b06: |00000000000000000000000000000000| delays: -
  m2, b07: |00000000000000000000000000000000| delays: -
  best: m2, b04 delays: 240+-134
  m3, b00: |00000000000000000000000000000000| delays: -
  m3, b01: |00000000000000000000000000000000| delays: -
  m3, b02: |00000000000000000000000000000000| delays: -
  m3, b03: |10000000000000000000000000000000| delays: -
  m3, b04: |00011111111111111111000000000000| delays: 180+-129
  m3, b05: |00000000000000000000000111111111| delays: 433+-77
  m3, b06: |00000000000000000000000000000000| delays: -
  m3, b07: |00000000000000000000000000000000| delays: -
  best: m3, b04 delays: 179+-126
  m4, b00: |00000000000000000000000000000000| delays: -
  m4, b01: |00000000000000000000000000000000| delays: -
  m4, b02: |00000000000000000000000000000000| delays: -
  m4, b03: |11111111111111110000000000000000| delays: 125+-125
  m4, b04: |00000000000000000001111111111111| delays: 408+-103
  m4, b05: |00000000000000000000000000000000| delays: -
  m4, b06: |00000000000000000000000000000000| delays: -
  m4, b07: |00000000000000000000000000000000| delays: -
  best: m4, b03 delays: 125+-125
  m5, b00: |00000000000000000000000000000000| delays: -
  m5, b01: |00000000000000000000000000000000| delays: -
  m5, b02: |00000000000000000000000000000000| delays: -
  m5, b03: |11110000000000000000000000000000| delays: 31+-31
  m5, b04: |00000001111111111111111100000000| delays: 239+-132
  m5, b05: |00000000000000000000000000111111| delays: 462+-49
  m5, b06: |00000000000000000000000000000000| delays: -
  m5, b07: |00000000000000000000000000000000| delays: -
  best: m5, b04 delays: 239+-131
  m6, b00: |00000000000000000000000000000000| delays: -
  m6, b01: |00000000000000000000000000000000| delays: -
  m6, b02: |00000000000000000000000000000000| delays: -
  m6, b03: |11111111111000000000000000000000| delays: 83+-83
  m6, b04: |00000000000001111111111111111100| delays: 338+-138
  m6, b05: |00000000000000000000000000000000| delays: 508+-03
  m6, b06: |00000000000000000000000000000000| delays: -
  m6, b07: |00000000000000000000000000000000| delays: -
  best: m6, b04 delays: 337+-140
  m7, b00: |00000000000000000000000000000000| delays: -
  m7, b01: |00000000000000000000000000000000| delays: -
  m7, b02: |11100000000000000000000000000000| delays: 15+-15
  m7, b03: |00000111111111111111110000000000| delays: 207+-131
  m7, b04: |00000000000000000000000001111111| delays: 446+-64
  m7, b05: |00000000000000000000000000000000| delays: -
  m7, b06: |00000000000000000000000000000000| delays: -
  m7, b07: |00000000000000000000000000000000| delays: -
  best: m7, b03 delays: 207+-131
Switching SDRAM to hardware control.
Memtest at 0x40000000 (2.0MiB)...
  Write: 0x40000000-0x40200000 2.0MiB     
   Read: 0x40000000-0x40200000 2.0MiB     
Memtest OK
Memspeed at 0x40000000 (Sequential, 2.0MiB)...
  Write speed: 108.8MiB/s
   Read speed: 93.6MiB/s

--============== Boot ==================--
Booting from serial...
Press Q or ESC to abort boot completely.
sL5DdSMmkekro
             Timeout
No boot medium found

--============= Console ================--

litex> 
```

offnaria, 2023-02-28:
Thanks, @vietthanh85 !
I tried your bitstream today on my AU200, but the initialization still failed...

This is the log.
The meaningful difference between this and my first log may be...

- "delays" value appears while "Read leveling"
- The number of "data errors" decreased

```
        __   _ __      _  __
       / /  (_) /____ | |/_/
      / /__/ / __/ -_)>  <
     /____/_/\__/\__/_/|_|
   Build your hardware, easily!

 (c) Copyright 2012-2022 Enjoy-Digital
 (c) Copyright 2007-2015 M-Labs

 BIOS built on Feb 23 2023 13:21:59
 BIOS CRC passed (38049151)

 LiteX git sha1: 8b14e649

--=============== SoC ==================--
CPU:            VexRiscv @ 125MHz
BUS:            WISHBONE 32-bit @ 4GiB
CSR:            32-bit data
ROM:            128.0KiB
SRAM:           8.0KiB
L2:             8.0KiB
SDRAM:          16.0GiB 64-bit @ 1000MT/s (CL-9 CWL-9)
MAIN-RAM:       1.0GiB

--========== Initialization ============--
Initializing SDRAM @0x40000000...
Switching SDRAM to software control.
Write leveling:
  tCK equivalent taps: 596
  Cmd/Clk scan (0-298)
  |00011  |00000  |000000000  |000000000| best: 16
  Setting Cmd/Clk delay to 16 taps.
  Data scan:
  m0: |11111111111100000000000| delay: 00
  m1: |11111111110000000000000| delay: 00
  m2: |11111111111111000000000| delay: 00
  m3: |11111111111111111111111| delay: 00
  m4: |00000000000000000111111| delay: 264
  m5: |11111111111111111111111| delay: 00
  m6: |11111111111111000000000| delay: 00
  m7: |11110000000000000000001| delay: -
Write latency calibration:
m0:6 m1:6 m2:6 m3:6 m4:0 m5:0 m6:6 m7:6
Read leveling:
  m0, b00: |00000000000000000000000000000000| delays: -
  m0, b01: |00000000000000000000000000000000| delays: -
  m0, b02: |11111110000000000000000000000000| delays: 50+-50
  m0, b03: |00000000011111111111111110000000| delays: 261+-131
  m0, b04: |00000000000000000000000000011111| delays: 469+-42
  m0, b05: |00000000000000000000000000000000| delays: -
  m0, b06: |00000000000000000000000000000000| delays: -
  m0, b07: |00000000000000000000000000000000| delays: -
  best: m0, b03 delays: 260+-132
  m1, b00: |00000000000000000000000000000000| delays: -
  m1, b01: |00000000000000000000000000000000| delays: -
  m1, b02: |11111111111000000000000000000000| delays: 86+-86
  m1, b03: |00000000000001111111111111111100| delays: 335+-136
  m1, b04: |00000000000000000000000000000001| delays: 504+-07
  m1, b05: |00000000000000000000000000000000| delays: -
  m1, b06: |00000000000000000000000000000000| delays: -
  m1, b07: |00000000000000000000000000000000| delays: -
  best: m1, b03 delays: 333+-135
  m2, b00: |00000000000000000000000000000000| delays: -
  m2, b01: |00000000000000000000000000000000| delays: -
  m2, b02: |00000000000000000000000000000000| delays: -
  m2, b03: |11111111111111111000000000000000| delays: 131+-131
  m2, b04: |00000000000000000001111111111111| delays: 406+-104
  m2, b05: |00000000000000000000000000000000| delays: -
  m2, b06: |00000000000000000000000000000000| delays: -
  m2, b07: |00000000000000000000000000000000| delays: -
  best: m2, b03 delays: 131+-129
  m3, b00: |00000000000000000000000000000000| delays: -
  m3, b01: |00000000000000000000000000000000| delays: -
  m3, b02: |00000000000000000000000000000000| delays: -
  m3, b03: |00000000000000000000000000000000| delays: -
  m3, b04: |00000000000000000000000000000000| delays: -
  m3, b05: |00000000000000000000000000000000| delays: -
  m3, b06: |00000000000000000000000000000000| delays: -
  m3, b07: |00000000000000000000000000000000| delays: -
  best: m3, b04 delays: -
  m4, b00: |00000000000000000000000000000000| delays: -
  m4, b01: |00000000000000000000000000000000| delays: -
  m4, b02: |11111111100000000000000000000000| delays: 66+-66
  m4, b03: |00000000000111111111111111100000| delays: 297+-131
  m4, b04: |00000000000000000000000000000011| delays: 490+-21
  m4, b05: |00000000000000000000000000000000| delays: -
  m4, b06: |00000000000000000000000000000000| delays: -
  m4, b07: |00000000000000000000000000000000| delays: -
  best: m4, b03 delays: 297+-129
  m5, b00: |00000000000000000000000000000000| delays: -
  m5, b01: |00000000000000000000000000000000| delays: -
  m5, b02: |00000000000000000000000000000000| delays: -
  m5, b03: |00000000000000000000000000000000| delays: -
  m5, b04: |00000000000000000000000000000000| delays: -
  m5, b05: |00000000000000000000000000000000| delays: -
  m5, b06: |00000000000000000000000000000000| delays: -
  m5, b07: |00000000000000000000000000000000| delays: -
  best: m5, b00 delays: -
  m6, b00: |00000000000000000000000000000000| delays: -
  m6, b01: |00000000000000000000000000000000| delays: -
  m6, b02: |11110000000000000000000000000000| delays: 29+-29
  m6, b03: |00000011111111111111111000000000| delays: 220+-135
  m6, b04: |00000000000000000000000011111111| delays: 448+-62
  m6, b05: |00000000000000000000000000000000| delays: -
  m6, b06: |00000000000000000000000000000000| delays: -
  m6, b07: |00000000000000000000000000000000| delays: -
  best: m6, b03 delays: 218+-133
  m7, b00: |00000000000000000000000000000000| delays: -
  m7, b01: |00000000000000000000000000000000| delays: -
  m7, b02: |11111111111111100000000000000000| delays: 112+-112
  m7, b03: |00000000000000001111111111111111| delays: 384+-127
  m7, b04: |00000000000000000000000000000000| delays: -
  m7, b05: |00000000000000000000000000000000| delays: -
  m7, b06: |00000000000000000000000000000000| delays: -
  m7, b07: |00000000000000000000000000000000| delays: -
  best: m7, b03 delays: 386+-125
Switching SDRAM to hardware control.
Memtest at 0x40000000 (2.0MiB)...
  Write: 0x40000000-0x40200000 2.0MiB
   Read: 0x40000000-0x40200000 2.0MiB
  bus errors:  256/256
  addr errors: 0/8192
  data errors: 506692/524288
Memtest KO
Memory initialization failed

--============= Console ================--

litex> 
```

offnaria, 2023-02-28:
I got this log after rebooting the SoC.
The values of "deay" while "Data scan" differ from the previous one.
Also "bus errors" and "data errors" values are almost half of the previous.

```
litex> reboot▒
        __   _ __      _  __
       / /  (_) /____ | |/_/
      / /__/ / __/ -_)>  <
     /____/_/\__/\__/_/|_|
   Build your hardware, easily!

 (c) Copyright 2012-2022 Enjoy-Digital
 (c) Copyright 2007-2015 M-Labs

 BIOS built on Feb 23 2023 13:21:59
 BIOS CRC passed (38049151)

 LiteX git sha1: 8b14e649

--=============== SoC ==================--
CPU:            VexRiscv @ 125MHz
BUS:            WISHBONE 32-bit @ 4GiB
CSR:            32-bit data
ROM:            128.0KiB
SRAM:           8.0KiB
L2:             8.0KiB
SDRAM:          16.0GiB 64-bit @ 1000MT/s (CL-9 CWL-9)
MAIN-RAM:       1.0GiB

--========== Initialization ============--
Initializing SDRAM @0x40000000...
Switching SDRAM to software control.
Write leveling:
  tCK equivalent taps: 596
  Cmd/Clk scan (0-298)
  |00011  |000011111  |111111111| best: 216
  Setting Cmd/Clk delay to 216 taps.
  Data scan:
  m0: |00001111111111111111111| delay: 58
  m1: |00111111111111111111110| delay: 20
  m2: |00000011111111111111111| delay: 86
  m3: |11111111111111111100000| delay: 00
  m4: |11111111111100000000000| delay: 00
  m5: |11111111111111111111111| delay: 00
  m6: |00000011111111111111111| delay: 95
  m7: |11111111111111110000000| delay: 00
Write latency calibration:
m0:6 m1:6 m2:6 m3:6 m4:6 m5:6 m6:6 m7:6
Read leveling:
  m0, b00: |00000000000000000000000000000000| delays: -
  m0, b01: |00000000000000000000000000000000| delays: -
  m0, b02: |00000000000000000000000000000000| delays: -
  m0, b03: |11111111111110000000000000000000| delays: 99+-99
  m0, b04: |00000000000000011111111111111111| delays: 365+-129
  m0, b05: |00000000000000000000000000000000| delays: -
  m0, b06: |00000000000000000000000000000000| delays: -
  m0, b07: |00000000000000000000000000000000| delays: -
  best: m0, b04 delays: 363+-130
  m1, b00: |00000000000000000000000000000000| delays: -
  m1, b01: |00000000000000000000000000000000| delays: -
  m1, b02: |00000000000000000000000000000000| delays: -
  m1, b03: |01111111111111111100000000000000| delays: 141+-130
  m1, b04: |00000000000000000001111111111111| delays: 406+-105
  m1, b05: |00000000000000000000000000000000| delays: -
  m1, b06: |00000000000000000000000000000000| delays: -
  m1, b07: |00000000000000000000000000000000| delays: -
  best: m1, b03 delays: 139+-133
  m2, b00: |00000000000000000000000000000000| delays: -
  m2, b01: |00000000000000000000000000000000| delays: -
  m2, b02: |00000000000000000000000000000000| delays: -
  m2, b03: |11111000000000000000000000000000| delays: 33+-33
  m2, b04: |00000001111111111111111000000000| delays: 234+-129
  m2, b05: |00000000000000000000000000111111| delays: 460+-50
  m2, b06: |00000000000000000000000000000000| delays: -
  m2, b07: |00000000000000000000000000000000| delays: -
  best: m2, b04 delays: 236+-127
  m3, b00: |00000000000000000000000000000000| delays: -
  m3, b01: |00000000000000000000000000000000| delays: -
  m3, b02: |00000000000000000000000000000000| delays: -
  m3, b03: |10000000000000000000000000000000| delays: 04+-04
  m3, b04: |00011111111111111111000000000000| delays: 176+-132
  m3, b05: |00000000000000000000001111111111| delays: 429+-82
  m3, b06: |00000000000000000000000000000000| delays: -
  m3, b07: |00000000000000000000000000000000| delays: -
  best: m3, b04 delays: 177+-132
  m4, b00: |00000000000000000000000000000000| delays: -
  m4, b01: |00000000000000000000000000000000| delays: -
  m4, b02: |00000000000000000000000000000000| delays: -
  m4, b03: |11111111111111100000000000000000| delays: 116+-116
  m4, b04: |00000000000000000111111111111111| delays: 391+-119
  m4, b05: |00000000000000000000000000000000| delays: -
  m4, b06: |00000000000000000000000000000000| delays: -
  m4, b07: |00000000000000000000000000000000| delays: -
  best: m4, b04 delays: 392+-118
  m5, b00: |00000000000000000000000000000000| delays: -
  m5, b01: |00000000000000000000000000000000| delays: -
  m5, b02: |00000000000000000000000000000000| delays: -
  m5, b03: |00000000000000000000000000000000| delays: -
  m5, b04: |00000000000000000000000000000000| delays: -
  m5, b05: |00000000000000000000000000000000| delays: -
  m5, b06: |00000000000000000000000000000000| delays: -
  m5, b07: |00000000000000000000000000000000| delays: -
  best: m5, b04 delays: -
  m6, b00: |00000000000000000000000000000000| delays: -
  m6, b01: |00000000000000000000000000000000| delays: -
  m6, b02: |00000000000000000000000000000000| delays: -
  m6, b03: |11111111110000000000000000000000| delays: 79+-79
  m6, b04: |00000000000011111111111111111000| delays: 322+-132
  m6, b05: |00000000000000000000000000000001| delays: 499+-11
  m6, b06: |00000000000000000000000000000000| delays: -
  m6, b07: |00000000000000000000000000000000| delays: -
  best: m6, b04 delays: 321+-132
  m7, b00: |00000000000000000000000000000000| delays: -
  m7, b01: |00000000000000000000000000000000| delays: -
  m7, b02: |11000000000000000000000000000000| delays: 15+-15
  m7, b03: |00001111111111111111000000000000| delays: 190+-130
  m7, b04: |00000000000000000000000111111111| delays: 436+-75
  m7, b05: |00000000000000000000000000000000| delays: -
  m7, b06: |00000000000000000000000000000000| delays: -
  m7, b07: |00000000000000000000000000000000| delays: -
  best: m7, b03 delays: 189+-131
Switching SDRAM to hardware control.
Memtest at 0x40000000 (2.0MiB)...
  Write: 0x40000000-0x40200000 2.0MiB
   Read: 0x40000000-0x40200000 2.0MiB
  bus errors:  128/256
  addr errors: 0/8192
  data errors: 261146/524288
Memtest KO
Memory initialization failed

--============= Console ================--

litex>
```

vietthanh85, 2023-03-04:
@offnaria Could you share me your bitstream. I'll flash it to my board. If it works, the problem might be from the board.

offnaria, 2023-03-06:
@vietthanh85  Sorry for the delay.
Please test this bitstream.
[xilinx_alveo_u200.zip](https://github.com/enjoy-digital/litex/files/10898089/xilinx_alveo_u200.zip)

vietthanh85, 2023-03-10:
@offnaria 
Sorry for the late reply. Here is my test with your bitstream on my board:

```
        __   _ __      _  __
       / /  (_) /____ | |/_/
      / /__/ / __/ -_)>  <
     /____/_/\__/\__/_/|_|
   Build your hardware, easily!

 (c) Copyright 2012-2023 Enjoy-Digital
 (c) Copyright 2007-2015 M-Labs

 BIOS built on Feb 17 2023 21:21:58
 BIOS CRC passed (4c32b058)

 LiteX git sha1: 4fa0ea2f

--=============== SoC ==================--
CPU:		VexRiscv @ 125MHz
BUS:		WISHBONE 32-bit @ 4GiB
CSR:		32-bit data
ROM:		128.0KiB
SRAM:		8.0KiB
L2:		8.0KiB
SDRAM:		16.0GiB 64-bit @ 1000MT/s (CL-9 CWL-9)
MAIN-RAM:	1.0GiB

--========== Initialization ============--
Initializing SDRAM @0x40000000...
Switching SDRAM to software control.
Write leveling:
  tCK equivalent taps: 616
  Cmd/Clk scan (0-308)
  |00111  |000000011  |000000000  |000000000| best: 72
  Setting Cmd/Clk delay to 72 taps.
  Data scan:
  m0: |00111111111111111111100| delay: 31
  m1: |11111111111111111110000| delay: 00
  m2: |00000111111111111111111| delay: 78
  m3: |11111111111111110000000| delay: 00
  m4: |11111111100000000000000| delay: -
  m5: |11111111111000000000000| delay: 00
  m6: |00000111111111111111111| delay: 69
  m7: |11111111111110000000000| delay: 00
Write latency calibration:
m0:4 
m1:4 
m2:4 
m3:6 
m4:4 
m5:4 
m6:0 
m7:0 
Read leveling:
  m0, b00: |00000000000000000000000000000000| delays: -
  m0, b01: |00000000000000000000000000000000| delays: -
  m0, b02: |00000000000000000000000000000000| delays: -
  m0, b03: |00000000000000000000000000000000| delays: -
  m0, b04: |00000000000000000000000000000000| delays: -
  m0, b05: |00000000000000000000000000000000| delays: -
  m0, b06: |00000000000000000000000000000000| delays: -
  m0, b07: |00000000000000000000000000000000| delays: -
  best: m0, b06 delays: -
  m1, b00: |00000000000000000000000000000000| delays: -
  m1, b01: |00000000000000000000000000000000| delays: -
  m1, b02: |00000000000000000000000000000000| delays: -
  m1, b03: |00000000000000000000000000000000| delays: -
  m1, b04: |00000000000000000000000000000000| delays: -
  m1, b05: |00000000000000000000000000000000| delays: -
  m1, b06: |00000000000000000000000000000000| delays: -
  m1, b07: |00000000000000000000000000000000| delays: -
  best: m1, b05 delays: -
  m2, b00: |00000000000000000000000000000000| delays: -
  m2, b01: |00000000000000000000000000000000| delays: -
  m2, b02: |00000000000000000000000000000000| delays: -
  m2, b03: |00000000000000000000000000000000| delays: -
  m2, b04: |00000000000000000000000000000000| delays: -
  m2, b05: |00000000000000000000000000000000| delays: -
  m2, b06: |00000000000000000000000000000000| delays: -
  m2, b07: |00000000000000000000000000000000| delays: -
  best: m2, b06 delays: -
  m3, b00: |00000000000000000000000000000000| delays: -
  m3, b01: |00000000000000000000000000000000| delays: -
  m3, b02: |00000000000000000000000000000000| delays: -
  m3, b03: |00000000000000000000000000000000| delays: -
  m3, b04: |00000000000000000000000000000000| delays: -
  m3, b05: |00000000000000000000000000000000| delays: -
  m3, b06: |00000000000000000000000000000000| delays: -
  m3, b07: |00000000000000000000000000000000| delays: -
  best: m3, b04 delays: -
  m4, b00: |00000000000000000000000000000000| delays: -
  m4, b01: |00000000000000000000000000000000| delays: -
  m4, b02: |00000000000000000000000000000000| delays: -
  m4, b03: |00000000000000000000000000000000| delays: -
  m4, b04: |00000000000000000000000000000000| delays: -
  m4, b05: |00000000000000000000000000000000| delays: -
  m4, b06: |00000000000000000000000000000000| delays: -
  m4, b07: |00000000000000000000000000000000| delays: -
  best: m4, b05 delays: -
  m5, b00: |00000000000000000000000000000000| delays: -
  m5, b01: |00000000000000000000000000000000| delays: -
  m5, b02: |00000000000000000000000000000000| delays: -
  m5, b03: |00000000000000000000000000000000| delays: -
  m5, b04: |00000000000000000000000000000000| delays: -
  m5, b05: |00000000000000000000000000000000| delays: -
  m5, b06: |00000000000000000000000000000000| delays: -
  m5, b07: |00000000000000000000000000000000| delays: -
  best: m5, b06 delays: -
  m6, b00: |00000000000000000000000000000000| delays: -
  m6, b01: |00000000000000000000000000000000| delays: -
  m6, b02: |00000000000000000000000000000000| delays: -
  m6, b03: |00000000000000000000000000000000| delays: -
  m6, b04: |00000000000000000000000000000000| delays: -
  m6, b05: |00000000000000000000000000000000| delays: -
  m6, b06: |00000000000000000000000000000000| delays: -
  m6, b07: |00000000000000000000000000000000| delays: -
  best: m6, b00 delays: -
  m7, b00: |00000000000000000000000000000000| delays: -
  m7, b01: |00000000000000000000000000000000| delays: -
  m7, b02: |00000000000000000000000000000000| delays: -
  m7, b03: |00000000000000000000000000000000| delays: -
  m7, b04: |00000000000000000000000000000000| delays: -
  m7, b05: |00000000000000000000000000000000| delays: -
  m7, b06: |00000000000000000000000000000000| delays: -
  m7, b07: |00000000000000000000000000000000| delays: -
  best: m7, b00 delays: -
Switching SDRAM to hardware control.
Memtest at 0x40000000 (2.0MiB)...
  Write: 0x40000000-0x40200000 2.0MiB     
   Read: 0x40000000-0x40200000 2.0MiB     
  bus errors:  192/256
  addr errors: 0/8192
  data errors: 524288/524288
Memtest KO
Memory initialization failed

--============= Console ================--

litex> 
```
The problem may come for both board variants and build process.
I think we should check the board first.

offnaria, 2023-03-10:
Thank you so much @vietthanh85 !
Yesterday, I found another AU200 board in my organisation and test with it.
@vietthanh85 's bitstream works correct but mine didn't.
So I agree with you, both my build process and board variants may have problem.

offnaria, 2023-04-13:
From the last comment (on Mar 10), I tried to generate correct bitstream with several environments.
I want you to tell me how your environment (gcc version, OS, LiteX version, dependencies, ...) is, because my bitstream still not pass the memory test on AU200 boards (I found a board which can pass the test with @vietthanh85 's bitstream).
Could you tell me how your environment is and how did you set up it?

vietthanh85, 2023-04-16:
@offnaria Sorry for the late reply.

Maybe I found the cause.
Could you add `cmd_latency = 1` to file `litex_boards/targets/xilinx_alveo_u200.py` as follows?
(It should be similar as `xilinx_alveo_u280.py`)

```
            self.ddrphy = usddrphy.USPDDRPHY(platform.request("ddram"),
                memtype          = "DDR4",
                sys_clk_freq     = sys_clk_freq,
                cmd_latency      = 1,
                iodelay_clk_freq = 500e6,
                is_rdimm         = True)
```

I reproduced in a fresh environment with the following versions and above modification.

```
OS: Ubuntu 22.04.1 LTS 
litex: b367c27191511f36b10ec4103198978f86f9502c (latest now)
litex-boards: f5e51d72bca6ed0325c1213791a78362326002f8 (latest now)
litex-dram: e452da7a000814cc427ab5dd1b4924230a17cd5e (latest now)
riscv64-unknown-elf-gcc (g1ea978e3066) 12.1.0
```

It works well as follows:
```
        __   _ __      _  __
       / /  (_) /____ | |/_/
      / /__/ / __/ -_)>  <
     /____/_/\__/\__/_/|_|
   Build your hardware, easily!

 (c) Copyright 2012-2023 Enjoy-Digital
 (c) Copyright 2007-2015 M-Labs

 BIOS built on Apr 16 2023 02:20:53
 BIOS CRC passed (044e4aaf)

 LiteX git sha1: b367c271

--=============== SoC ==================--
CPU:		VexRiscv @ 125MHz
BUS:		WISHBONE 32-bit @ 4GiB
CSR:		32-bit data
ROM:		128.0KiB
SRAM:		8.0KiB
L2:		8.0KiB
SDRAM:		16.0GiB 64-bit @ 1000MT/s (CL-9 CWL-9)
MAIN-RAM:	1.0GiB

--========== Initialization ============--
Initializing SDRAM @0x40000000...
Switching SDRAM to software control.
Write leveling:
  tCK equivalent taps: 612
  Cmd/Clk scan (0-306)
  |00111  |000000111  |000000001  |000000000| best: 72
  Setting Cmd/Clk delay to 72 taps.
  Data scan:
  m0: |00111111111111111111100| delay: 32
  m1: |11111111111111111110000| delay: 00
  m2: |00000111111111111111111| delay: 80
  m3: |11111111111111110000000| delay: 00
  m4: |11111111100000000000000| delay: -
  m5: |11111111111100000000000| delay: 00
  m6: |00000111111111111111111| delay: 72
  m7: |11111111111111000000000| delay: 00
Write latency calibration:
m0:6 m1:6 m2:6 m3:6 m4:6 m5:6 m6:6 m7:6 
[... 69 lines cut ...]
  m7, b05: |00000000000000000000000000000000| delays: -
  m7, b06: |00000000000000000000000000000000| delays: -
  m7, b07: |00000000000000000000000000000000| delays: -
  best: m7, b03 delays: 207+-133
Switching SDRAM to hardware control.
Memtest at 0x40000000 (2.0MiB)...
  Write: 0x40000000-0x40200000 2.0MiB     
   Read: 0x40000000-0x40200000 2.0MiB     
Memtest OK
Memspeed at 0x40000000 (Sequential, 2.0MiB)...
  Write speed: 108.8MiB/s
   Read speed: 93.6MiB/s

--============== Boot ==================--
Booting from serial...
Press Q or ESC to abort boot completely.
sL5DdSMmkekro
             Timeout
No boot medium found

--============= Console ================--

litex> 

```

offnaria, 2023-04-16:
Thank you so much @vietthanh85 !
I'll try it ASAP!

offnaria, 2023-04-17:
By adding `cmd_latency = 1` in the code, the memtest was passed on my AU200!

```
self.ddrphy = usddrphy.USPDDRPHY(platform.request("ddram"),
    memtype          = "DDR4",
    sys_clk_freq     = sys_clk_freq,
    cmd_latency      = 1,
    iodelay_clk_freq = 500e6,
    is_rdimm         = True)
```

```
        __   _ __      _  __
       / /  (_) /____ | |/_/
      / /__/ / __/ -_)>  <
     /____/_/\__/\__/_/|_|
   Build your hardware, easily!

 (c) Copyright 2012-2023 Enjoy-Digital
 (c) Copyright 2007-2015 M-Labs

 BIOS built on Apr 16 2023 23:29:32
 BIOS CRC passed (0c7a24e4)

 LiteX git sha1: fb1cd22a

--=============== SoC ==================--
CPU:            VexRiscv @ 125MHz
BUS:            WISHBONE 32-bit @ 4GiB
CSR:            32-bit data
ROM:            128.0KiB
SRAM:           8.0KiB
L2:             8.0KiB
SDRAM:          16.0GiB 64-bit @ 1000MT/s (CL-9 CWL-9)
MAIN-RAM:       1.0GiB

--========== Initialization ============--
Initializing SDRAM @0x40000000...
Switching SDRAM to software control.
Write leveling:
  tCK equivalent taps: 612
  Cmd/Clk scan (0-306)
  |00111  |000000111  |000000000  |000000000| best: 65
  Setting Cmd/Clk delay to 65 taps.
  Data scan:
  m0: |00011111111111111111110| delay: 40
  m1: |01111111111111111111000| delay: 02
  m2: |00001111111111111111111| delay: 57
  m3: |11111111111111100000000| delay: 00
  m4: |11111111100000000000000| delay: -
  m5: |11111111111100000000000| delay: 00
  m6: |00000011111111111111111| delay: 83
  m7: |11111111111110000000000| delay: 00
Write latency calibration:
m0:6
m1:6
m2:6
m3:6
m4:6
m5:6
m6:6
m7:6
Read leveling:
  m0, b00: |00000000000000000000000000000000| delays: -
  m0, b01: |00000000000000000000000000000000| delays: -
  m0, b02: |00000000000000000000000000000000| delays: -
  m0, b03: |11111111111111000000000000000000| delays: 106+-106
  m0, b04: |00000000000000001111111111111111| delays: 379+-131
  m0, b05: |00000000000000000000000000000000| delays: -
  m0, b06: |00000000000000000000000000000000| delays: -
[... 60 lines cut ...]
  m7, b04: |00000000000000000000000001111111| delays: 453+-58
  m7, b05: |00000000000000000000000000000000| delays: -
  m7, b06: |00000000000000000000000000000000| delays: -
  m7, b07: |00000000000000000000000000000000| delays: -
  best: m7, b03 delays: 222+-134
Switching SDRAM to hardware control.
Memtest at 0x40000000 (2.0MiB)...
  Write: 0x40000000-0x40200000 2.0MiB
   Read: 0x40000000-0x40200000 2.0MiB
Memtest OK
Memspeed at 0x40000000 (Sequential, 2.0MiB)...
  Write speed: 108.8MiB/s
   Read speed: 93.6MiB/s

--============== Boot ==================--
Booting from serial...
Press Q or ESC to abort boot completely.
sL5DdSMmkekro
Timeout
No boot medium found

--============= Console ================--

litex>
```

offnaria, 2023-04-17:
By adding `cmd_latency = 1` in the code, the memtest was passed on my AU200!
It is build in my existing environment...
```
OS: Ubuntu 22.04.2 LTS
Vivado v2022.2 (64-bit)
litex: fb1cd22ac2fc97300f0e7bb2d5bf08c59558a619
litex-boards: bf4f6881641c668ae48cadb92bb0638cad53f59b
litedram: 6c8df7cc7be2f88e2d78d6282ada4213918de477
riscv64-unknown-elf-gcc () 12.2.0
```

The code I've editted in `xilinx_alveo_u200.py` is below.
```
self.ddrphy = usddrphy.USPDDRPHY(platform.request("ddram"),
    memtype          = "DDR4",
    sys_clk_freq     = sys_clk_freq,
    cmd_latency      = 1,
    iodelay_clk_freq = 500e6,
    is_rdimm         = True)
```

and the result is...
```
        __   _ __      _  __
       / /  (_) /____ | |/_/
      / /__/ / __/ -_)>  <
     /____/_/\__/\__/_/|_|
   Build your hardware, easily!

 (c) Copyright 2012-2023 Enjoy-Digital
 (c) Copyright 2007-2015 M-Labs

 BIOS built on Apr 16 2023 23:29:32
 BIOS CRC passed (0c7a24e4)

 LiteX git sha1: fb1cd22a

--=============== SoC ==================--
CPU:            VexRiscv @ 125MHz
BUS:            WISHBONE 32-bit @ 4GiB
CSR:            32-bit data
ROM:            128.0KiB
SRAM:           8.0KiB
L2:             8.0KiB
SDRAM:          16.0GiB 64-bit @ 1000MT/s (CL-9 CWL-9)
MAIN-RAM:       1.0GiB

--========== Initialization ============--
Initializing SDRAM @0x40000000...
Switching SDRAM to software control.
Write leveling:
  tCK equivalent taps: 612
  Cmd/Clk scan (0-306)
  |00111  |000000111  |000000000  |000000000| best: 65
  Setting Cmd/Clk delay to 65 taps.
  Data scan:
  m0: |00011111111111111111110| delay: 40
  m1: |01111111111111111111000| delay: 02
  m2: |00001111111111111111111| delay: 57
  m3: |11111111111111100000000| delay: 00
  m4: |11111111100000000000000| delay: -
  m5: |11111111111100000000000| delay: 00
  m6: |00000011111111111111111| delay: 83
  m7: |11111111111110000000000| delay: 00
Write latency calibration:
m0:6
m1:6
m2:6
m3:6
m4:6
[... 71 lines cut ...]
  m7, b04: |00000000000000000000000001111111| delays: 453+-58
  m7, b05: |00000000000000000000000000000000| delays: -
  m7, b06: |00000000000000000000000000000000| delays: -
  m7, b07: |00000000000000000000000000000000| delays: -
  best: m7, b03 delays: 222+-134
Switching SDRAM to hardware control.
Memtest at 0x40000000 (2.0MiB)...
  Write: 0x40000000-0x40200000 2.0MiB
   Read: 0x40000000-0x40200000 2.0MiB
Memtest OK
Memspeed at 0x40000000 (Sequential, 2.0MiB)...
  Write speed: 108.8MiB/s
   Read speed: 93.6MiB/s

--============== Boot ==================--
Booting from serial...
Press Q or ESC to abort boot completely.
sL5DdSMmkekro
Timeout
No boot medium found

--============= Console ================--

litex>
```

offnaria, 2023-04-17:
Well, I have a question.
@vietthanh85 , had you build your first bitstream of Feb 23 without `cmd_latency = 1` option and passed the memtest?
@lapnd , did you do so?

If so, I don't know why my past bitstreams never pass the test...

vietthanh85, 2023-04-17:
@offnaria 
> @vietthanh85 , had you build your first bitstream of Feb 23 without cmd_latency = 1 option and passed the memtest?

I don't think so, I think that bitstream was built with `cmd_latency = 1`.
Did you pass the test with `cmd_latency = 1`?

offnaria, 2023-04-17:
@vietthanh85 
Yes I passed it with `cmd_latency = 1`.

offnaria, 2023-04-19:
Thank you for the update.
Well, I faced with a new problem.
When I select **vexriscv_smp** as the CPU, the memtest fails again.

This is the output when I select vexriscv_smp.
(Built with `cmd_latency = 1` and `--cpu-type vexriscv_smp` option)
```
        __   _ __      _  __
       / /  (_) /____ | |/_/
      / /__/ / __/ -_)>  <
     /____/_/\__/\__/_/|_|
   Build your hardware, easily!

 (c) Copyright 2012-2023 Enjoy-Digital
 (c) Copyright 2007-2015 M-Labs

 BIOS built on Apr 19 2023 12:54:46
 BIOS CRC passed (8d92b136)

 LiteX git sha1: fb1cd22a

--=============== SoC ==================--
CPU:            VexRiscv SMP-STANDARD @ 125MHz
BUS:            WISHBONE 32-bit @ 4GiB
CSR:            32-bit data
ROM:            128.0KiB
SRAM:           8.0KiB
L2:             8.0KiB
SDRAM:          16.0GiB 64-bit @ 1000MT/s (CL-9 CWL-9)
MAIN-RAM:       1.0GiB

--========== Initialization ============--
Initializing SDRAM @0x40000000...
Switching SDRAM to software control.
Write leveling:
  tCK equivalent taps: 612
  Cmd/Clk scan (0-306)
  |00111  |000001111  |000011111  |010111111| best: 77
  Setting Cmd/Clk delay to 77 taps.
  Data scan:
  m0: |00011111111111111111110| delay: 43
  m1: |01111111111111111111000| delay: 06
  m2: |00000111111111111111111| delay: 79
  m3: |11111111111111110000000| delay: 00
  m4: |11111111110000000000000| delay: -
  m5: |11111111111110000000000| delay: 00
  m6: |00000011111111111111111| delay: 87
  m7: |11111111111111000000000| delay: 00
Write latency calibration:
m0:6
m1:6
m2:6
m3:6
m4:6
m5:6
m6:6
m7:6
Read leveling:
  m0, b00: |00000000000000000000000000000000| delays: -
  m0, b01: |00000000000000000000000000000000| delays: -
  m0, b02: |00000000000000000000000000000000| delays: -
  m0, b03: |11111111111110000000000000000000| delays: 105+-105
  m0, b04: |00000000000000001111111111111111| delays: 378+-133
  m0, b05: |00000000000000000000000000000000| delays: -
  m0, b06: |00000000000000000000000000000000| delays: -
  m0, b07: |00000000000000000000000000000000| delays: -
  best: m0, b04 delays: 378+-132
  m1, b00: |00000000000000000000000000000000| delays: -
  m1, b01: |00000000000000000000000000000000| delays: -
  m1, b02: |00000000000000000000000000000000| delays: -
  m1, b03: |00111111111111111110000000000000| delays: 154+-134
  m1, b04: |00000000000000000000111111111111| delays: 416+-95
  m1, b05: |00000000000000000000000000000000| delays: -
  m1, b06: |00000000000000000000000000000000| delays: -
  m1, b07: |00000000000000000000000000000000| delays: -
  best: m1, b03 delays: 153+-136
  m2, b00: |00000000000000000000000000000000| delays: -
  m2, b01: |00000000000000000000000000000000| delays: -
  m2, b02: |00000000000000000000000000000000| delays: -
  m2, b03: |11111000000000000000000000000000| delays: 32+-32
  m2, b04: |00000001111111111111111000000000| delays: 236+-128
  m2, b05: |00000000000000000000000000111111| delays: 461+-50
  m2, b06: |00000000000000000000000000000000| delays: -
  m2, b07: |00000000000000000000000000000000| delays: -
  best: m2, b04 delays: 237+-129
  m3, b00: |00000000000000000000000000000000| delays: -
  m3, b01: |00000000000000000000000000000000| delays: -
  m3, b02: |00000000000000000000000000000000| delays: -
  m3, b03: |00000000000000000000000000000000| delays: -
  m3, b04: |00001111111111111110000000000000| delays: 176+-123
  m3, b05: |00000000000000000000001111111111| delays: 431+-80
  m3, b06: |00000000000000000000000000000000| delays: -
  m3, b07: |00000000000000000000000000000000| delays: -
  best: m3, b04 delays: 175+-122
  m4, b00: |00000000000000000000000000000000| delays: -
  m4, b01: |00000000000000000000000000000000| delays: -
  m4, b02: |00000000000000000000000000000000| delays: -
  m4, b03: |11111111111111110000000000000000| delays: 123+-123
  m4, b04: |00000000000000000001111111111111| delays: 400+-110
  m4, b05: |00000000000000000000000000000000| delays: -
  m4, b06: |00000000000000000000000000000000| delays: -
  m4, b07: |00000000000000000000000000000000| delays: -
  best: m4, b03 delays: 125+-125
  m5, b00: |00000000000000000000000000000000| delays: -
  m5, b01: |00000000000000000000000000000000| delays: -
  m5, b02: |00000000000000000000000000000000| delays: -
  m5, b03: |11110000000000000000000000000000| delays: 27+-27
  m5, b04: |00000011111111111111111000000000| delays: 225+-127
  m5, b05: |00000000000000000000000000111111| delays: 457+-53
  m5, b06: |00000000000000000000000000000000| delays: -
  m5, b07: |00000000000000000000000000000000| delays: -
  best: m5, b04 delays: 226+-132
  m6, b00: |00000000000000000000000000000000| delays: -
  m6, b01: |00000000000000000000000000000000| delays: -
  m6, b02: |00000000000000000000000000000000| delays: -
  m6, b03: |11111111110000000000000000000000| delays: 75+-75
  m6, b04: |00000000000001111111111111111000| delays: 330+-126
  m6, b05: |00000000000000000000000000000000| delays: 508+-03
  m6, b06: |00000000000000000000000000000000| delays: -
  m6, b07: |00000000000000000000000000000000| delays: -
  best: m6, b04 delays: 330+-125
  m7, b00: |00000000000000000000000000000000| delays: -
  m7, b01: |00000000000000000000000000000000| delays: -
  m7, b02: |11100000000000000000000000000000| delays: 15+-15
  m7, b03: |00000111111111111111110000000000| delays: 205+-136
  m7, b04: |00000000000000000000000011111111| delays: 442+-69
  m7, b05: |00000000000000000000000000000000| delays: -
  m7, b06: |00000000000000000000000000000000| delays: -
  m7, b07: |00000000000000000000000000000000| delays: -
  best: m7, b03 delays: 204+-136
Switching SDRAM to hardware control.
Memtest at 0x40000000 (2.0MiB)...
  Write: 0x40000000-0x40200000 2.0MiB
   Read: 0x40000000-0x40200000 2.0MiB
  bus errors:  0/256
  addr errors: 0/8192
  data errors: 491520/524288
Memtest KO
Memory initialization failed

--============= Console ================--

litex>
```

Is this problem related to our first problem?
Also, can anyone check this phenomenon on your environment?

offnaria, 2023-04-19:
With vexriscv (not vexriscv_smp) of `standard`, `minimal`, `lite`, `full` configurations, the memtest was passed.
(Built with `cmd_latency = 1` and `--cpu-variant` option)

vietthanh85, 2023-04-19:
@offnaria 
> When I select vexriscv_smp as the CPU, the memtest fails again.

I'll check and get back to you.

vietthanh85, 2023-04-20:
@offnaria 
For VexRiscv SMP, I got the same result as yours (built with `cmd_latency = 1`)
I am not clear the reason so far:

```
        __   _ __      _  __
       / /  (_) /____ | |/_/
      / /__/ / __/ -_)>  <
     /____/_/\__/\__/_/|_|
   Build your hardware, easily!

 (c) Copyright 2012-2023 Enjoy-Digital
 (c) Copyright 2007-2015 M-Labs

 BIOS built on Apr 20 2023 00:33:14
 BIOS CRC passed (1672158a)

 LiteX git sha1: b367c271

--=============== SoC ==================--
CPU:            VexRiscv SMP-STANDARD @ 125MHz
BUS:            WISHBONE 32-bit @ 4GiB
CSR:            32-bit data
ROM:            128.0KiB
SRAM:           8.0KiB
L2:             8.0KiB
SDRAMm:         16.0GiB 64-bit @ 1000MT/s (CL-9 CWL-9)
MAIN-RAM:       1.0GiB

--========== Initialization ============--
Initializing SDRAM @0x40000000...
Switching SDRAM to software control.
Write leveling:
  tCK equivalent taps: 612
  Cmd/Clk scan (0-306)
  |00111  |000000111  |000000000  |000000000| best: 73
  Setting Cmd/Clk delay to 73 taps.
  Data scan:
  m0: |000111111111111111100| delay: 33
  m1: |1111111111111111110000| delay: 00
  m2: |00000111111111111111111| delay: 79
  m3: |11111111111111110000| delay: 00
  m4: |111110000000| delay: -
  m5: |111111111100000| delay: 00
  m6: |00000111 delay: 72
  m7: |1111111000 delay: 00
Write latency calibration:
m0:6 m1:6 m2:6 m3:6 m4:6 m5:6 m6:6 m7:6 
Read leveling:
  m0, b00: |00000000000000000000000000000000| delays: -
  m0, b01: |00000000000000000000000000000000| delays: -
  m0, b02: |00000000000000000000000000000000| delays: -
  m0, b03: |11111111111111000000000000000000| delays: 107+-107
  m0, b04: |00000000000000001111111111111111| delays: 382+-126
  m0, b05: |0000000000000000000000000000000| delays: -
  m0, b06: |0000000000000000000000000000000| delays: -
  m0, b07: |0000000000000000000000000000000| delays: -
  best: m0, b04 delays: 383+-127
  m1, b00: |000000000000000000000000000000| delays: -
  m1, b01: |0000000000000000000000000000000| delays: -
  m1, b02: |0000000000000000000000000000000| delays: -
  m1, b03: |01111111111111111100000000000000| delays: 151+-135
  m1, b04: |00000000000000000000011111111111| delays: 416+-95
  m1, b05: |00000000000000000000000000000000| delays: -
  m1, b06: |0000000000000000000000000000000| delays: -
  m1, b07: |00000000000000000000000000000000| delays: -
  best: m1, b03 delays: 150+-134
  m2, b00: |00000000000000000000000000000000| delays: -
  m2, b01: |00000000000000000000000000000000| delays: -
  m2, b02: |000000000000000000000000000000| elays: -
  m2, b03: |111110000000000000000000000000| delays: 34+-34
  m2, b04: |000000111111111111111110000000| elays: 242+-134
  m2, b05: |00000000000000000000000111111| elays: 462+-49
  m2, b06: |0000000000000000000000000000| delays: -
  m2, b07: |00000000000000000000000000000000| delays: -
  best: m2, b04 delays: 243+-135
  m3, b00: |00000000000000000000000000000000| elays: -
  m3, b01: |00000000000000000000000000000000| delays: -
  m3, b02: |000000000000000000000000000000| delays: -
  m3, b03: |00000000000000000000000000000000| delays: -
  m3, b04: |00001111111111111111000000000000| delays: 181+-126
  m3, b05: |0000000000000000000000111111111| delays: 436+-74
  m3, b06: |0000000000000000000000000000000| delays: -
  m3, b07: |00000000000000000000000000000000 delays: -
  best: m3, b04 delays: 182+-125
  m4, b00: |000000000000000000000000000000| delays: -
  m4, b01: |0000000000000000000000000000| delays: -
  m4, b02: |0000000000000000000000000000000| delays: -
  m4, b03: |1111111111111111100000000000000| delays: 128+-128
  m4, b04: |0000000000000000000111111111111| delays: 405+-106
  m4, b05: |00000000000000000000000000000000| delays: -
  m4, b06: |0000000000000000000000000000000| delays: -
  m4, b07: |000000000000000000000000000000| delays: -
  best: m4, b03 delays: 128+-128
  m5, b00: |00000000000000000000000000000000| delays: -
  m5, b01: |00000000000000000000000000000000| delays: -
  m5, b02: |00000000000000000000000000000000| delays: -
  m5, b03: |11110000000000000000000000000000| delays: 28+-28
  m5, b04: |00000000111111111111111100000000| delays: 241+-126
  m5, b05: |00000000000000000000000000011111| delays: 464+-46
  m5, b06: |00000000000000000000000000000000| delays: -
  m5, b07: |00000000000000000000000000000000| delays: -
  best: m5, b04 delays: 241+-127
  m6, b00: |000000000000000000000000000000 delays: -
  m6, b01: |00000000000000000000000000| delays: -
  m6, b0: |000000000000000000000000000000| delays: -
  m6, b03: |1111111111100000000000000000000| delays: 84+-84
  m6, b04: |0000000000001111111111111111100| delays: 338+-134
  m6, b05: |00000000000000000000000000000000| delays: 510+-00
  m6, b06: |00000000000000000000000000000000| delays: -
  m6, b07: |00000000000000000000000000000000| delays: -
  best: m6, b04 elays: 338+-134
  m7, b00: |00000000000000000000000000000000| elays: -
  m7, b01: |0000000000000000000000000000000| delays: -
  m7, b02: |11100000000000000000000000000000| delays: 17+-17
  m7, b03: |00000111111111111111110000000000| delays: 205+-131
  m7, b04: |00000000000000000000000011111111| delays: 444+-66
  m7, b05: |0000000000000000000000000000000| delays: -
  m7, b06: |00000000000000000000000000000000| delays: -
  m7, b07: |00000000000000000000000000000000| delays: -
  best: m7, b03 delays: 206+-132
Switching SDRAM to hardware control.
Memtest at 0x40000000 (2.0MiB)...
  Write: 0x40000000-0x40200000 2.0MiB     
   Read: 0x40000000-0x40200000 2.0MiB     
  bus errors:  0/256
  addr errors: 0/8192
  data errors: 491520/524288
Memory initialization failed

--============= Console ================--

litex> 

```

offnaria, 2023-04-20:
@vietthanh85 
Thank you for the verification.
I tried `serv` and `ibex` as the CPU and they passed the memtest.
I'm not sure but I think these data errors come from vexriscv_smp's implementation.

offnaria, 2023-04-24:
Thanks to the issue #1652 , I found a solution for the problem of vexriscv_smp.
We should build the SoC with `--with-wishbone-memory` option like:
`xilinx_alveo_u200.py --build--cpu-type vexriscv_smp --output-dir $MY_OUTPUT --with-wishbone-memory`

This is the log with the option (and also `cmd_latency = 1`).
```
        __   _ __      _  __
       / /  (_) /____ | |/_/
      / /__/ / __/ -_)>  <
     /____/_/\__/\__/_/|_|
   Build your hardware, easily!

 (c) Copyright 2012-2023 Enjoy-Digital
 (c) Copyright 2007-2015 M-Labs

 BIOS built on Apr 24 2023 15:33:55
 BIOS CRC passed (8c93966f)

 LiteX git sha1: fb1cd22a

--=============== SoC ==================--
CPU:            VexRiscv SMP-STANDARD @ 125MHz
BUS:            WISHBONE 32-bit @ 4GiB
CSR:            32-bit data
ROM:            128.0KiB
SRAM:           8.0KiB
L2:             8.0KiB
SDRAM:          16.0GiB 64-bit @ 1000MT/s (CL-9 CWL-9)
MAIN-RAM:       1.0GiB

--========== Initialization ============--
Initializing SDRAM @0x40000000...
Switching SDRAM to software control.
Write leveling:
  tCK equivalent taps: 612
  Cmd/Clk scan (0-306)
  |00111  |000000111  |000000000  |000000000| best: 70
  Setting Cmd/Clk delay to 70 taps.
  Data scan:
  m0: |00011111111111111111110| delay: 35
  m1: |11111111111111111111000| delay: 00
  m2: |00000111111111111111111| delay: 72
  m3: |11111111111111110000000| delay: 00
  m4: |11111111110000000000000| delay: -
  m5: |11111111111100000000000| delay: 00
  m6: |00000111111111111111111| delay: 79
  m7: |11111111111111000000000| delay: 00
Write latency calibration:
m0:6
m1:6
m2:6
m3:6
m4:6
m5:6
m6:6
m7:6
Read leveling:
  m0, b00: |00000000000000000000000000000000| delays: -
  m0, b01: |00000000000000000000000000000000| delays: -
  m0, b02: |00000000000000000000000000000000| delays: -
  m0, b03: |11111111111111000000000000000000| delays: 108+-108
  m0, b04: |00000000000000001111111111111111| delays: 382+-128
  m0, b05: |00000000000000000000000000000000| delays: -
  m0, b06: |00000000000000000000000000000000| delays: -
  m0, b07: |00000000000000000000000000000000| delays: -
  best: m0, b04 delays: 382+-129
  m1, b00: |00000000000000000000000000000000| delays: -
  m1, b01: |00000000000000000000000000000000| delays: -
  m1, b02: |00000000000000000000000000000000| delays: -
  m1, b03: |00111111111111111110000000000000| delays: 159+-132
[... 54 lines cut ...]
  m7, b04: |00000000000000000000000011111111| delays: 446+-65
  m7, b05: |00000000000000000000000000000000| delays: -
  m7, b06: |00000000000000000000000000000000| delays: -
  m7, b07: |00000000000000000000000000000000| delays: -
  best: m7, b03 delays: 212+-136
Switching SDRAM to hardware control.
Memtest at 0x40000000 (2.0MiB)...
  Write: 0x40000000-0x40200000 2.0MiB
   Read: 0x40000000-0x40200000 2.0MiB
Memtest OK
Memspeed at 0x40000000 (Sequential, 2.0MiB)...
  Write speed: 93.7MiB/s
   Read speed: 78.0MiB/s

--============== Boot ==================--
Booting from serial...
Press Q or ESC to abort boot completely.
sL5DdSMmkekro
Timeout
No boot medium found

--============= Console ================--

litex>
```

enjoy-digital, 2023-04-24:
Nice collaboration everyone! I was indeed going to suggest `--with-wishbone-memory` but great that you found it before. I'll still try to see if the limitation is expected, if so, we should prevent building the SoC and if not, fixing the issue.