enjoy-digital/litex #1609: FT245PHYSynchronous how to use it ?

Prigana, 2023-02-22:
Hello, I have a problem with FT245PHYSynchronous fonctions from "litex.soc.cores.usb_fifo".

I connected my FT2232 chip to my Butterstick board (FPGA ECP5). Then I configured my inputs and outputs like this example: https://github.com/litex-hub/litex-boards/blob/a8c92cd86ffecde9b0c8302f18353c00a0c894b1/litex_boards/platforms/limesdr_mini_v2.py#:~:text=%22LVCMOS25%22))%2C-,%23%20USB%2DFIFO.,)%2C,-%23%20RF%2DIC%20/%20LMS7002M 

[image]

Then I wanted to do an echo: send a signal PC -> FT2232 -> FPGA -> FT2232 -> PC but it did not work. I reused this part:
https://github.com/litex-hub/litex-boards/blob/a8c92cd86ffecde9b0c8302f18353c00a0c894b1/litex_boards/targets/limesdr_mini_v2.py#:~:text=%22i2c%22))-,%23%20USB%2DFIFO%20%2D%2D%2D%2D%2D%2D%2D%2D%2D%2D%2D%2D%2D%2D%2D%2D%2D%2D%2D%2D%2D%2D%2D%2D%2D%2D%2D%2D%2D%2D%2D%2D%2D%2D%2D%2D%2D%2D%2D%2D%2D%2D%2D%2D%2D%2D%2D%2D%2D%2D%2D%2D%2D%2D%2D%2D%2D%2D%2D%2D%2D%2D%2D%2D%2D%2D%2D%2D%2D%2D%2D%2D%2D%2D%2D%2D%2D%2D%2D%2D%2D,%5D,-else%3A
In my litex code, it looks like this :
[image]
My data coming from my FT2232 chip, only the litex part is not working, could you help me to understand how to use this module?

enjoy-digital, 2023-02-23:
Hello @Prigana,

I would indeed recommend using the LimeSDRMini integration as a  basis and use the JTAGBone bridge to have more visibility on what's going on:
1. Add a JTAGBone bridge to your design and `--csr-csv=csr.csv` to your build command line.
2. Start LiteX Server: `litex_server --jtag --jtag-config=openocd_limesdr_mini_v2.cfg` on the LimeSDR, possible you'll have to create a specific configuration for your case with the FT2232.
3. Verify JTAGBone operation: `litex_cli --regs` (will dumps the registers of the design).
4. Add LiteScope to your design and rebuild: https://github.com/litex-hub/litex-boards/blob/master/litex_boards/targets/limesdr_mini_v2.py#L115-L121
5. Use LiteScope (litescope_cli --help) to trigger on the signals you want to observe, initiate a transfer from the PC and analyse the traces.

trabucayre, 2023-02-23:
Hi,
According to the [ft2232 datasheet](https://ftdichip.com/wp-content/uploads/2020/07/DS_FT2232H.pdf) section 4.4 FT2232's interface used must be configured, at the eeprom level, to act as a FIFO.

If your converter isn't configured as required this change may be done with the proprietary tool or with a bit of c + *libftdi*

```
ret = ftdi_set_eeprom_value(ftdi, CHANNEL_X_TYPE, CHANNEL_IS_FIFO);
```

where `CHANNEL_X_TYPE` may be `CHANNEL_A_TYPE` or `CHANNEL_B_TYPE`

[This example](https://github.com/trabucayre/fixFT2232_ecp5evn/blob/master/fixFT2232_ecp5evn.c#L109) may help to see how to change this unique parameter.

enjoy-digital, 2023-02-23:
Indeed @trabucayre, in case you have a Windows computer at hand, you can also use `FT_PROG`.

trabucayre, 2023-02-23:
@enjoy-digital thanks! By "*proprietary tool*" it's was `FT_PROG` in my mind (but unable to remember name :))

Prigana, 2023-02-24:
THANK YOU ! I just added a processor to my SoC and it worked although some of the data is still not totally good. Is the module FT245PHYSynchronous  usable in a SoC without processor?