enjoy-digital/litex #1610: Auto end LiteX simulation

hardiselg, 2023-02-22:
Hello,

I am running trace collection simulation experiments on a cv32e40p core.
My issue is that I do not want the simulation session to remain open after the trace generation program has ended (return 0).

Would it be possible to add some macro calls to the C code or other environmental variables elsewhere to close the simulation session automatically?

I know that there is a function called  "litex_sim_got_finish" defined in veril.cpp that is supposed to signal Verilator to close the simulation //If I understood this correctly.
Would it be possible to call that function for example in the end of the "[demo](https://github.com/enjoy-digital/litex/tree/master/litex/soc/software/demo)" program?

enjoy-digital, 2023-02-23:
Hello,

if you run the simulation with `--sim-debug`, you'll then have access to debug CSRs and adding `sim_finish_finish_write(1);` in your C code will allow you to finish the simulation.

For more info, you can have a closer look at [sim_debug.c](https://github.com/enjoy-digital/litex/blob/master/litex/soc/software/bios/sim_debug.c).

hardiselg, 2023-02-23:
Thank you!

`sim_finish_finish_write(1);` in combination with ``--sim-debug`` work perfectly!

closed 2023-02-23 by hardiselg