enjoy-digital/litex #1611: BlackParrot not working

YiminGao0113, 2023-02-23:
Hi,

I wanted to experiment with BlackParrot a bit. First, I test using the Xilinx VC707 board the bitstream didn't work, then I simply use the litex_sim with cpu-type = blackparrot. It didn't work as well. 
[image]

I got stuck here in simulation, and observed nothing on VC707 board. Is the blackparrot currently down in LiteX?
Thanks.

developandplay, 2023-02-23:
Unfortunately yes it's broken.
I was the last maintainer of the project but didn't have the time to keep
up with the changes in Litex lately.
I was planning to look into it but probably it won't be too soon.

developandplay, 2023-03-26:
Hi @YiminGao0113 sorry for looking into it so late.
This part of Blackparrot is not broken for me. It was just the Linux boot...
You can get succesfully get to the console if you press Q or ESC.

```

        __   _ __      _  __
       / /  (_) /____ | |/_/
      / /__/ / __/ -_)>  <
     /____/_/\__/\__/_/|_|
   Build your hardware, easily!

 (c) Copyright 2012-2023 Enjoy-Digital
 (c) Copyright 2007-2015 M-Labs

 BIOS built on Mar 26 2023 18:32:22
 BIOS CRC passed (41de9955)

 LiteX git sha1: 4b72dd04

--=============== SoC ==================--
CPU:            BlackParrotRV64[imafd] @ 1MHz
BUS:            WISHBONE 32-bit @ 4GiB
CSR:            32-bit data
ROM:            128.0KiB
SRAM:           8.0KiB

--============== Boot ==================--
Booting from serial...
Press Q or ESC to abort boot completely.
sL5DdSMmkekro
Cancelled

--============= Console ================--

litex> 
```

developandplay, 2023-03-26:
There is however an issue with the timer such that it does not time out quickly like the vexriscv version for example does.

developandplay, 2023-03-26:
@YiminGao0113 Can you try if you still can't reproduce it with pressing ESC or Q?