enjoy-digital/litex #1612: Hardware interrupt handling functions not all generated in csr.h file

mohamedElbouazzati, 2023-02-23:
Hi @enjoy-digital, 

I'm using an external interrupt with my CPU named dio0, I'm trying to generate the required functions to handle my interrupt from the software. 
But the generated csr.h is limited to a couple of functions, not all the ones I'm used to generate with an older version of litex.
Do you have any idea about this situation?

Thank you
 
**Used in target :** 
```
 self.submodules.dio0 = GPIOIn(self.platform.request("dio0"),with_irq=True )
 self.irq.add("dio0")
```
**Generated csr.h :**
```
/* dio0 */
#define CSR_DIO0_BASE (CSR_BASE + 0x1800L)
#define CSR_DIO0_IN_ADDR (CSR_BASE + 0x1800L)
#define CSR_DIO0_IN_SIZE 1
static inline uint32_t dio0_in_read(void) {
	return csr_read_simple((CSR_BASE + 0x1800L));
}
#define CSR_DIO0_MODE_ADDR (CSR_BASE + 0x1804L)
#define CSR_DIO0_MODE_SIZE 1
static inline uint32_t dio0_mode_read(void) {
	return csr_read_simple((CSR_BASE + 0x1804L));
}
static inline void dio0_mode_write(uint32_t v) {
	csr_write_simple(v, (CSR_BASE + 0x1804L));
}
#define CSR_DIO0_EDGE_ADDR (CSR_BASE + 0x1808L)
#define CSR_DIO0_EDGE_SIZE 1
static inline uint32_t dio0_edge_read(void) {
	return csr_read_simple((CSR_BASE + 0x1808L));
}
static inline void dio0_edge_write(uint32_t v) {
	csr_write_simple(v, (CSR_BASE + 0x1808L));
}

/* identifier_mem */
```

**Desired csr.h( generated with a litex old version):**
```
/* dio0 */
#define CSR_DIO0_BASE (CSR_BASE + 0x0L)
#define CSR_DIO0_IN_ADDR (CSR_BASE + 0x0L)
#define CSR_DIO0_IN_SIZE 1
static inline uint32_t dio0_in_read(void) {
	return csr_read_simple(CSR_BASE + 0x0L);
}
#define CSR_DIO0_MODE_ADDR (CSR_BASE + 0x4L)
#define CSR_DIO0_MODE_SIZE 1
static inline uint32_t dio0_mode_read(void) {
	return csr_read_simple(CSR_BASE + 0x4L);
}
static inline void dio0_mode_write(uint32_t v) {
	csr_write_simple(v, CSR_BASE + 0x4L);
}
#define CSR_DIO0_EDGE_ADDR (CSR_BASE + 0x8L)
#define CSR_DIO0_EDGE_SIZE 1
static inline uint32_t dio0_edge_read(void) {
	return csr_read_simple(CSR_BASE + 0x8L);
}
static inline void dio0_edge_write(uint32_t v) {
	csr_write_simple(v, CSR_BASE + 0x8L);
}
#define CSR_DIO0_EV_STATUS_ADDR (CSR_BASE + 0xcL)
#define CSR_DIO0_EV_STATUS_SIZE 1
static inline uint32_t dio0_ev_status_read(void) {
	return csr_read_simple(CSR_BASE + 0xcL);
}
#define CSR_DIO0_EV_STATUS_I0_OFFSET 0
#define CSR_DIO0_EV_STATUS_I0_SIZE 1
static inline uint32_t dio0_ev_status_i0_extract(uint32_t oldword) {
	uint32_t mask = ((1 << 1)-1);
	return ( (oldword >> 0) & mask );
}
static inline uint32_t dio0_ev_status_i0_read(void) {
	uint32_t word = dio0_ev_status_read();
	return dio0_ev_status_i0_extract(word);
}
#define CSR_DIO0_EV_PENDING_ADDR (CSR_BASE + 0x10L)
#define CSR_DIO0_EV_PENDING_SIZE 1
static inline uint32_t dio0_ev_pending_read(void) {
	return csr_read_simple(CSR_BASE + 0x10L);
}
static inline void dio0_ev_pending_write(uint32_t v) {
	csr_write_simple(v, CSR_BASE + 0x10L);
}
#define CSR_DIO0_EV_PENDING_I0_OFFSET 0
#define CSR_DIO0_EV_PENDING_I0_SIZE 1
static inline uint32_t dio0_ev_pending_i0_extract(uint32_t oldword) {
	uint32_t mask = ((1 << 1)-1);
	return ( (oldword >> 0) & mask );
}
static inline uint32_t dio0_ev_pending_i0_read(void) {
	uint32_t word = dio0_ev_pending_read();
	return dio0_ev_pending_i0_extract(word);
}
static inline uint32_t dio0_ev_pending_i0_replace(uint32_t oldword, uint32_t plain_value) {
	uint32_t mask = ((1 << 1)-1);
	return (oldword & (~(mask << 0))) | (mask & plain_value)<< 0 ;
}
static inline void dio0_ev_pending_i0_write(uint32_t plain_value) {
	uint32_t oldword = dio0_ev_pending_read();
	uint32_t newword = dio0_ev_pending_i0_replace(oldword, plain_value);
	dio0_ev_pending_write(newword);
}
#define CSR_DIO0_EV_ENABLE_ADDR (CSR_BASE + 0x14L)
#define CSR_DIO0_EV_ENABLE_SIZE 1
static inline uint32_t dio0_ev_enable_read(void) {
	return csr_read_simple(CSR_BASE + 0x14L);
}
static inline void dio0_ev_enable_write(uint32_t v) {
	csr_write_simple(v, CSR_BASE + 0x14L);
}
#define CSR_DIO0_EV_ENABLE_I0_OFFSET 0
#define CSR_DIO0_EV_ENABLE_I0_SIZE 1
static inline uint32_t dio0_ev_enable_i0_extract(uint32_t oldword) {
	uint32_t mask = ((1 << 1)-1);
	return ( (oldword >> 0) & mask );
}
static inline uint32_t dio0_ev_enable_i0_read(void) {
	uint32_t word = dio0_ev_enable_read();
	return dio0_ev_enable_i0_extract(word);
}
static inline uint32_t dio0_ev_enable_i0_replace(uint32_t oldword, uint32_t plain_value) {
	uint32_t mask = ((1 << 1)-1);
	return (oldword & (~(mask << 0))) | (mask & plain_value)<< 0 ;
}
static inline void dio0_ev_enable_i0_write(uint32_t plain_value) {
	uint32_t oldword = dio0_ev_enable_read();
	uint32_t newword = dio0_ev_enable_i0_replace(oldword, plain_value);
	dio0_ev_enable_write(newword);
}

```

adamhlt, 2023-02-23:
Hi @mohamedElbouazzati
I have the same result as you when I try to add a GPIO IN, after looking a bit at gpio.py I noticed that modifying the file **gpio.py** like this should work :

```python
class _GPIOIRQ:
    def add_irq(self, in_pads):
        self._mode = CSRStorage(len(in_pads), description="GPIO IRQ Mode: 0: Edge, 1: Change.")
        self._edge = CSRStorage(len(in_pads), description="GPIO IRQ Edge (when in Edge mode): 0: Rising Edge, 1: Falling Edge.")

        # # #

        self.submodules.ev = EventManager()
        for n in range(len(in_pads)):
            in_pads_n_d = Signal()
            self.sync += in_pads_n_d.eq(in_pads[n])
            esp = EventSourceProcess(name=f"i{n}", edge="rising")
            self.comb += [
                # Change mode.
                If(self._mode.storage[n],
                    esp.trigger.eq(in_pads[n] ^ in_pads_n_d)
                # Edge mode.
                ).Else(
                    esp.trigger.eq(in_pads[n] ^ self._edge.storage[n])
                )
            ]
            setattr(self.ev, f"i{n}", esp)

            self.ev.finalize() # Add this line
```

I'm not used to using interrupts so I would like someone more expert to confirm my solution.

## Result : 

Target file : 

```python 
    soc.submodules.irq_test = GPIOIn(soc.platform.request("test"), with_irq=True)
    soc.irq.add("irq_test")
```

Generated csr.h :

```c
/* irq_test */
#define CSR_IRQ_TEST_BASE (CSR_BASE + 0x1000L)
#define CSR_IRQ_TEST_IN_ADDR (CSR_BASE + 0x1000L)
#define CSR_IRQ_TEST_IN_SIZE 1
static inline uint32_t irq_test_in_read(void) {
	return csr_read_simple((CSR_BASE + 0x1000L));
}
#define CSR_IRQ_TEST_MODE_ADDR (CSR_BASE + 0x1004L)
#define CSR_IRQ_TEST_MODE_SIZE 1
static inline uint32_t irq_test_mode_read(void) {
	return csr_read_simple((CSR_BASE + 0x1004L));
}
static inline void irq_test_mode_write(uint32_t v) {
	csr_write_simple(v, (CSR_BASE + 0x1004L));
}
#define CSR_IRQ_TEST_EDGE_ADDR (CSR_BASE + 0x1008L)
#define CSR_IRQ_TEST_EDGE_SIZE 1
static inline uint32_t irq_test_edge_read(void) {
	return csr_read_simple((CSR_BASE + 0x1008L));
}
static inline void irq_test_edge_write(uint32_t v) {
	csr_write_simple(v, (CSR_BASE + 0x1008L));
}
#define CSR_IRQ_TEST_EV_STATUS_ADDR (CSR_BASE + 0x100cL)
#define CSR_IRQ_TEST_EV_STATUS_SIZE 1
static inline uint32_t irq_test_ev_status_read(void) {
	return csr_read_simple((CSR_BASE + 0x100cL));
}
#define CSR_IRQ_TEST_EV_STATUS_I0_OFFSET 0
#define CSR_IRQ_TEST_EV_STATUS_I0_SIZE 1
static inline uint32_t irq_test_ev_status_i0_extract(uint32_t oldword) {
	uint32_t mask = 0x1;
	return ( (oldword >> 0) & mask );
}
static inline uint32_t irq_test_ev_status_i0_read(void) {
	uint32_t word = irq_test_ev_status_read();
	return irq_test_ev_status_i0_extract(word);
}
#define CSR_IRQ_TEST_EV_PENDING_ADDR (CSR_BASE + 0x1010L)
#define CSR_IRQ_TEST_EV_PENDING_SIZE 1
static inline uint32_t irq_test_ev_pending_read(void) {
	return csr_read_simple((CSR_BASE + 0x1010L));
}
static inline void irq_test_ev_pending_write(uint32_t v) {
	csr_write_simple(v, (CSR_BASE + 0x1010L));
}
#define CSR_IRQ_TEST_EV_PENDING_I0_OFFSET 0
#define CSR_IRQ_TEST_EV_PENDING_I0_SIZE 1
static inline uint32_t irq_test_ev_pending_i0_extract(uint32_t oldword) {
	uint32_t mask = 0x1;
	return ( (oldword >> 0) & mask );
}
static inline uint32_t irq_test_ev_pending_i0_read(void) {
	uint32_t word = irq_test_ev_pending_read();
	return irq_test_ev_pending_i0_extract(word);
}
static inline uint32_t irq_test_ev_pending_i0_replace(uint32_t oldword, uint32_t plain_value) {
	uint32_t mask = 0x1;
	return (oldword & (~(mask << 0))) | (mask & plain_value)<< 0 ;
}
static inline void irq_test_ev_pending_i0_write(uint32_t plain_value) {
	uint32_t oldword = irq_test_ev_pending_read();
	uint32_t newword = irq_test_ev_pending_i0_replace(oldword, plain_value);
	irq_test_ev_pending_write(newword);
}
#define CSR_IRQ_TEST_EV_ENABLE_ADDR (CSR_BASE + 0x1014L)
#define CSR_IRQ_TEST_EV_ENABLE_SIZE 1
static inline uint32_t irq_test_ev_enable_read(void) {
	return csr_read_simple((CSR_BASE + 0x1014L));
}
static inline void irq_test_ev_enable_write(uint32_t v) {
	csr_write_simple(v, (CSR_BASE + 0x1014L));
}
#define CSR_IRQ_TEST_EV_ENABLE_I0_OFFSET 0
#define CSR_IRQ_TEST_EV_ENABLE_I0_SIZE 1
static inline uint32_t irq_test_ev_enable_i0_extract(uint32_t oldword) {
	uint32_t mask = 0x1;
	return ( (oldword >> 0) & mask );
}
static inline uint32_t irq_test_ev_enable_i0_read(void) {
	uint32_t word = irq_test_ev_enable_read();
	return irq_test_ev_enable_i0_extract(word);
}
static inline uint32_t irq_test_ev_enable_i0_replace(uint32_t oldword, uint32_t plain_value) {
	uint32_t mask = 0x1;
	return (oldword & (~(mask << 0))) | (mask & plain_value)<< 0 ;
}
static inline void irq_test_ev_enable_i0_write(uint32_t plain_value) {
	uint32_t oldword = irq_test_ev_enable_read();
	uint32_t newword = irq_test_ev_enable_i0_replace(oldword, plain_value);
	irq_test_ev_enable_write(newword);
}
```

mohamedElbouazzati, 2023-02-23:
@adamhlt Thanks alot for your answer.
Now, it works fine for me.
I think it is required to be updated with a pull request.

enjoy-digital, 2023-02-24:
Thanks @mohamedElbouazzati and @adamhlt, I'll re-opening to have a closer look at it and the PR.

enjoy-digital, 2023-02-24:
https://github.com/enjoy-digital/litex/pull/1613 has been merged, thanks!

closed 2023-02-24 by enjoy-digital