enjoy-digital/litex #1614: AttributeError: 'Platform' object has no attribute 'startswith'

jwfaye, 2023-02-23:
Hello, 

I updated litex to retrieve last changes but I got an error with the parser. 

Here is what I do : 

from litex.soc.integration.soc import LiteXSoCArgumentParser
    parser = LiteXSoCArgumentParser(description="LiteX AMP Dual-Core SoC generator on De10Lite")
    target_group = parser.add_argument_group(title="Target options")
    target_group.add_argument("--platform",       default=terasic_de10lite.Platform())
    target_group.add_argument("--toolchain",      default="quartus",           help="FPGA toolchain (vivado, symbiflow or yosys+nextpnr).")
    target_group.add_argument("--sys-clk-freq",   default=50e6,                help="System clock frequency.")
    target_group.add_argument("--bus_data_width", default=16,                  help="Super SoC bus data width.")
    target_group.add_argument('--config_file',    help='Configuration file',   required=True)
    target_group.add_argument('--config',         help='Configuration number', required=True)
    target_group.add_argument("--build",          action="store_true",         help="Build bitstream.")
    target_group.add_argument("--build_dir",      default='',                  help="Base output directory.")
    target_group.add_argument("--load",           action="store_true",         help="Load bitstream.")
    builder_args(parser)
    args = parser.parse_args()

And I got this trace : 
Traceback (most recent call last):
  File "./amp.py", line 354, in <module>
    main()
  File "/amp.py", line 318, in main
    args = parser.parse_args()
           ^^^^^^^^^^^^^^^^^^^
  File "/opt/litex_root/litex/litex/build/parser.py", line 181, in parse_args
    platform_cls = importlib.import_module(platform).Platform
                   ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/home/jofaye/miniconda3/envs/litex/lib/python3.11/importlib/__init__.py", line 117, in import_module
    if name.startswith('.'):
       ^^^^^^^^^^^^^^^
AttributeError: 'Platform' object has no attribute 'startswith'

trabucayre, 2023-02-23:
Hi,
Your target uses `LiteXSoCArgumentParser`. This class now inherit for `LiteXArgumentParser`.
Argument `--platform` is a specific case (as used by **simple.py**) to let user provides platform name. It a must be a string and `LiteXArgumentParser` use it to instanciate corresponding platform's class. In your case instead of a string you provide a class instance. This why an error is.

`LiteXArgumentParser` also inject some default, common args (`build`, `toolchain`, `load`) and create group `parser.add_argument_group(title="Target options")`.

Since your code seems based on *de10lite* you can check [this part](https://github.com/litex-hub/litex-boards/blob/master/litex_boards/targets/terasic_de10lite.py#L91) to see how to use `LiteXArgumentParser`

It may be looks like:

```
from litex.build.parser import LiteXArgumentParser
parser = LiteXArgumentParser(platform=terasic_de10lite.Platform, description="LiteX SoC on DE10-Lite.")
parser.add_target_argument("--sys-clk-freq",        default=50e6, type=float, help="System clock frequency.")
target_group.add_argument("--bus_data_width", default=16, help="Super SoC bus data width.")
target_group.add_argument('--config_file', help='Configuration file', required=True)
target_group.add_argument('--config', help='Configuration number', required=True)
target_group.add_argument("--build_dir", default='', help="Base output directory.")
args = parser.parse_args()
```

jwfaye, 2023-02-24:
Hello,

Thank you @trabucayre your answer helped me to understand the logic of the LiteXArgumentParser. I managed to modify the code to make it work . thank you. I am closing the issue.