enjoy-digital/litex #1615: json2dts issue for Linux on Litex

AEW2015, 2023-02-25:
When building Linux on Litex for Nexys Video, the lastest change is causing an issue.

`dtc  -O dtb -o build/nexys_video/nexys_video.dtb build/nexys_video/nexys_video.dts

build/nexys_video/nexys_video.dts:103.36-108.15: ERROR (phandle_references): /soc/clint@f0010000: Reference to non-existent node or label "L4"

build/nexys_video/nexys_video.dts:103.36-108.15: ERROR (phandle_references): /soc/clint@f0010000: Reference to non-existent node or label "L4"

`

AEW2015, 2023-02-25:
It looks like its this change:
https://github.com/enjoy-digital/litex/commit/01e9a5432114e9487f9e56ebc8860e54b9118e20

enjoy-digital, 2023-02-27:
Thanks @AEW2015 for the feedback. This should be fixed with https://github.com/enjoy-digital/litex/commit/991198ec2ee119c405a21376cb9a47088655edf6. I'll have a closer look at this when continuing Rocket support in the json2dts script.

Dolu1990, 2023-02-27:
Hi,

Right.
L4 would refer to the internal interrupt controller of a given RISC-V code.

Here is from another project : 

CPU definition :
https://github.com/SpinalHDL/buildroot-spinal-saxon/blob/main/boards/common/dts/linux_cpu.dts#L12

Here is an example for the PLIC (not the CLINT) but that's similar concerning the Lx stuff : 
https://github.com/SpinalHDL/buildroot-spinal-saxon/blob/main/boards/common/dts/linux_plic_link.dts

enjoy-digital, 2023-02-27:
Thanks @Dolu1990. I was just preparing Rocket and NaxRiscv support while doing and should have tested regression with VexRiscv-SMP. I'll do this when continuing this.

Dolu1990, 2023-02-27:
Just for understanding : 
&L4 3 &L4 7

mean  : 
"CPU named L4" "Machine software interrupt"  "CPU named L4" "Machine timer interrupt"
Numbers being translated from 
[image]

enjoy-digital, 2023-02-27:
Thanks @Dolu1990, this will be very useful!

gsomlo, 2023-02-27:
If you take a look at one of the auto-generated *.dts sample files in
pythondata-cpu-rocket (e.g.,
pythondata_cpu_rocket/verilog/generated-src/freechips.rocketchip.system.LitexFull4DConfig.dts):

blocks (and sub-blocks) are assigned auto-generated labels, which can
be used elsewhere in the DT source file to reference them. It happens
that the *first* CPU is assigned label "L4" during chisel elaboration
(the overall "cpus" node containing all CPUs is "L24" in that file,
but other variants will have a different label).

It's IMO a *coincidence* that the first CPU is labeled "L4" across
several of the pre-generated variant sample DTS files.

Whatever the first CPU's assigned label is (let's use "L42" as an
example), it can be referenced from elsewhere in the DTS as "&L42".

HTH,
--Gabriel

Renemiess, 2023-11-09:
Hi, I wanna know how to transfer json to dts

closed 2024-01-12 by enjoy-digital