enjoy-digital/litex #1618: PHY for LCDs with 8080 bus

suarezvictor, 2023-02-28:
With some glue I wrote for cores from others I was able to show the LiteX boot console on a LCD with a 8080 bus. Since it doesn't need changes to the LiteX API for video handling (it's just a new PHY), it could be mainstreamed with low effort.

https://github.com/suarezvictor/micropython/commit/94903839bb6391f5aba9ef2f0d743af2f7b43d12#diff-f689a3f83aaaa1076106b25044284fc4d4c04e2cd3e173ef6a9d9e74e6840348R32

[image]

enjoy-digital, 2023-02-28:
Nice @suarezvictor, I'll have a closer look. If it's only a Video PHY, it would indeed be very useful to integrate it. Do you want to do a PR? Even if you don't have time to polish it, I could have a look at it once integrated.

suarezvictor, 2023-02-28:
Yes in the example you can see how I just call `add_video_terminal` with the glue PHY.
I'm afraid I don't have time to prepare proper PRs now, that may need some more polishing (i.e. it needs some more core to support other resolutions)

suarezvictor, 2023-02-28:
Yes, just a PHY that glues to a existing LCD controller, in the code you can see how I just call `add_video_terminal`
I'm afraid I have no time to prepare the PRs, I'm publishing that in the hope that someone else can contribute